reserve85/HoymilesZeroExport #1: Hilfe beim Ausführen unter Windows

Ollipop030, 2023-02-13:
Hi,

ich würde das Script gern auf einem Windows Rechner laufen lassen. Leider habe ich vom Programmieren oder von python keinen Plan. Ich komme aus der Hardwarecke, und verstehe noch Basic vom C64 und Batchdateien in DOS.

Kann ich das Script so unter zB Tonny ausführen? Ich habs mal probiert, klappt aber nicht. Die IP Adressen habe ich natürlich geändert. Desweiteren habe ich keinen Tasmota, sondern einen anderen Lesekopf, aber der spukt auch json aus. Die richtige Adresse ist bei mir
http://{tasmotaIP}/pages/getinformation.php?heute&meterindex=1

Was aber noch geändert werden muss: Anders als Tasmota zeigt mein Zähler nicht negativ bei Einspeisung, sondern ich habe zwei positive Werte (1.8.0 für Bezug und 2.7.0 für Einspeisung. Um aber einfach einen negativen Wert zu bekommen kann man Netzbezug (1.8.0) abzüglich Einspeisung (2.7.0) rechnen. Beispiel: Bei Netzbezug von 500W ist ja 1.8.0=500, 2.7.0=0, also logischerweise 500W. Bei Einspeisung ist 1.8.0=0, 2.7.0=500, also 0 - 500 = -500.

Vielleicht hast du ja Bock mir ein wenig zu helfen.

reserve85, 2023-02-13:
Hi,

du kannst das Script einfach ausführen wenn du Python in Version 3 installiert hast + du musst nach der Installation das Python-Paket "requests" nachinstallieren. 
Das geht sehr einfach in der Kommandozeile (cmd) mit `pip3 install requests`. 
Ansonsten nutze ich zum debuggen und entwickeln Visual Code mit dem Python Plugin, das funktioniert bei mir perfekt.

So wie ich das jetzt verstehe brauchst du nur den Rückgabewert von Verbrauch und Einspeisung aus dem JSON exportieren und subtrahieren (klingt zumindest einfach).
Was kommt da zurück wenn du deine URL aufrufst? Poste mal die JSON Rückgabe...

Ollipop030, 2023-02-13:
Und da geht es schon los. Das hatte ich schon probiert, bekomme immer SyntaxError: invalid syntax. 

Der Ausgabe vom Stromzähler sieht so aus:

{
  "Leistung170":475,
  "Leistung171":0,
  "Leistung172":0,
  "Leistung173":0,
  "Leistung270":0,
  "Leistung271":0,
  "Leistung272":0,
  "Leistung273":0,
  "Waehrung":"EUR",
  "Stand300":null,
  "M300":0,
  "Stand180":0,
  "Stand181":0,
  "Stand182":0,
  "Stand280":0,
  "Stand281":0,
  "Stand282":0,
  "Kwh180":0,
  "Kwh181":0,
  "Kwh182":0,
  "Kwh280":0,
  "Kwh281":0,
  "Kwh282":0,
  "Kwh300":0,
  "Betrag180":0,
  "Betrag181":0,
  "Betrag182":0,
  "Betrag280":0,
  "Betrag281":0,
  "Betrag282":0,
  "Betrag300":0,
  "Metertype":"Strom"
}

Ollipop030, 2023-02-13:
Ok, requests ist installiert. Dachte ich führe das einfach in der Kommandozeile aus.

reserve85, 2023-02-13:
Das JSON sollte bei dir dann so auszuwerten sein:

ParsedData = requests.get(url = f'http://{tasmotaIP}/pages/getinformation.php?heute&meterindex=1).json()
Watts = int(parsed_data['Leistung170'] - parsed_data['Leistung270'])

in GetPowermeterWatts ersetzen.

Ollipop030, 2023-02-13:
Läuft! Heute ist allerdings das Wetter totaler Mist. Das Limit Limit wird logischerweise auf 100% gesetzt, Komme gerade auf knapp 170 Watt Leistung mit einem HM1500 Ost/West und HM300 Süd.
Das Limit setze ich beim HM1500, den HM300 lasse ich einfach auf 100%. Der allein bekommt eh nichtmal die Grundlast abgedeckt. Dazu eine Frage:
Wird das Script noch dahingehend aktualisiert, das mehrere Inverter unterstützt werden? Das Script weiß ja, wieviel Leistung zur Verfügung steht. Somit könnte man einen Prozenzwert errechnen, der dann an alle WR geschickt wird.

reserve85, 2023-02-13:
Schön zu hören dass es läuft.
Ich selbst habe nur einen 1500 Hoymiles am Netz und kann somit keine zwei Inverter testen. 
Somit fällt das von meiner Seite aus leider ins Wasser, war auch erstmal nicht mein Ziel.

Ollipop030, 2023-02-13:
Kurzes Update: Das Script meldet gerade, das der WR nicht erreichbar wäre, AHOY sagt das selbe. Den Interval im Script habe ich auf 10 Sekunden Sekunden gesetzt, ist wahrscheinlich zuviel für die kleine DTU? Ich setze es wieder hoch. Die Kommunikation ging auch erst wieder nach Neustart der DTU.

reserve85, 2023-02-13:
Ja, das habe ich auch festgestellt und daher etwas erhöht. Mit 20s läuft es absolut zuverlässig. 
Im Ahoy habe ich das Intervall auf 6s eingestellt.

Ollipop030, 2023-02-13:
Guter Tipp. Mein Lesekopf aktualisiert auch nur alle 10 Sekunden die Daten, Tasmota ist da glaub ich schneller.

Hier kam grad kurz ein bisschen Überschuss, ist das richtig das im Sekundentakt zB "powermeter -24" ausgegeben wird? Und die Meldung "Not enough energy producing: limit already at maximum" kommt warscheinlich, weil die Toleranz mit 25 angegeben ist?

Ollipop030, 2023-02-13:
Verstanden. Gibt es da einen Wert, den ich im Script ändern kann? Wie gesagt, mein Lesekopf haut die Werte eh nur alle 10 Sekunden raus.

reserve85, 2023-02-13:
ne lass das mal so, du kannst das ja nicht 1:1 synchronisieren wann du im Script abfragst du wann der Zähler sich aktualisiert. Wenn du es sekündlich abfragst kann das Script sofort reagieren und das Limit erhöhen. Andernfalls wäre das im worst case erst 10s später der Fall bei dem du Energie einkaufst obwohl du sie generieren könntest.

Ollipop030, 2023-02-13:
Stimmt auch wieder. Aber noch was: Wenn, so wie heute, das Limit eh immer auf 100% gesetzt wird, muss es denn dann überhaupt immer wieder zur DTU gesendet werden? Das könnte man ja vermeiden, indem man prüft, ob die letzte Änderung auf 100% erfolgte, und die Leistung weiterhin positiv ist. So lange muss ja kein Limit geändert werden. Erst wenn der Wert kleiner 100% sein muss, den dann zur DTU schicken. Entlastet den ESP vielleicht ein wenig. Hab eh das Gefühl, das die kleinen ESPs schnell überfordert sind.

reserve85, 2023-02-13:
Ja, stimmt prinzipiell. Ich hatte das auch schon so implementiert, allerdings hatte ich dann auch schonmal den Fall, dass das Kommando zum Limitieren von Ahoy nicht geklappt hat. Dann bekomme ich das nicht mit. Daher habe ich mich dazu entschieden alle 20s das zu senden.

Ollipop030, 2023-02-13:
Läuft bisher immer noch, manchmal kommt aber die Meldung "ERROR    'NoneType' object is not subscriptable". Das Script selbst  läuft aber weiter. Was könnte das heißen?

reserve85, 2023-02-13:
Puh, kannst du mal das log posten? Hatte ich bisher noch nicht.

Ollipop030, 2023-02-13:
Kann ich machen, aber mehr kam auch nicht als Fehlermeldung. Trat auch komplett random auf. Manchmal innerhalb von 5 Minuten, dann von 17:45 bis Sonnenuntergang gar nicht mehr.
Ich vermute aber den bereits erwähnten HM300. Dieser kommuniziert ja auch regelmäßig mit der DTU, und heute Nachmittag zeigte die DTU diesen WR als nicht erreichbar. Vielleicht überschneidet sich da was bei Limitsetzen. Ich bleibe am Ball und berichte...

reserve85, 2023-02-13:
Ja, konnte das gerade auch provozieren: der Fehler erscheint wenn das empfangene JSON Objekt "None" ist, also leer. Entweder vom Ahoy oder von deinem Stromzähler.

du könntest z.B. mal eine Abfrage auf None unter jedes ParsedData = ... einfügen:
```
def GetHoymilesAvailable():
    ParsedData = requests.get(url = f'http://{ahoyIP}/api/index').json()
    if ParsedData == None:
      logging.info("ParsedData from GetHoymilesAvailable is empty!")
      return False
    Reachable = bool(ParsedData["inverter"][0]["is_avail"])
    logging.info("HM reachable: %s",Reachable)
    return Reachable
```
dann kannst du im Log sehen wo es herkommt.

Ollipop030, 2023-02-14:
Guten morgen,

ich habe die neuen Zeilen hinzugefügt und lasse es jetzt laufen, sobald ich was sehe melde ich mich.

Ollipop030, 2023-02-14:
Hier kam der erste Mal die Meldung:

2023-02-14 08:49:09 INFO     HM reachable: True
2023-02-14 08:49:09 INFO     powermeter: 157  Watt
2023-02-14 08:49:09 INFO     setting new limit to 1500  Watt
2023-02-14 08:49:29 INFO     Error: ParsedData is empty (in function GetHoymilesAvailable)
2023-02-14 08:49:49 INFO     HM reachable: True
2023-02-14 08:49:49 INFO     powermeter: 184  Watt
2023-02-14 08:49:49 INFO     setting new limit to 1500  Watt

reserve85, 2023-02-14:
Vermutlich ist Ahoy bei dir zu ausgelastet. Deaktiviere mal den zweiten Inverter? Bei mir kam das noch nie. Welche Version nutzt du?

Ollipop030, 2023-02-14:
Gerade die 0.5.87 auf einem 8266. Ich finde die ESP32 eigentlich besser, aber da gibt es gerade ein paar Probleme. Bisher kam der Fehler noch nicht wieder, ich lasse den WR noch ein wenig drin bis zum nächsten Fehler.

reserve85, 2023-02-14:
ok ich hab den ESP32 38Pin und auch V0.5.87

Ollipop030, 2023-02-14:
So, HM300 ist raus, trotzdem der Fehler:

2023-02-14 09:42:49 INFO     powermeter: 41  Watt
2023-02-14 09:42:49 INFO     setting new limit to 1500  Watt
2023-02-14 09:43:09 INFO     Error: ParsedData is empty (in function GetHoymilesAvailable)
2023-02-14 09:43:29 INFO     Error: ParsedData is empty (in function GetHoymilesAvailable)
2023-02-14 09:43:49 INFO     HM reachable: True
2023-02-14 09:43:49 INFO     powermeter: 32  Watt
2023-02-14 09:43:49 INFO     setting new limit to 1500  Watt
2023-02-14 09:44:09 INFO     HM reachable: True

Gleichzeitig macht die DTU das hier:

09:43:05 I: resetPayload: id: 0
09:43:05 I: (#0) prepareDevInformCmd
09:43:05 I: TX 27B Ch23 | 15 83 92 49 15 81 92 14 22 80 05 00 63 eb 49 99 00 00 00 00 00 00 00 00 ca 4c 26 
09:43:17 I: resetPayload: id: 0
09:43:17 I: (#0) prepareDevInformCmd
09:43:17 I: TX 27B Ch40 | 15 83 92 49 15 81 92 14 22 80 05 00 63 eb 49 a5 00 00 00 00 00 00 00 00 cb 4d 1a 
09:43:17 I: procPyld: cmd:  0x5
09:43:17 I: procPyld: txid: 0x95
09:43:29 I: resetPayload: id: 0
09:43:29 I: (#0) enqueuedCmd: 0xb
09:43:29 I: (#0) prepareDevInformCmd

Keine Ahnung ob dir das hilft.

Ollipop030, 2023-02-14:
Vielleicht könnte man das noch ein wenig verschönern:

2023-02-14 10:54:14 INFO     powermeter: 3  Watt
2023-02-14 10:54:14 INFO     setting new limit to 1500  Watt

Das Limit muss ja nur um 3 Watt erhöht werden. Wegen meines langsamen Zählers speise ich dann im ungünstigsten Fall 10 Sekunden mit allem was da ist ins Netz.

reserve85, 2023-02-14:
jein, kann auch sein dass der Zähler "nach oben schießt" und du nur den kurzen zwischenstand mitbekommst. Daher hab ich das so programmiert, dass der Hoymiles immer voll aussteuert und dann von oben wieder runterregelt. Die 20s bis er wieder das Limit nachregelt nehm ich in kauf.

Ollipop030, 2023-02-14:
Der Gedanke kam mir gerade so: Könnte man das Script ein wenig variabler gestalten? ZB die ganze Domäne mit den json Daten oben im Script angibt, statt nur der IP. Dann natürlich die Variablen für Bezug/Einspeisung. Und einen Switch, ob der Zähler negativ zeigt, oder wie bei mir 2 positiven Werten arbeitet. Das wäre dann ein wenig benutzfreundlicher für die ganz Ahnungslosen wie mich. 
Übrigens: Kein Fehler mehr vorgekommen bisher, liegt tatsächlich am zweiten Inverter.

reserve85, 2023-02-15:
Hi,

ja das kann man in Zukunft bestimmt noch verbessern. Ich habe das ja ursprünglich explizit nur für die Tasmota Web-API entwickelt, daher ist das alles eben nicht generisch. Ist auch schwierig, da jede API und jeder Zähler eben ein anderes JSON zurückschickt.

Ollipop030, 2023-02-15:
Läuft übrigens wunderbar, heute (bis jetzt) 2,74 kWh produziert, davon 0,18 eingespeist.