reserve85/HoymilesZeroExport #2: Prüfung ob DTU online ist

Ollipop030, 2023-02-21:
Guten Morgen, mittlerweile läuft das Script auf meinem Pi Hole, da ist mir folgendes aufgefallen. Sobald die FritzBox die Nachtschaltung (WLAN aus) aktiviert, kommt es zu einer Fehlermeldung:

2023-02-21 00:29:25 INFO     HM reachable: False
2023-02-21 00:29:42 INFO     HM reachable: False
2023-02-21 00:29:59 INFO     HM reachable: False
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 159, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 80, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 70, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1260, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1306, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1255, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1030, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 970, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in connect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 168, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x763b0f10>: Failed to establish a new connection: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.0.19', port=80): Max retries exceeded with url: /api/index (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x763b0f10>: Failed to establish a new connection: [Errno 113] No route to host'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/smb/test.py", line 72, in <module>
    if GetHoymilesAvailable():
  File "/mnt/smb/test.py", line 32, in GetHoymilesAvailable
    ParsedData = requests.get(url = f'http://{ahoyIP}/api/index').json()
  File "/usr/lib/python3/dist-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.0.19', port=80): Max retries exceeded with url: /api/index (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x763b0f10>: Failed to establish a new connection: [Errno 113] No route to host'))

Man kann hier schön sehen, das pünktlinch um 00:30 WLAN abgeschaltet wird. Das Script ist dann aber beendet, und nimmt am nächsten morgen natürlich die Arbeit nicht wieder auf.

Könnte man evtl. eine Prüfung einbauen, die nachschaut, ob die DTU online ist, aber bei einem false Ergebnis nicht direkt aussteigt? Ähnlich wie GetHoymilesAvailable, nur als GetDTUAvailable.

reserve85, 2023-02-21:
Danke für die Meldung, ich habe gesehen, dass das Fehlerhandling in den letzten Zeilen falsch war. Ich habe das nun geändert und aktualisiert. Kannst du das bitte auch bei dir testen?

Ollipop030, 2023-02-21:
Danke für die schnelle Antwort. Habe ich bei mir geändert und berichte dann.

Ollipop030, 2023-02-22:
Hat geklappt. Die Leistung wird wieder geregelt, sobald WLAN wieder da ist (und der Wechselrichter ausgeschlafen ist).

reserve85, 2023-02-22:
perfekt, dann mach ich hier zu

commit 8022dd70 (2024-09-08): Update HoymilesZeroExport.py

fix #2
closed 2023-02-22 by reserve85