reserve85/HoymilesZeroExport #3: Powerlimit bei kleinstem Netzbezug immer 100%

Ollipop030, 2023-02-25:
Hi, haben wir schonmal kurz drüber gesprochen: Wenn der Netzbezug kurzzeitig wieder steigt, wird das Limit immer auf 100% geregelt. Könnte man das Limit nicht um Netzbezug + powermeterTolerance erhöhen? Ich merke, gerade wenn der 3D Drucker läuft, das Limit ständig auf Max erhöht wird. Die Schrittmotoren ziehen sehr viel Energie beim Anfahren. Manchmal ist dann der Netzbezug nur bei 50 Watt, aber Limit geht dann direkt wieder auf 100%. Es ist erst halb drei, habe heute schon über 1 kWh eingespeist.

Kannst du mir einen Tipp geben welche Änderungen dafür nötig sind?

reserve85, 2023-02-25:
ja, das kannst du ändern, hoffentlich funktioniert es auch:

```
if powermeterWatts > 0:
	newLimitSetpoint = hoymilesMaxWatt
	setLimit(hoymilesInverterID, newLimitSetpoint)
	if int(LoopIntervalInSeconds) - SetLimitDelay - x * PollInterval <= 0:
		break
	else:
		time.sleep(int(LoopIntervalInSeconds) - SetLimitDelay - x * PollInterval)
	break
```

ändern in 

```
if powermeterWatts > 0:
	break
```

so wird nicht ausgeregelt sondern weiter unten im code dann "angeglichen".

Ollipop030, 2023-02-25:
Danke, ich teste mal. Wetter soll ja gut werden die nächsten Tage.

Ollipop030, 2023-02-26:
Guten Morgen. Das scheint zu klappen:

2023-02-26 10:09:08 INFO     powermeter: -80  Watt
2023-02-26 10:09:09 INFO     powermeter: -80  Watt
2023-02-26 10:09:10 INFO     powermeter: -80  Watt
2023-02-26 10:09:11 INFO     powermeter: -80  Watt
2023-02-26 10:09:12 INFO     powermeter: 113  Watt
2023-02-26 10:09:12 INFO     setting new limit to 713  Watt
2023-02-26 10:09:25 INFO     HM reachable: True
2023-02-26 10:09:25 INFO     powermeter: 113  Watt
2023-02-26 10:09:25 INFO     setting new limit to 901  Watt
2023-02-26 10:09:42 INFO     HM reachable: True
2023-02-26 10:09:42 INFO     powermeter: 122  Watt
2023-02-26 10:09:42 INFO     setting new limit to 1098  Watt
2023-02-26 10:09:59 INFO     HM reachable: True
2023-02-26 10:09:59 INFO     powermeter: 123  Watt
2023-02-26 10:09:59 INFO     setting new limit to 1296  Watt
2023-02-26 10:10:17 INFO     HM reachable: True
2023-02-26 10:10:17 INFO     powermeter: 123  Watt
2023-02-26 10:10:17 INFO     setting new limit to 1494  Watt

Allerdings macht mich das Timing etwas stutzig. Sobald der Wert positiv wird, wird auch das Limit erhöht, im Log bei 10:09:12. Um 10:09:25 wird aber schon wieder erhöht, eingestellt sind aber 17 Sekunden bei mir. Noch krasser war es hier:

2023-02-26 10:02:08 INFO     powermeter: -102  Watt
2023-02-26 10:02:09 INFO     powermeter: -102  Watt
2023-02-26 10:02:10 INFO     powermeter: -102  Watt
2023-02-26 10:02:11 INFO     powermeter: -102  Watt
2023-02-26 10:02:12 INFO     powermeter: 24  Watt
2023-02-26 10:02:12 INFO     setting new limit to 986  Watt
2023-02-26 10:02:17 INFO     HM reachable: True
2023-02-26 10:02:17 INFO     powermeter: 24  Watt
2023-02-26 10:02:17 INFO     setting new limit to 1085  Watt
2023-02-26 10:02:34 INFO     HM reachable: True
2023-02-26 10:02:34 INFO     powermeter: -151  Watt

Hier wurde um 10:02:12 erhöht, dann fünf Sekunden später wieder. Der geringe Zeitunterschied ist aber nur zwischen der ersten und zweiten Änderung vorhanden, danach sind es immer wieder die eingestellten 17 Sekunden.

Ollipop030, 2023-02-26:
Hier noch ein Beispiel:

2023-02-26 12:11:19 INFO     powermeter: 72  Watt
2023-02-26 12:11:19 INFO     setting new limit to 465  Watt
2023-02-26 12:11:24 INFO     HM reachable: True
2023-02-26 12:11:24 INFO     powermeter: 72  Watt
2023-02-26 12:11:24 INFO     setting new limit to 612  Watt
2023-02-26 12:11:41 INFO     HM reachable: True
2023-02-26 12:11:41 INFO     powermeter: -325  Watt

Auch hier nur 5 Sekunden zwischen der Änderung. Wäre schön, wenn dann noch die 17 Sekunden eingehalten werden. Habe übrigens das alle deine Änderung von gestern übernommen.

reserve85, 2023-02-26:
ja das ist mit Absicht von mir so, damit halt das limit sofort erhöht wird als "Zwischenlimit setzen". Danach wird nur noch die "Restzeit" zu den 17sekunden abgewartet. So bin ich am besten gefahren. Kannst das natürlich auch anpassen wenn du willst: 

```
if int(loopIntervalInSeconds) - setLimitDelay - x * pollInterval <= 0:
	break
else:
	time.sleep(int(loopIntervalInSeconds) - setLimitDelay - x * pollInterval)
```
ändern in 
`time.sleep(loopIntervalInSeconds)`

dann wartest du nach der sofortigen Anpassung wieder 17s.

Ollipop030, 2023-02-26:
Macht bei mir mehr Sinn, denn ich bekomme ja nur alle 10 Sekunden neue Werte vom Stromzähler. Ich probiere das mal. Ansonst läuft das sehr gut.

Übrigens ein Hinweis:
jumpToMaxlimitOnGridUsage = bool(True) # when powermeter > 0: (true): always jump to maxLimit of inverter; (false): increase limit based on previous limit

sollte geändert werden in: 
jumpToMaxlimitOnGridUsage = bool(True) # when powermeter > 0: (True): always jump to maxLimit of inverter; (False): increase limit based on previous limit

True oder False müssen jeweils groß geschrieben werden.

reserve85, 2023-02-26:
danke, übernehme ich noch

Ollipop030, 2023-02-28:
` time.sleep(loopIntervalInSeconds)` habe ich wieder rausgenommen, gefällt mir andersherum besser. Ich mach hier dicht.

closed 2023-02-28 by Ollipop030