reserve85/HoymilesZeroExport #4: GetPowermeterWatts konfigurierbar

schliepi, 2023-02-28:
Gutes Script, danke!

`    Watts = int(ParsedData["StatusSNS"]["SML"]["curr_w"])`
extrahiert bei meinem Stromzähler leider nichts, da das Tasmota-Script anders aufgebaut ist und diesen JSON liefert:

{"StatusSNS":{"Time":"2023-02-28T11:41:19","DWS74":{"Total":1234.567,"TotalOut":1.234,"Power":2,"Server_id":"xxx"}}}

Wie wäre es hier mit zwei weiteren Konfigurationsmöglichkeiten für die Werte "SML" und "curr_w"?

reserve85, 2023-02-28:
ja, schwieriges Thema. Ist eben bei jedem Zähler unterschiedlich (je nach Script) und muss daher individuell angepasst werden.
Bei dir läuft es dann wenn du "SML" durch "DWS74" ersetzt und "curr_w" durch "Power"?
Ich kann die beiden Werte prinzipiell mit oben in die Types einfügen...

schliepi, 2023-02-28:
Ja, läuft mit den beiden Werten ersetzt super.

Konfiguration der Werte analog zur Tasmota-Konfiguration

https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-definition
<jsonPrefix> | Prefix for Web UI and MQTT JSON payload. Up to 7 characters.
Bei Dir "SML", bei mir "DWS74" - Vorschlag Konfigurationstype: meterDefinitionMqttPrefix

https://tasmota.github.io/docs/Smart-Meter-Interface/#meter-metrics
<var> | MQTT label (max. 23 characters)
Bei Dir "curr_w", bei mir "Power"  - Vorschlag Konfigurationstype: meterMetricsMqttLabel 

Kriegst auch noch ein paar weitere Issues :-)

reserve85, 2023-02-28:
Ja habs mal so benannt:

```
# the following three constants describes how to navigate through the Tasmota-JSON
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TAS_JSON_STATUS = 'StatusSNS'
TAS_JSON_SCRIPT_PREFIX = 'SML'
TAS_JSON_POWER = 'curr_w'
```

schliepi, 2023-02-28:
Ja, gut so. Danke!

closed 2023-02-28 by reserve85

Ollipop030, 2023-02-28:
Habe die Änderungen auch bei mir übernommen (obwohl es keine wirkliche Änderung seit vorgestern gab). Bei mir muss das ganze so aussehen:

`#TAS_JSON_STATUS = 'StatusSNS'
#TAS_JSON_PAYLOAD_MQTT_PREFIX = 'SML' # Prefix for Web UI and MQTT JSON payload
TAS_JSON_POWER_MQTT_LABEL = 'Leistung170' # Power-MQTT label
TAS_JSON_E_POWER_MQTT_LABEL = 'Leistung270' # ExportPower-MQTT label`

Und dann natürlich `Watts = int(ParsedData[TAS_JSON_POWER_MQTT_LABEL] - ParsedData[TAS_JSON_E_POWER_MQTT_LABEL])`

da der Zähler bei mir zwei Werte ausgibt. Sollte klappen, Sonne ist schon zu niedrig um Einspeiseleistung zu begrenzen. Aber schön zu sehen, dass es immer flexibler wird. 
Vielleicht wäre noch eine variable Domänenangabe interessant, sowas wie `TAS_JSON_DOMAIN = '/cm?cmnd=status%2010'`