reserve85/HoymilesZeroExport #6: Suche einen Tester für Shelly 3EM und OpenDTU

reserve85, 2023-03-03:
Es wäre nett, wenn jemand einen Shelly 3EM oder eine OpenDTU (oder auch beides) zu Hause eingerichtet hat und damit die Version aus dem DEV Zweig testen könnte.

Kante001, 2023-03-09:
Hallo, habe eine Shelly 3em direkt hinter dem Zähler und eine OPEN DTU mit einem Balkonkraftwerk.

munter1506, 2023-03-09:
Hallo, ich habe sowohl einen Shelly 3EM als auch einen HICHI IR mit Tasmota Firmware.

Alles in Verbindung mit einem Hoymiles 1500 und OPEN DTU.

Kante001, 2023-03-11:
> Hallo @munter1506<https://github.com/munter1506> und @Kante001<https://github.com/Kante001>: das wäre natürlich bombastisch wenn ihr die Version aus dem DEV Pfad bei euch mal testen könntet:
> https://github.com/reserve85/HoymilesZeroExport/archive/refs/heads/Dev_Positive_TargetPoint.zip
>
> Shelly 3M sollte eigentlich funktionieren, bei OpenDTU bin ich mir eben nicht sicher. Am einfachsten ist es mit Visual Studio Code auszuführen, ihr müsst nur python installieren (Falls ihr mit Windows unterwegs seid).

Hallo,

ich weis garnicht ob man hier kommunizeren kann wenn man antwortet ? Also ich habe jetzt die Datei am laufen und sie arbeitet gut. Die Installation nach Anleitung habe ich nicht hinbekommen. Beim punkt launch installscript to install python and create zero export service

sudo chmod +x install.sh

./install.sh
Hat nicht funktionoiert bzw ich weis nicht ob ich da etwas ersetzen muss.
Falls man hier eine antowort bekommt würde mich interessieren ob man mehrere Wechselrichter damit steuern kann ? also muss man einfach mehrere Dateien starten also für jeden Wechselrichter eine oder irgendwie anders ? verbaue die erst nächsten Monat deswegen kann ich das nicht ausprobieren.

Viele grüße falls das ankommt

reserve85, 2023-03-11:
Hi, kannst du sagen was genau bei dem Install Script nicht funktioniert (log oder ähnliches)? Und welches System du hast? Bei mir auf dem RasPi mit Debian funktionierte es… mehrere Inverter werden aktuell nicht unterstützt, eventuell gehe ich das Thema mal an wenn ich dafür Zeit habe.

reserve85, 2023-03-13:
Ja, ich implementiere das demnächst mal (mehrere Inverter) - dauert aber noch. Testen muss das allerdings jemand anderes, ich habe halt nur einen. 

Ps: dein Screenshot kommt über E-Mail hier nicht an.

Kante001, 2023-03-16:
Hallo,

sorry für die späte antwort.
Also wenn ich es eingebe sieht es so aus

sudo chmod +x install.sh
=
chmod: Zugriff auf 'install.sh' nicht möglich: Datei oder Verzeichnis nicht gefunden

./install.sh
=
bash: ./install.sh: Datei oder Verzeichnis nicht gefunden

bestimmt hätte ich noch irgendwas an dem befehl ändern müssen steht aber so ja nicht in der Anleitung.

Als tester für die Multi wechselregler würde ich mich dann wieder anbieten 😊

Viele grüße

reserve85, 2023-03-16:
ok, du musst natürlich vorher in das angelegte Verzeichnis gehen:
`cd HoymilesZeroExport/`

nehme ich mit in die Readme...

Kante001, 2023-03-19:
Guten Morgen. Wie kann ich dieses automatische Programm eigentlich wieder abschalten ? Ich wollte nochmals was probieren.Mit sudo chmod+x Stop hat es nicht geklappt was wohl zu erwarten war. Viele grüße

reserve85, 2023-03-19:
Das geht mit 
`sudo systemctl stop HoymilesZeroExport.service`

Kante001, 2023-03-19:
Super danke. Ich berichte falls mir was auffällt