reserve85/HoymilesZeroExport #8: Support for multiple inverters

IronicSven, 2023-03-06:
Great work!
Are you planning on adding support for multiple inverters?
Because Ahoy and OpenDTU both support more than one and there are many users using two or more inverters. I'am also planning on buying a second one.

reserve85, 2023-03-06:
Hi Sven,

no that is not planned by myself. I have only one inverter and i can´t install another one at home. I think that needs a lot of testing and tries to calculate the limit for each inverter, especially if they have different output powers.
Sorry!

reserve85, 2023-03-14:
Hi, i just opened a new DEV branch for multiple inverters. 
@Ollipop030 : can you support me here? i can´t test it at home with more than one inverter.
Until now it is completely untestet and i am going to test it with one inverter at the end of this week.

Ollipop030, 2023-03-14:
Sure. But first test results in an error:

2023-03-14 08:32:47 INFO     read config file: D:\Download\HoymilesZeroExport_Config.ini
2023-03-14 08:32:47 INFO     Inverter 0 reachable: True
2023-03-14 08:32:47 INFO     Inverter 1 reachable: True
2023-03-14 08:32:47 INFO     setting new limit to 1800 Watt
Traceback (most recent call last):
  File "D:\Download\HoymilesZeroExport.py", line 204, in <module>
    SetLimit(newLimitSetpoint)
  File "D:\Download\HoymilesZeroExport.py", line 41, in SetLimit
    Factor = HOY_MAX_WATT[i] / GetMaxWattFromAllInverters
TypeError: unsupported operand type(s) for /: 'int' and 'function'

reserve85, 2023-03-14:
ah, i forgot the brackets (): GetMaxWattFromAllInverters()

Ollipop030, 2023-03-14:
Step after step:

2023-03-14 08:45:29 INFO     read config file: D:\Download\Dev\HoymilesZeroExport_Config.ini
2023-03-14 08:45:29 INFO     Inverter 0 reachable: True
2023-03-14 08:45:29 INFO     Inverter 1 reachable: True
2023-03-14 08:45:29 INFO     setting new limit to 1800 Watt
2023-03-14 08:45:29 INFO     Inverter 0: setting new limit from 0 Watt to 1500 Watt
2023-03-14 08:45:31 INFO     Inverter 1: setting new limit from 0 Watt to 300 Watt
2023-03-14 08:45:42 INFO     Inverter 0 reachable: True
2023-03-14 08:45:42 INFO     Inverter 1 reachable: True
2023-03-14 08:45:42 INFO     powermeter: 151  Watt
ApplyLimitsToSetpoint() missing 1 required positional argument: 'pSetpoint'
2023-03-14 08:45:55 INFO     Inverter 0 reachable: True
2023-03-14 08:45:55 INFO     Inverter 1 reachable: True
2023-03-14 08:45:55 INFO     powermeter: 156  Watt
ApplyLimitsToSetpoint() missing 1 required positional argument: 'pSetpoint'

reserve85, 2023-03-14:
ah, overloading does not work in python.
can you try renaming:

```
def ApplyLimitsToSetpoint(pInverter, pSetpoint): -> def ApplyLimitsToSetpointInverter(pInverter, pSetpoint):

NewLimit = ApplyLimitsToSetpoint(i, NewLimit) -> NewLimit = ApplyLimitsToSetpointInverter(i, NewLimit)
```

Ollipop030, 2023-03-14:
Looks ok:

2023-03-14 09:01:38 INFO     read config file: D:\Download\Dev\HoymilesZeroExport_Config.ini
2023-03-14 09:01:38 INFO     Inverter 0 reachable: True
2023-03-14 09:01:38 INFO     Inverter 1 reachable: True
2023-03-14 09:01:38 INFO     setting new limit to 1800 Watt
2023-03-14 09:01:38 INFO     Inverter 0: setting new limit from 0 Watt to 1500 Watt
2023-03-14 09:01:40 INFO     Inverter 1: setting new limit from 0 Watt to 300 Watt
2023-03-14 09:01:50 INFO     Inverter 0 reachable: True
2023-03-14 09:01:50 INFO     Inverter 1 reachable: True
2023-03-14 09:01:50 INFO     powermeter: 106  Watt
2023-03-14 09:01:50 INFO     setting new limit to 1800 Watt
2023-03-14 09:01:50 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 09:01:52 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 09:02:05 INFO     Inverter 0 reachable: True
2023-03-14 09:02:05 INFO     Inverter 1 reachable: True
2023-03-14 09:02:05 INFO     powermeter: 82  Watt
2023-03-14 09:02:05 INFO     setting new limit to 1800 Watt
2023-03-14 09:02:05 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 09:02:07 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 09:02:21 INFO     Inverter 0 reachable: True
2023-03-14 09:02:21 INFO     Inverter 1 reachable: True
2023-03-14 09:02:21 INFO     powermeter: 55  Watt
2023-03-14 09:02:21 INFO     setting new limit to 1800 Watt
2023-03-14 09:02:21 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 09:02:23 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt

I will let it run on my Windows machine and hope for the sun to come out to see if the inverters are limited when overproducing. But I don´t think it will get this much power today, typical weather in northern germany.

Ollipop030, 2023-03-14:
Thank you, I changed the lines and will report.

reserve85, 2023-03-14:
ok, did 
`GetHoymilesActualPower() `
work?

Ollipop030, 2023-03-14:
Yes.

Ollipop030, 2023-03-14:
It seems to work:

2023-03-14 11:47:01 INFO     Inverter 0 power producing: 302  Watt
2023-03-14 11:47:01 INFO     Inverter 1 power producing: 56  Watt
2023-03-14 11:47:01 INFO     overproducing: reduce limit based on actual power
2023-03-14 11:47:01 INFO     setting new limit to 358 Watt
2023-03-14 11:47:01 INFO     Inverter 0: setting new limit from 1500 Watt to 298 Watt
2023-03-14 11:47:03 INFO     Inverter 1: setting new limit from 300 Watt to 59 Watt
2023-03-14 11:47:08 INFO     Inverter 0 reachable: True
2023-03-14 11:47:08 INFO     Inverter 1 reachable: True
2023-03-14 11:47:08 INFO     powermeter: -109  Watt
2023-03-14 11:47:10 INFO     powermeter: -116  Watt
2023-03-14 11:47:11 INFO     powermeter: -116  Watt
2023-03-14 11:47:12 INFO     powermeter: -116  Watt
2023-03-14 11:47:13 INFO     powermeter: -116  Watt
2023-03-14 11:47:14 INFO     powermeter: -116  Watt
2023-03-14 11:47:15 INFO     powermeter: -116  Watt
2023-03-14 11:47:16 INFO     powermeter: -116  Watt
2023-03-14 11:47:17 INFO     powermeter: -116  Watt
2023-03-14 11:47:18 INFO     powermeter: -116  Watt
2023-03-14 11:47:19 INFO     powermeter: -116  Watt
2023-03-14 11:47:20 INFO     powermeter: -116  Watt
2023-03-14 11:47:21 INFO     powermeter: -116  Watt
2023-03-14 11:47:22 INFO     overproducing: reduce limit based on previous limit setpoint
2023-03-14 11:47:22 INFO     setting new limit to 317 Watt
2023-03-14 11:47:22 INFO     Inverter 0: setting new limit from 298 Watt to 264 Watt
2023-03-14 11:47:24 INFO     Inverter 1: setting new limit from 59 Watt to 52 Watt
2023-03-14 11:47:29 INFO     Inverter 0 reachable: True
2023-03-14 11:47:29 INFO     Inverter 1 reachable: True
2023-03-14 11:47:29 INFO     powermeter: -118  Watt
2023-03-14 11:47:30 INFO     powermeter: -118  Watt
2023-03-14 11:47:31 INFO     powermeter: -118  Watt
2023-03-14 11:47:32 INFO     powermeter: -118  Watt
2023-03-14 11:47:33 INFO     powermeter: -118  Watt
2023-03-14 11:47:34 INFO     powermeter: -118  Watt
2023-03-14 11:47:35 INFO     powermeter: -118  Watt
2023-03-14 11:47:36 INFO     powermeter: -118  Watt
2023-03-14 11:47:38 INFO     powermeter: -118  Watt
2023-03-14 11:47:39 INFO     powermeter: -75  Watt
2023-03-14 11:47:40 INFO     powermeter: -75  Watt
2023-03-14 11:47:41 INFO     powermeter: -75  Watt
2023-03-14 11:47:42 INFO     powermeter: -75  Watt

But something is strange:

setting new limit to 358 Watt
Inverter 0: setting new limit from 1500 Watt to 298 Watt
Inverter 1: setting new limit from 300 Watt to 59 Watt
298+59=357W

and here:

setting new limit to 317 Watt
Inverter 0: setting new limit from 298 Watt to 264 Watt
Inverter 1: setting new limit from 59 Watt to 52 Watt
264+52=316W

Just a little calculation error?

reserve85, 2023-03-14:
Yes, it´s a small calculation error because i need to round the decimal value to full integer value. But i think 1 Watt doesn´t really matter.

I´m very happy that it works out of the box. I will test it at home in the next days and if it works i´ll branch it into main. Please watch this over the next few days and let me know if any errors or problems occur.

I will update the dev version with all the bugfixes above included.

Thanks for your help!

Ollipop030, 2023-03-14:
I will keep an eye on that today. I can later post a log of a few hours of the running script, so you can look for yourself if there is any inconsistency.

reserve85, 2023-03-14:
That would be great. Thanks

Ollipop030, 2023-03-14:
Script is running for some time now, no errors. The weather is very bad today. Here is the log of the last 30 Minutes (keep in mind, my powermeter only updates every 10 seconds):

2023-03-14 16:26:05 INFO     setting new limit to 1800 Watt
2023-03-14 16:26:05 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:26:07 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:26:20 INFO     Inverter 0 reachable: True
2023-03-14 16:26:20 INFO     Inverter 1 reachable: True
2023-03-14 16:26:20 INFO     powermeter: 74  Watt
2023-03-14 16:26:20 INFO     setting new limit to 1800 Watt
2023-03-14 16:26:20 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:26:22 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:26:35 INFO     Inverter 0 reachable: True
2023-03-14 16:26:35 INFO     Inverter 1 reachable: True
2023-03-14 16:26:35 INFO     powermeter: 104  Watt
2023-03-14 16:26:35 INFO     setting new limit to 1800 Watt
2023-03-14 16:26:35 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:26:37 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:26:50 INFO     Inverter 0 reachable: True
2023-03-14 16:26:50 INFO     Inverter 1 reachable: True
2023-03-14 16:26:51 INFO     powermeter: 131  Watt
2023-03-14 16:26:51 INFO     setting new limit to 1800 Watt
2023-03-14 16:26:51 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:26:53 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:27:06 INFO     Inverter 0 reachable: True
2023-03-14 16:27:06 INFO     Inverter 1 reachable: True
2023-03-14 16:27:06 INFO     powermeter: 304  Watt
2023-03-14 16:27:06 INFO     setting new limit to 1800 Watt
2023-03-14 16:27:06 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:27:08 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:27:21 INFO     Inverter 0 reachable: True
2023-03-14 16:27:21 INFO     Inverter 1 reachable: True
2023-03-14 16:27:21 INFO     powermeter: 201  Watt
2023-03-14 16:27:21 INFO     setting new limit to 1800 Watt
2023-03-14 16:27:21 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:27:23 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:27:36 INFO     Inverter 0 reachable: True
2023-03-14 16:27:36 INFO     Inverter 1 reachable: True
2023-03-14 16:27:36 INFO     powermeter: 194  Watt
2023-03-14 16:27:36 INFO     setting new limit to 1800 Watt
2023-03-14 16:27:36 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:27:38 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:27:52 INFO     Inverter 0 reachable: True
2023-03-14 16:27:52 INFO     Inverter 1 reachable: True
2023-03-14 16:27:52 INFO     powermeter: 161  Watt
2023-03-14 16:27:52 INFO     setting new limit to 1800 Watt
2023-03-14 16:27:52 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:27:54 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:28:07 INFO     Inverter 0 reachable: True
2023-03-14 16:28:07 INFO     Inverter 1 reachable: True
2023-03-14 16:28:07 INFO     powermeter: 133  Watt
2023-03-14 16:28:07 INFO     setting new limit to 1800 Watt
2023-03-14 16:28:07 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:28:09 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:28:22 INFO     Inverter 0 reachable: True
2023-03-14 16:28:22 INFO     Inverter 1 reachable: True
2023-03-14 16:28:22 INFO     powermeter: 262  Watt
2023-03-14 16:28:22 INFO     setting new limit to 1800 Watt
2023-03-14 16:28:22 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:28:24 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:28:37 INFO     Inverter 0 reachable: True
2023-03-14 16:28:37 INFO     Inverter 1 reachable: True
2023-03-14 16:28:37 INFO     powermeter: 305  Watt
2023-03-14 16:28:37 INFO     setting new limit to 1800 Watt
2023-03-14 16:28:37 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 16:28:39 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 16:28:53 INFO     Inverter 0 reachable: True
2023-03-14 16:28:53 INFO     Inverter 1 reachable: True
2023-03-14 16:28:53 INFO     powermeter: 208  Watt
2023-03-14 16:28:53 INFO     setting new limit to 1800 Watt
2023-03-14 16:28:53 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
[... 887 lines cut ...]
2023-03-14 17:06:33 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 17:06:35 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 17:06:48 INFO     Inverter 0 reachable: True
2023-03-14 17:06:48 INFO     Inverter 1 reachable: True
2023-03-14 17:06:48 INFO     powermeter: 126  Watt
2023-03-14 17:06:48 INFO     setting new limit to 1800 Watt
2023-03-14 17:06:48 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 17:06:50 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 17:07:03 INFO     Inverter 0 reachable: True
2023-03-14 17:07:03 INFO     Inverter 1 reachable: True
2023-03-14 17:07:03 INFO     powermeter: 140  Watt
2023-03-14 17:07:03 INFO     setting new limit to 1800 Watt
[... cut ...]
2023-03-14 17:07:19 INFO     powermeter: 164  Watt
2023-03-14 17:07:19 INFO     setting new limit to 1800 Watt
2023-03-14 17:07:19 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-14 17:07:21 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-14 17:07:34 INFO     Inverter 0 reachable: True
2023-03-14 17:07:34 INFO     Inverter 1 reachable: True
2023-03-14 17:07:34 INFO     powermeter: 140  Watt

Maybe tomorrow is a better change for some "overproducing", I will report any strange behaviour.

reserve85, 2023-03-14:
ok thanks - you can add the next logfile via attach files :-)

reserve85, 2023-03-15:
perfect, thanks for your help! I´ll merge it into main.

closed 2023-03-15 by reserve85