reserve85/HoymilesZeroExport #12: Smartmeter mit OBIS-Code

tqma1, 2023-03-18:
Hallo,

tolles Projekt und schon mal ein kräftiges Danke hierzu!

Ist es geplant auch Leseköpfe zu unterstützen welche mit sogenannten OBIS-Codes arbeiten?

1.7.0 ist der aktuelle Bezug
2.7.0 ist die aktuelle Einspeisung.

Mein Zähler liefert z.Bsp über eine REST API folgende Daten:

{
"1.8.0":"123456",
"2.8.0":"123456",
"3.8.1":"123456",
"4.8.1":"123456",
**"1.7.0":"200",
"2.7.0":"300",**
"3.7.0":"123456",
"4.7.0":"123456",
"1.128.0":"123456",
"uptime":"0000:11:42:35"
}

Abfragelink:
http://IP_Adresse/getLastData?user=Benutzername&password=Passwort

Es gibt hier keine negativen Daten sondern man müsste beim Wert von 1.7.0 den Wert von 2.7.0 abziehen und so hätte man in meinem Beispiel eine aktuelle Einspeisung von 100 Watt.

Gruß, Thomas

reserve85, 2023-03-18:
Hi,

ja das wäre wahrscheinlich problemlos als weitere Variante möglich, ich müsste das nur implementieren. Das kann ich die nächsten Tage mal machen, am Wochenende siehts zeitlich aktuell schlecht aus.
Du müsstest das dann aber testen?

Kannst du mir mal Infos dazu geben welches Modul das ist? Wie oft aktualisiert der Wert?

Gruß Tobias

tqma1, 2023-03-18:
Das wäre super!
Testen würde ich selbstverständlich.

mein Modul ist von shrdzm, da in Österreich die Daten ja im Modul mit dem entsprechenden Key entschlüsselt werden müssen. Geht meines Wissens nach bei Tasmota (leider) noch nicht.

reserve85, 2023-03-18:
Ok, ich gebe dir bescheid.

Grüße nach Österreich,
Tobias

Ollipop030, 2023-03-18:
Hallo @tqma1 

habe auch keinen Tasmota Lesekopf, aber meiner gibt auch zwei Werte aus. Du kannst mit ein paar Änderungen das Ding ans Laufen bekommen:

HoymilesZeroExport_Config.ini:

[TASMOTA]
# --- defines for Tasmota ---
TASMOTA_IP = IP Adresse
# the following three constants describes how to navigate through the Tasmota-JSON
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
#TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
#TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label
TASMOTA_JSON_POWER_MQTT_LABEL = 1.7.0
TASMOTA_JSON_POWER_E_MQTT_LABEL = 2.7.0

HoymilesZeroExport.py:

def GetPowermeterWattsTasmota():
    url = f'http://{TASMOTA_IP}getLastData?user=Benutzername&password=Passwort'
    ParsedData = requests.get(url).json()
    Watts = int(ParsedData[TASMOTA_JSON_POWER_MQTT_LABEL] - ParsedData[TASMOTA_JSON_POWER_E_MQTT_LABEL])
    logging.info("powermeter: %s %s",Watts," Watt")
    return int(Watts)

und weiter unten im Script unter # read config:

#TASMOTA_JSON_STATUS = config.get('TASMOTA', 'TASMOTA_JSON_STATUS')
#TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = config.get('TASMOTA', 'TASMOTA_JSON_PAYLOAD_MQTT_PREFIX')
TASMOTA_JSON_POWER_MQTT_LABEL = config.get('TASMOTA', 'TASMOTA_JSON_POWER_MQTT_LABEL')
TASMOTA_JSON_POWER_E_MQTT_LABEL = config.get('TASMOTA', 'TASMOTA_JSON_POWER_E_MQTT_LABEL')

tqma1, 2023-03-18:
Hallo @Ollipop030

Herzlichen Dank für Deinen Input, so wie das aussieht läuft das schon mit den von Dir ausgearbeiteten Änderungen!

reserve85, 2023-03-18:
Perfekt, so weit hab ich ehrlich gesagt gar nicht gedacht. Danke @Ollipop030 ich würde es trotzdem noch als neue Variante mit implementieren, sonst ist die Namensgebung verwirrend und das Script muss immer bei jeder neuen Version mit angepasst werden. Ich würde gerne die Einstellungen komplett in die config verfrachten.

tqma1, 2023-03-19:
Preisfrage:
Habe gestern später noch auf python3.10 upgedated nun läuft das script nicht mehr.
Führe ich "sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f" aus
sehe das es Verweise auf python3.9/
z.Bsp: File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 565, in send
Muss ich es  nun neu installieren - oder?

reserve85, 2023-03-19:
Gute Frage, ich habe die 3.9 - kann schon sein dass mit 3.10 da was geändert wurde. Kannst du mal das LogOutput hier posten?

Ollipop030, 2023-03-19:
Auf dem Pi läuft es bei mir auf 3.7.3. Es lief bei mir allerdings testweise aber auf einer Windows Maschine mit python 3.10. Ich denke beim Update bei dir ist was schiefgelaufen. Evtl. fehlen irgendwelche Abhängigkeiten?

@reserve85 zu der geplanten Version hätte ich ein paar Anmerkungen:

1. Wäre es in der *.ini nicht besser die ganze Domäne mit den json Daten anzugeben statt nur der IP? Hilft dann auch Leuten mit einem Powerfox Lesekopf. Wobei es dann ja auch nicht mehr Tasmota heißen müsste.

2. In dem Beispiel habe ich ja die Variable TASMOTA_JSON_POWER_E_MQTT_LABEL = 2.7.0 hinzugefügt. Hier wäre es doch schlau, wenn man hier seinen Wert eintragen kann, und wenn der Lesekopf dafür keinen Wert ausgibt (wie bei Tasmota) den Wert "none" einträgt. Das Script sollte dann nur unterscheiden, 

wenn Wert "none" dann "Watts = int(ParsedData[TASMOTA_JSON_POWER_MQTT_LABEL])", 

sonst 
"Watts = int(ParsedData[TASMOTA_JSON_POWER_MQTT_LABEL] - ParsedData[TASMOTA_JSON_POWER_E_MQTT_LABEL])"

tqma1, 2023-03-19:
Habe die Installation nochmals gemacht, jetzt läuft es wieder - Danke

@reserve85 

Hatte da gestern Abend eine Idee …
Wäre es möglich eine zweite Instanz für die Einspeisung in der Nacht laufen zu lassen?
Ich denke hierfür müsste man das zwingend über eine zweite DTU laufen lassen?
Oder wäre hier die Aktivierung/Deaktivierung über Sonnen- Auf bzw. Untergang denkbar?

Schöne Grüße!

reserve85, 2023-03-20:
@Ollipop030 : danke für deinen Input, ich weiss auch nicht so recht wie man das möglichst "Variabel" gestalten kann. Problem ist halt, dass du es eigentlich immer irgendwie anpassen musst. Am einfachsten ist da wahrscheinlich noch die "Request" URL, da es ja meist nur eine Zeile ist. Allerdings kommt bei manchen dann auch noch HTTP Authentifizierung mit dazu. 
Problematischer sind die JSON Results, da die immer komplett unterschiedlich aufgebaut sind. Da musst du dann teilweise über mehrere Arrays springen und suchen. Keine Ahnung wie ich das variabel implementieren könnte. 
Meine Idee war daher, dass man die zwei Funktionen dann halt für unterschiedliche Geräte erweitert. Keine Ahnung ob das der richtige Weg ist. Ich bin auch kein Python Profi, programmiere berufsbedingt normalerweise Delphi...

@tqma1 : Was verstehst du unter Nachteinspeisung? Ich Frage die DTU´s ab ob die Inverter erreichbar sind, da ist das Sonnenaufgangsthema eigentlich dann schon mit abgedeckt.

Ollipop030, 2023-03-20:
Ich verstehe die Idee von @tqma1 so:

Man trägt die Maximalleistung der WR ein (zB 600W), und dieser Wert wird dann nach Sonnenuntergang geändert auf 100W. Und dementsprechend morgens wieder hoch auch 600W. Dazu müsste man aber die Zeiten aus der DTU abfragen.

Ollipop030, 2023-03-20:
Ich schätze mal, dass er einen Akku an einem separaten Wechselrichter hat. Der WR wird tagsüber begrenzt, damit der Akku geladen werden kann und nicht einspeist, und erst Abends soll der WR dann einspeisen um die Grundlast zu decken.

curtosvienna, 2023-03-20:
Denke auch an Akku Nulleinspeisung

reserve85, 2023-03-20:
Ahso, also wäre das dann ein zweiter WR der eben nachts "produziert" während der andere mit Panels ausgeschaltet ist? 
Ich glaube am einfachsten wäre es da zwei Scripte und zwei DTUs zu haben, getrennt jeweils eins für Tagbetrieb und eins für Nachtbetrieb. 
Das könnte man dann abends z.B. über ein Shellscript switchen ("stop Service Tagbetrieb -> start Service Nachtbetrieb") und hätte dadurch kein Programmieraufwand im Script selbst. Gibt ja dafür diverse API´s (z.B. auf die schnelle https://domes-finest.de/2014/09/17/linux-shell-script-sonnenaufgang-sonnenuntergang-bestimmen/)

Ollipop030, 2023-03-20:
So würde ich es auch machen, einfach zwei DTUs und zwei Dienste, wobei man die Dienst Tagesbetrieb gar nicht beenden müsste, da ja vom Dach eh nichts mehr kommt. Könnte man mit cron machen, wird dann aber komplizierter mit der Berechnung für Sonnenauf- und Untergang. Das über dieses Script zu lösen ist aber wohl der falsche Weg.

reserve85, 2023-03-20:
ich noch mal: @tqma1 : ich habe das SHRDZM Modul eingebunden, wäre super wenn du das mal laufen lassen könntest. Ist im DEV-Zweig (https://github.com/reserve85/HoymilesZeroExport/tree/dev_shrdzm)

@Ollipop030 : Die Option für 1.7.0 - 2.7.0 bei Tasmota habe ich auch mit reingenommen. Kannst dir es ja mal anschauen. Ich habe es mit einem boolean-flag umgesetzt ob es berechnet werden muss oder nicht.

tqma1, 2023-03-20:
Ui, schnell mal etwas Arbeiten gegangen und ich komme mit dem lesen fast nicht mehr hinterher…

Richtig, Nachteinspeisung kommt von einem separates Wechselrichter am Akku, daher die Idee mit der zweiten Instanz.
Werde morgen mal die Testversion installieren und hoffen das die Sonne ordentlich strahlt :-)

Ollipop030, 2023-03-21:
Du brauchst die nur die *.ini und *.py aus dem dev Branch. Eben wieder anpassen, so wie du sie brauchst und starten.

reserve85, 2023-03-21:
kannst auch rechts oben auf "Code" und "Download ZIP" klicken.

tqma1, 2023-03-21:
So installiert und getestet - regelt leider nur nicht auf die eingestellte Einspeisung von 75 Watt

reserve85, 2023-03-21:
Weil Wetter wie in Frankfurt? Oder geht was nicht?

tqma1, 2023-03-21:
Sonnenschein pur!

Output:

Mär 21 15:20:01 Debian-Server python3[1118]: 2023-03-21 15:20:01 INFO     Author: reserve85 / Script Version: 1.9
Mär 21 15:20:01 Debian-Server python3[1118]: 2023-03-21 15:20:01 INFO     read config file: /home/user/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mär 21 15:20:01 Debian-Server python3[1118]: 2023-03-21 15:20:01 INFO     Inverter 0 reachable: True
Mär 21 15:20:01 Debian-Server python3[1118]: 2023-03-21 15:20:01 INFO     setting new limit to 800 Watt
Mär 21 15:20:01 Debian-Server python3[1118]: 2023-03-21 15:20:01 INFO     Inverter 0: setting new limit from 0 Watt to 800 Watt
Mär 21 15:20:11 Debian-Server python3[1118]: 2023-03-21 15:20:11 INFO     Inverter 0 reachable: True
Mär 21 15:20:31 Debian-Server python3[1118]: 2023-03-21 15:20:31 INFO     Inverter 0 reachable: True
Mär 21 15:20:51 Debian-Server python3[1118]: 2023-03-21 15:20:51 INFO     Inverter 0 reachable: True
Mär 21 15:21:11 Debian-Server python3[1118]: 2023-03-21 15:21:11 INFO     Inverter 0 reachable: True
Mär 21 15:21:31 Debian-Server python3[1118]: 2023-03-21 15:21:31 INFO     Inverter 0 reachable: True
Mär 21 15:21:52 Debian-Server python3[1118]: 2023-03-21 15:21:52 INFO     Inverter 0 reachable: True
Mär 21 15:22:12 Debian-Server python3[1118]: 2023-03-21 15:22:12 INFO     Inverter 0 reachable: True

tqma1, 2023-03-21:
Mär 21 15:52:21 Debian-Server systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mär 21 15:52:21 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mär 21 15:52:21 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Mär 21 15:52:21 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Mär 21 15:52:21 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Mär 21 15:52:21 Debian-Server python3[616]: 2023-03-21 15:52:21 INFO     Author: reserve85 / Script Version: 1.9
Mär 21 15:52:21 Debian-Server python3[616]: 2023-03-21 15:52:21 INFO     read config file: /home/user/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mär 21 15:52:21 Debian-Server python3[616]: Traceback (most recent call last):
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 169, in _new_conn
Mär 21 15:52:21 Debian-Server python3[616]:     conn = connection.create_connection(
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 96, in create_connection
Mär 21 15:52:21 Debian-Server python3[616]:     raise err
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 86, in create_connection
Mär 21 15:52:21 Debian-Server python3[616]:     sock.connect(sa)
Mär 21 15:52:21 Debian-Server python3[616]: OSError: [Errno 101] Network is unreachable
Mär 21 15:52:21 Debian-Server python3[616]: During handling of the above exception, another exception occurred:
Mär 21 15:52:21 Debian-Server python3[616]: Traceback (most recent call last):
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 699, in urlopen
Mär 21 15:52:21 Debian-Server python3[616]:     httplib_response = self._make_request(
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 394, in _make_request
Mär 21 15:52:21 Debian-Server python3[616]:     conn.request(method, url, **httplib_request_kw)
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 234, in request
Mär 21 15:52:21 Debian-Server python3[616]:     super(HTTPConnection, self).request(method, url, body=body, headers=headers)
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3.9/http/client.py", line 1255, in request
Mär 21 15:52:21 Debian-Server python3[616]:     self._send_request(method, url, body, headers, encode_chunked)
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3.9/http/client.py", line 1301, in _send_request
Mär 21 15:52:21 Debian-Server python3[616]:     self.endheaders(body, encode_chunked=encode_chunked)
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3.9/http/client.py", line 1250, in endheaders
Mär 21 15:52:21 Debian-Server python3[616]:     self._send_output(message_body, encode_chunked=encode_chunked)
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3.9/http/client.py", line 1010, in _send_output
Mär 21 15:52:21 Debian-Server python3[616]:     self.send(msg)
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3.9/http/client.py", line 950, in send
Mär 21 15:52:21 Debian-Server python3[616]:     self.connect()
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 200, in connect
Mär 21 15:52:21 Debian-Server python3[616]:     conn = self._new_conn()
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in _new_conn
Mär 21 15:52:21 Debian-Server python3[616]:     raise NewConnectionError(
Mär 21 15:52:21 Debian-Server python3[616]: urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0x7f3bdffb05e0>: Failed to establish a new connection: [Errno 101] Network is unreachable
Mär 21 15:52:21 Debian-Server python3[616]: During handling of the above exception, another exception occurred:
Mär 21 15:52:21 Debian-Server python3[616]: Traceback (most recent call last):
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/local/lib/python3.9/dist-packages/requests/adapters.py", line 489, in send
Mär 21 15:52:21 Debian-Server python3[616]:     resp = conn.urlopen(
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 755, in urlopen
Mär 21 15:52:21 Debian-Server python3[616]:     retries = retries.increment(
Mär 21 15:52:21 Debian-Server python3[616]:   File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 574, in increment
Mär 21 15:52:21 Debian-Server python3[616]:     raise MaxRetryError(_pool, url, error or ResponseError(cause))
Mär 21 15:52:21 Debian-Server python3[616]: urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.1.4', port=80): Max retries exceeded with url: /api/index (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f3bdffb05e0>: Failed to establish a new connection: [Errno 101] Network is unreachable'))
Mär 21 15:52:21 Debian-Server python3[616]: During handling of the above exception, another exception occurred:
Mär 21 15:52:21 Debian-Server python3[616]: Traceback (most recent call last):
[... cut ...]
Mär 21 15:52:21 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mär 21 15:52:22 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Mär 21 15:52:22 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Mär 21 15:52:22 Debian-Server systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Mär 21 15:52:22 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mär 21 15:52:22 Debian-Server systemd[1]: Failed to start HoymilesZeroExport Service.

tqma1, 2023-03-21:
warum gibt er immer noch python3.9/ pfade aus ?
Habe ja auf  python3.10 upgedated - liegt hier der Fehler?

reserve85, 2023-03-21:
Und die IP ist korrekt?
HTTPConnectionPool(host='192.168.1.4', port=80): Max retries exceeded with url: /api/index (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f3bdffb05e0>: Failed to establish a new connection: [Errno 101] Network is unreachable'))

Für mich sieht das so aus als kann er Ahoy nicht abfragen.

tqma1, 2023-03-21:
IP ist definitiv korrekt

reserve85, 2023-03-21:
Oder kannst du die AHOY IP mal von dem Debian Server pingen ob der erreichbar ist? Ist definitiv ein Fehler bei GetHoymilesAvailable... Kann ich auch nachstellen wenn ich eine falsche IP eintrage oder eine die es nicht gibt.

tqma1, 2023-03-21:
Anpingen ist absolut stressfrei: ca. 2,5ms im Schnitt

tqma1, 2023-03-21:
Aber da sollte doch auch nie die Meldung kommen:
Inverter reachable true

reserve85, 2023-03-22:
du könntest auch das hier noch zum Testen in den Code einfügen, evtl hilft es:

direkt nach der Zeile 
`SLOW_APPROX_LIMIT = int(GetMaxWattFromAllInverters() * config.getint('COMMON', 'SLOW_APPROX_LIMIT_IN_PERCENT') / 100)
`
das hier einfügen und das script dann mal starten
```
try:
    # Test
    logging.info("---Startup-Test---")
    logging.info("GetHoymilesAvailable")
    GetHoymilesAvailable()
    logging.info("GetPowermeterWatts")
    GetPowermeterWatts()
    logging.info("GetHoymilesActualPower")
    GetHoymilesActualPower()
    logging.info("SetLimit")
    SetLimit(GetMaxWattFromAllInverters())
    logging.info("---Startup-Test complete---")
except Exception as e:
    if hasattr(e, 'message'):
        print(e.message)
    else:
        print(e)
```

tqma1, 2023-03-22:
Habe den Teil eingefügt und das Script manuell gestartet:

Mär 22 16:19:05 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     Author: reserve85 / Script Version: 1.9
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     read config file: /home/user/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     ---Startup-Test---
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     GetHoymilesAvailable
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     Inverter 0 reachable: True
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     GetPowermeterWatts
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     Inverter 0 reachable: True
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     setting new limit to 800 Watt
Mär 22 16:19:05 Debian-Server python3[805]: 2023-03-22 16:19:05 INFO     Inverter 0: setting new limit from 0 Watt to 800 Watt
Mär 22 16:19:15 Debian-Server python3[805]: 2023-03-22 16:19:15 INFO     Inverter 0 reachable: True
Mär 22 16:19:36 Debian-Server python3[805]: 2023-03-22 16:19:36 INFO     Inverter 0 reachable: True

Starte ich allerdings den Server neu und er startet selbst das Script erhalte ich nach Eingabe von:
sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f
eine Unmenge von Fehlermeldungen ....

reserve85, 2023-03-22:
OK, ein Schritt weiter: vom Powermeter kommt nix an. Bitte die Funktion mal ersetzen:

`def GetPowermeterWattsShrdzm():
    logging.info("GetPowermeterWattsShrdzm")
    logging.info(f'http://{SHRDZM_IP}/getLastData?user={SHRDZM_USER}&password={SHRDZM_PASS}')
    url = f'http://{SHRDZM_IP}/getLastData?user={SHRDZM_USER}&password={SHRDZM_PASS}'
    ParsedData = requests.get(url).json()
    logging.info(ParsedData)
    logging.info(ParsedData['1.7.0'])
    logging.info(ParsedData['2.7.0'])
    Watts = int(ParsedData['1.7.0'] - ParsedData['2.7.0'])
    logging.info("powermeter: %s %s",Watts," Watt")
    return int(Watts)`

tqma1, 2023-03-22:
Mär 22 16:58:35 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     Author: reserve85 / Script Version: 1.9
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     read config file: /home/user/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     ---Startup-Test---
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     GetHoymilesAvailable
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     Inverter 0 reachable: True
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     GetPowermeterWatts
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     GetPowermeterWattsShrdzm
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     http://192.168.1.8/getLastData?user=USER&password=PASSWORT
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     {'1.8.0': '18334860', '2.8.0': '6849', '3.8.1': '60184', '4.8.1': '4749016', '1.7.0': '357', '2.7.0': '0', '3.7.0': '0', '4.7.0': '97', '1.128.0': '0', 'uptime': '0000:13:44:56'}
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     357
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     0
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     Inverter 0 reachable: True
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     setting new limit to 800 Watt
Mär 22 16:58:35 Debian-Server python3[870]: 2023-03-22 16:58:35 INFO     Inverter 0: setting new limit from 0 Watt to 800 Watt
Mär 22 16:58:45 Debian-Server python3[870]: 2023-03-22 16:58:45 INFO     Inverter 0 reachable: True
Mär 22 16:58:45 Debian-Server python3[870]: 2023-03-22 16:58:45 INFO     GetPowermeterWattsShrdzm
Mär 22 16:58:45 Debian-Server python3[870]: 2023-03-22 16:58:45 INFO     http://192.168.1.8/getLastData?user=USER&password=PASSWORT
Mär 22 16:58:46 Debian-Server python3[870]: 2023-03-22 16:58:46 INFO     {'1.8.0': '18334862', '2.8.0': '6849', '3.8.1': '60184', '4.8.1': '4749017', '1.7.0': '364', '2.7.0': '0', '3.7.0': '0', '4.7.0': '96', '1.128.0': '0', 'uptime': '0000:13:45:09'}

reserve85, 2023-03-22:
diese Zeile ersetzen:
`Watts = int(int(ParsedData['1.7.0']) - int(ParsedData['2.7.0']))`

tqma1, 2023-03-22:
sieht jetzt ganz gut aus:

Mär 22 17:14:27 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     Author: reserve85 / Script Version: 1.9
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     read config file: /home/user/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     ---Startup-Test---
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     GetHoymilesAvailable
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     Inverter 0 reachable: True
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     GetPowermeterWatts
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     GetPowermeterWattsShrdzm
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     http://192.168.1.8/getLastData?user=USER&password=PASSWORT
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     {'1.8.0': '18334984', '2.8.0': '6849', '3.8.1': '60184', '4.8.1': '4749042', '1.7.0': '328', '2.7.0': '0', '3.7.0': '0', '4.7.0': '97', '1.128.0': '0', 'uptime': '0000:14:00:51'}
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     328
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     0
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     powermeter: 328  Watt
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     GetHoymilesActualPower
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     Inverter 0 power producing: 29  Watt
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     SetLimit
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     setting new limit to 800 Watt
Mär 22 17:14:28 Debian-Server python3[1049]: 2023-03-22 17:14:28 INFO     Inverter 0: setting new limit from 0 Watt to 800 Watt
Mär 22 17:14:33 Debian-Server python3[1049]: 2023-03-22 17:14:33 INFO     ---Startup-Test complete---
Mär 22 17:14:33 Debian-Server python3[1049]: 2023-03-22 17:14:33 INFO     Inverter 0 reachable: True
Mär 22 17:14:33 Debian-Server python3[1049]: 2023-03-22 17:14:33 INFO     setting new limit to 800 Watt
Mär 22 17:14:33 Debian-Server python3[1049]: 2023-03-22 17:14:33 INFO     Inverter 0: setting new limit from 800 Watt to 800 Watt
Mär 22 17:14:43 Debian-Server python3[1049]: 2023-03-22 17:14:43 INFO     Inverter 0 reachable: True
Mär 22 17:14:43 Debian-Server python3[1049]: 2023-03-22 17:14:43 INFO     GetPowermeterWattsShrdzm
Mär 22 17:14:43 Debian-Server python3[1049]: 2023-03-22 17:14:43 INFO     http://192.168.1.8/getLastData?user=USER&password=PASSWORT
Mär 22 17:14:44 Debian-Server python3[1049]: 2023-03-22 17:14:44 INFO     {'1.8.0': '18334985', '2.8.0': '6849', '3.8.1': '60184', '4.8.1': '4749043', '1.7.0': '588', '2.7.0': '0', '3.7.0': '0', '4.7.0': '97', '1.128.0': '0', 'uptime': '0000:14:01:07'}
Mär 22 17:14:44 Debian-Server python3[1049]: 2023-03-22 17:14:44 INFO     588
Mär 22 17:14:44 Debian-Server python3[1049]: 2023-03-22 17:14:44 INFO     0
Mär 22 17:14:44 Debian-Server python3[1049]: 2023-03-22 17:14:44 INFO     powermeter: 588  Watt
Mär 22 17:14:44 Debian-Server python3[1049]: 2023-03-22 17:14:44 INFO     setting new limit to 800 Watt
Mär 22 17:14:44 Debian-Server python3[1049]: 2023-03-22 17:14:44 INFO     Inverter 0: setting new limit from 800 Watt to 800 Watt

reserve85, 2023-03-22:
perfekt. SORRY! ich konnte das schlecht testen. Ich übernehme den Bugfix in der nächsten Version mit in den Quellcode. Du kannst im Prinzip vorerst das Originalfile nehmen und nur die eine Zeile austauschen.

tqma1, 2023-03-22:
Nichts zu entschuldigen, war ja vereinbart das ich das ganze teste!
Danke nochmals für die prompte Integration und Unterstützung.

Hättest Du noch einen Hinweis warum das nach einem restart vom Server nicht läuft?

Erst nachdem ich:
sudo chmod +x restart.sh
./restart.sh
gemacht habe läuft das Script wie erwartet.

closed 2023-03-23 by reserve85