reserve85/HoymilesZeroExport #14: Feature Request: Changelog, Log als Datei ausgeben

Ollipop030, 2023-03-21:
Hi,

ich mach der Übersichtlichkeit halber mal ein neues Issue auf. Wäre es möglich eine Changelog Datei anzulegen? Ich speichere für den Fehlerfall immer die alten Versionen in ein Verzeichnis, weiß aber gar nicht welche Änderungen es da immer gab.

Des Weiteren:
Gibt es eine Möglichkeit eine Logdatei zu schreiben beim Starten des Scripts? Wenn das auf Linux als Dienst läuft geht das ja ohne Weiteres, aber testweise starte ich neue Versionen einfach so über die Kommandozeile, da wird dann nichts geloggt. Reicht ja, wenn einfach ein Textfile erzeugt wird pro Tag.

reserve85, 2023-03-21:
Changelog kann ich gerne machen. Bei den Logs bin ich überfragt, ich recherchiere mal wenn ich Zeit habe

reserve85, 2023-03-23:
Logfiles in Datei speichern ist nun implementiert, Changelog ebenso

closed 2023-03-23 by reserve85