reserve85/HoymilesZeroExport #22: Unnötiges Setzen der maximalen Leistung bei zu wenig Erzeugung

Ollipop030, 2023-03-28:
Hi,

hatte da noch einen Einfall, keine Ahnung ob das eine gute Idee oder Quatsch ist.

Wenn zu wenig erzeugt wird, wird immer wieder aufs neue das Limit 100% an alle Inverter geschickt. Das ist gerade morgens, kurz nach Sonnenaufgang und Abends unnötig. Bei mir ist zB die Grundlast 120W, darunter komme ich nie. Wenn jetzt nicht mehr als 200W (Einspeisewert + Toleranz) erzeugt werden, wird immer aufs neue das Limit auf 100% gesetzt. Allein heute morgen fing die PV Anlage an zu produzieren, das Limit wurde aber erstmalig um kurz vor 9 reduziert.

Jetzt hattest du in einem anderen Issue mal erwähnt, das ab und zu das Limit nicht angenommen wird. Könnte man, wenn der Fall eintritt, nicht einen Counter setzen, das nur 10x hintereinander 100% Limit gesetzt werden? Und bei jedem Durchlauf prüft das Script was das Powermeter meldet, und ob das Limit schon bei 100% ist, und ob der Counter schon bei 10 steht. Dann weiter bis zum nächsten Durchlauf ohne neues Setzen des Limits.

Falls du keinen Handlungsbedarf siehst mach hier ruhig wieder zu, vielleicht hast du ja auch eine bessere Idee.

reserve85, 2023-03-28:
https://github.com/reserve85/HoymilesZeroExport/tree/dev_LimitRetry

Ollipop030, 2023-03-28:
Wow, das ging ja mal schnell. Wird gerade getestet, sieht aber gut aus:

2023-03-28 13:16:25 INFO     powermeter: -9  Watt
2023-03-28 13:16:26 INFO     powermeter: -9  Watt
2023-03-28 13:16:27 INFO     powermeter: -9  Watt
2023-03-28 13:16:28 INFO     powermeter: -9  Watt
2023-03-28 13:16:29 INFO     Not enough energy producing: limit already at maximum
2023-03-28 13:16:29 INFO     Inverter 0 reachable: True
2023-03-28 13:16:30 INFO     Inverter 1 reachable: True
2023-03-28 13:16:30 INFO     powermeter: 3  Watt
2023-03-28 13:16:30 INFO     setting new limit to 1800 Watt
2023-03-28 13:16:30 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:16:31 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:16:44 INFO     Inverter 0 reachable: True
2023-03-28 13:16:44 INFO     Inverter 1 reachable: True
2023-03-28 13:16:44 INFO     powermeter: 3  Watt
2023-03-28 13:16:44 INFO     setting new limit to 1800 Watt
2023-03-28 13:16:44 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:16:45 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:16:58 INFO     Inverter 0 reachable: True
2023-03-28 13:16:58 INFO     Inverter 1 reachable: True
2023-03-28 13:16:58 INFO     powermeter: 9  Watt
2023-03-28 13:16:58 INFO     setting new limit to 1800 Watt
2023-03-28 13:16:58 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:16:59 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:17:12 INFO     Inverter 0 reachable: True
2023-03-28 13:17:12 INFO     Inverter 1 reachable: True
2023-03-28 13:17:12 INFO     powermeter: 1  Watt
2023-03-28 13:17:12 INFO     setting new limit to 1800 Watt
2023-03-28 13:17:12 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:17:13 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:17:27 INFO     Inverter 0 reachable: True
2023-03-28 13:17:27 INFO     Inverter 1 reachable: True
2023-03-28 13:17:27 INFO     powermeter: 6  Watt
2023-03-28 13:17:27 INFO     setting new limit to 1800 Watt
2023-03-28 13:17:27 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:17:28 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:17:41 INFO     Inverter 0 reachable: True
2023-03-28 13:17:41 INFO     Inverter 1 reachable: True
2023-03-28 13:17:41 INFO     powermeter: 10  Watt
2023-03-28 13:17:41 INFO     setting new limit to 1800 Watt
2023-03-28 13:17:41 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:17:42 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:17:55 INFO     Inverter 0 reachable: True
2023-03-28 13:17:55 INFO     Inverter 1 reachable: True
2023-03-28 13:17:55 INFO     powermeter: 28  Watt
2023-03-28 13:17:55 INFO     setting new limit to 1800 Watt
2023-03-28 13:17:55 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:17:56 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:18:09 INFO     Inverter 0 reachable: True
2023-03-28 13:18:09 INFO     Inverter 1 reachable: True
2023-03-28 13:18:09 INFO     powermeter: 31  Watt
2023-03-28 13:18:09 INFO     setting new limit to 1800 Watt
2023-03-28 13:18:09 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:18:10 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:18:24 INFO     Inverter 0 reachable: True
2023-03-28 13:18:24 INFO     Inverter 1 reachable: True
2023-03-28 13:18:24 INFO     powermeter: 31  Watt
2023-03-28 13:18:24 INFO     setting new limit to 1800 Watt
2023-03-28 13:18:24 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
2023-03-28 13:18:25 INFO     Inverter 1: setting new limit from 300 Watt to 300 Watt
2023-03-28 13:18:38 INFO     Inverter 0 reachable: True
2023-03-28 13:18:38 INFO     Inverter 1 reachable: True
2023-03-28 13:18:38 INFO     powermeter: 25  Watt
2023-03-28 13:18:38 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1800 Watt
2023-03-28 13:18:49 INFO     Inverter 0 reachable: True
2023-03-28 13:18:49 INFO     Inverter 1 reachable: True
2023-03-28 13:18:49 INFO     powermeter: 32  Watt
2023-03-28 13:18:49 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1800 Watt
2023-03-28 13:19:00 INFO     Inverter 0 reachable: True
2023-03-28 13:19:00 INFO     Inverter 1 reachable: True
2023-03-28 13:19:00 INFO     powermeter: 37  Watt
2023-03-28 13:19:00 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1800 Watt
2023-03-28 13:19:11 INFO     Inverter 0 reachable: True
2023-03-28 13:19:11 INFO     Inverter 1 reachable: True
2023-03-28 13:19:11 INFO     powermeter: 35  Watt
2023-03-28 13:19:11 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1800 Watt

Genau bei 10 mal wird das Leistungslimit nicht erneut gesendet. Perfekt.

Ollipop030, 2023-03-28:
Habe seit ein paar Tagen schon LOOP_INTERVAL_IN_SECONDS = 11 und SET_LIMIT_DELAY_IN_SECONDS = 1. Mein Zähler spuckt leider nur alle 10 Sekunde neue Werte aus, den Loop noch weiter zu reduzieren macht bei mir wohl keinen Sinn. Aber alles stabil und ohne Aussetzer. 
Habe Ahoy immer auf der aktuellstren Dev Version gehalten, heute morgen dann die Final 0.60 geflasht.

Tolle Arbeit von dir und danke nochmals.

reserve85, 2023-03-28:
Ja hab ich auch immer aktuell gehalten aber hab die Settings nie angepasst. Meinst es macht sinn das mit in die Defaults zu übernehmen? Oder besser etwas konservativer?

Ollipop030, 2023-03-28:
Schwierig. Ich habe nur noch ESP32 und keinen 8266, ich weiß nicht ob die so häufige Änderungen, gerade mit mehreren Invertern vertragen. Die ESP32 laufen viel stabiler, die 8266 (habe damals 4 Stck getestet) habe nicht mal 24h ohne Reboot durchgehalten.

Vorsichtshalber lieber die Defaults so lassen, die meisten werden wohl 8266 einsetzen.

closed 2023-03-29 by reserve85

D3R-ST3FAN, 2023-04-14:
ist das Problem wirklich behoben?
Im Moment erzeuge ich ~90W (Wetter sehr bescheiden) aber die WR sollen ständig ihr Limit neu setzen.
Das kommt auch so in der DTU (OpenDTU, aktuellster Build  an).

Ich sehe in der DTU ständig:

`TX ActivePowerControl`  Einträge

Wird das aktuelle Limit nicht gespeichert?

Warum kommt
`OpenDTU: Inverter "03-ZaunPV400": setting new limit from 400 Watt to 400 Watt`

`INFO     powermeter IOBROKER: 2423  Watt
INFO     setting new limit to 1500 Watt
INFO     OpenDTU: Inverter "02-HM1500-02-Unten": setting new limit from 1500 Watt to 1500 Watt`

Das scheint die DTU auch zu stressen.

reserve85, 2023-04-14:
Du kannst die Anzahl der "wiederholenden Limit-Befehle" einstellen: SET_LIMIT_RETRY.

D3R-ST3FAN, 2023-04-14:
Ahhh Danke Dir