reserve85/HoymilesZeroExport #24: Exception at GetPowermeterWatts - Ich finde den Fehler nicht

famlac9805, 2023-03-29:
Nach ersten Versuchen als Teilzeit iOBroker und Github Nutzer konnte ich den HoymilesZeroExport zum laufen bringen. Die Verbindung zu Ahoy scheint OK zu sein. Es bleibt jedoch ein Error übrig. Ich finds einfach nicht raus. Der Fehler kommt im Log permanent.

Mar 29 21:56:51 phoscon systemd[1]: Stopping HoymilesZeroExport Service...
Mar 29 21:56:51 phoscon systemd[1]: HoymilesZeroExport.service: Main process exited, code=killed, status=15/TERM
Mar 29 21:56:51 phoscon systemd[1]: HoymilesZeroExport.service: Succeeded.
Mar 29 21:56:51 phoscon systemd[1]: Stopped HoymilesZeroExport Service.
Mar 29 21:56:51 phoscon systemd[1]: Started HoymilesZeroExport Service.
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     Log write to file: False
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     Author: reserve85 / Script Version: 1.14
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     Config file V 1.14
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     Inverter 0 reachable: True
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     setting new limit to 1500 Watt
Mar 29 21:56:51 phoscon python3[21579]: 2023-03-29 21:56:51 INFO     Inverter 0: setting new limit from 0 Watt to 1500 Watt
Mar 29 21:57:02 phoscon python3[21579]: 2023-03-29 21:57:02 INFO     Inverter 0 reachable: True
Mar 29 21:57:02 phoscon python3[21579]: 2023-03-29 21:57:02 ERROR    Exception at GetPowermeterWatts
Mar 29 21:57:02 phoscon python3[21579]: 2023-03-29 21:57:02 ERROR    '1.7.0'
Mar 29 21:57:22 phoscon python3[21579]: 2023-03-29 21:57:22 INFO     Inverter 0 reachable: True
Mar 29 21:57:22 phoscon python3[21579]: 2023-03-29 21:57:22 ERROR    Exception at GetPowermeterWatts
Mar 29 21:57:22 phoscon python3[21579]: 2023-03-29 21:57:22 ERROR    '1.7.0'
Mar 29 21:57:42 phoscon python3[21579]: 2023-03-29 21:57:42 INFO     Inverter 0 reachable: True
Mar 29 21:57:43 phoscon python3[21579]: 2023-03-29 21:57:43 ERROR    Exception at GetPowermeterWatts

Vielleicht gibts da Abhilfe. Ich kann leider auch noch nicht sagen ob der ZeroExport funktioniert. Habe ich erst heute nach Sonnenuntergang eingerichtet.

Danke schon mal für die Hilfe

reserve85, 2023-03-29:
Da stimmt was nicht mit Tasmota. Eventuell heißen die Labels bei dir anders. Kannst du mal das JSON von Tasmota hier posten ( http://{TASMOTA_IP}/cm?cmnd=status%2010 )und deine Config Settings?

famlac9805, 2023-03-29:
Ich verwende Tasmota nicht. Habe lediglich den SHRDZM und die AhoyDTU. Ich hab schon zweimal kontrolliert ob ich alles richtig abgeändert habe im Skript finde aber keinen Fehler. 

Der im Error beschriebene Wert 1.7.0 ist bei meinem SmartMeter die Leistung die aktuell bezogen wird.

reserve85, 2023-03-29:
Ok ich schaue morgen nochmal rein und melde mich. 

Kannst du noch das jSON von SHRDZM posten?
http://SHRDZM_IP/getLastData?user=SHRDZM_USER&password=SHRDZM_PASS

famlac9805, 2023-03-29:
Dein Link hat das Problem gelöst.
Ich bin mit dem Link und meinen Login Daten vom SHRDZM nicht zum JSON vom SHRDZM gekommen. Über die Oberfläche von SHRDZM kann ich jedoch auf die Rest Adresse sehen. Und da ist ein User und Passwort vergeben das ich bis dato nicht kannte... Die im Skript geändert und siehe da. Kein Fehler mehr.

Somit alles erledigt. Freue mich schon morgen nicht einzuspeisen :) 

DANKE!!!!

Mar 29 22:47:38 phoscon systemd[1]: HoymilesZeroExport.service: Main process exited, code=killed, status=15/TERM
Mar 29 22:47:38 phoscon systemd[1]: HoymilesZeroExport.service: Succeeded.
Mar 29 22:47:38 phoscon systemd[1]: Stopped HoymilesZeroExport Service.
Mar 29 22:47:38 phoscon systemd[1]: Started HoymilesZeroExport Service.
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     Log write to file: False
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     Author: reserve85 / Script Version: 1.14
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     Config file V 1.14
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     Inverter 0 reachable: True
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     setting new limit to 1500 Watt
Mar 29 22:47:39 phoscon python3[31913]: 2023-03-29 22:47:39 INFO     Inverter 0: setting new limit from 0 Watt to 1500 Watt
Mar 29 22:47:49 phoscon python3[31913]: 2023-03-29 22:47:49 INFO     Inverter 0 reachable: True
Mar 29 22:47:49 phoscon python3[31913]: 2023-03-29 22:47:49 INFO     powermeter: 281  Watt
Mar 29 22:47:49 phoscon python3[31913]: 2023-03-29 22:47:49 INFO     setting new limit to 1500 Watt
Mar 29 22:47:49 phoscon python3[31913]: 2023-03-29 22:47:49 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt
Mar 29 22:48:09 phoscon python3[31913]: 2023-03-29 22:48:09 INFO     Inverter 0 reachable: True
Mar 29 22:48:10 phoscon python3[31913]: 2023-03-29 22:48:10 INFO     powermeter: 278  Watt
Mar 29 22:48:10 phoscon python3[31913]: 2023-03-29 22:48:10 INFO     setting new limit to 1500 Watt
Mar 29 22:48:10 phoscon python3[31913]: 2023-03-29 22:48:10 INFO     Inverter 0: setting new limit from 1500 Watt to 1500 Watt

reserve85, 2023-03-30:
Perfekt, freut mich zu hören!