reserve85/HoymilesZeroExport #26: Erhöhung Limit wenn der WR nicht mal das aktuelle Limit produziert

Leviathan09, 2023-03-30:
Hallo @reserve85 ,
Habe gestern dein Skript entdeckt und finde es bisher echt super da meine selbstgebastelte Steuerung über ioBroker nur mäßig gut lief.
Dafür erstmal vielen Dank.

Nun zu einer Frage:
Aktuell ist es ja so das das Limit erhöht wird wenn vom 'powermeter' außerhalb des Target Points +/- Toleranz ein Wert kommt.
Könnte man das noch um eine Prüfung ergänzen ob der WR überhaupt so viel Produziert wie das Limit ist?

Aktuell wird ja einfach erhöht und wenn dann wieder genug vom WR kommt, reduziert um den Target Point zu erreichen.
Aber das Limit zu erhöhen (zb. von 100W auf 300W) bringt ja nichts wenn vom WR aktuell durch Wolken nur (zb. 80W) kommen.

Habe noch nicht alle closde issues durch, also sry falls die Frage schon kam.

Ollipop030, 2023-03-30:
Moin moin,

ich misch mich hier mal, grad zufällig gelesen.
@Leviathan09 :

Dazu müsste man Ahoy ständig abfragen. Das Problem ist aber, das die Werte nur alle paar Minuten aktualisiert werden, das hilft einem dann auch nicht weiter.

Hast du einen Zwischenzähler an der PV Anlage der json Daten auswirft? Dann schau dir in der *.ini mal [INTERMEDIATE_TASMOTA] an, damit kannst das Limit regulieren, damit es nicht bis zum Maximum erhöht wird, wenn mal eine Wolke dazwischen ist.

reserve85, 2023-03-30:
Hi, freut mich dass dir das Script gefällt. 
Ja, das ist nicht so ohne weiteres möglich, weil du ohne Limiterhöhung gar nicht weisst wann der Himmel aufklart. Außerdem gibt es noch den Fall, dass ein oder mehrere Module verschattet werden, dann ist die Berechnung vom Limit auch abweichend zum Output. Außerdem ist AHOY teilweise mehrere Sekunden hinter dem "Ist-Wert" - dann ist es ungünstig darüber einen Wert zu errechnen.

Leviathan09, 2023-03-30:
@Ollipop030 
Danke für deine Antwort, das schau ich mir mal an.
Tue mir noch ein kleinwenig schwer mit manchen Parameter.
Ich hab ein Shelly Plus 1PM an der PV den ich abfragen könnte

Ollipop030, 2023-03-30:
Das sollte kein Problem sein, die Daten kann man von {ShellyIP}/status bekommen.

reserve85, 2023-03-30:
Beim 1PM ist das aber ein anderes Result-JSON, das geht geringfügig anders - ist dann unter 'meters' versteckt wie ich das lese. 
Ich kann den 1PM als Zwischenzähler gerne noch mit aufnehmen, ich bräuchte aber ein JSON.

@Leviathan09 Kannst du das Result von http://{SHELLY_IP}/status hier posten?

Leviathan09, 2023-03-30:
Klar ein Output kann ich dir liefer.
Die Abfrage ist bei den Plus Geräten etwas ander.
Die URL lautet: http://{SHELLY_IP}/rpc/Switch.GetStatus?id=0

Die Ausgabe sieht so aus:
`{"id":0, "source":"loopback", "output":true, "apower":114.0, "voltage":230.8, "current":0.494, "aenergy":{"total":61134.422,"by_minute":[95.664,1864.020,1677.475],"minute_ts":1680178322},"temperature":{"tC":33.5, "tF":92.3}}
`

Der Wert ist dementsprechen "apower"

reserve85, 2023-03-30:
> @reserve85 Also ich hatte es in eine Blockly unter ioBroker das er nichts gemacht hat solange Output < Limit Das funktionierte soweit auch nur eben mit großer verzögerung durch MQTT. Ich werde aber mal den Vorschlag von Oliipop020 testen
Ist ein anderes Konzept, so ist das bei mir nicht aufgebaut - da ändert auch der Shelly nix. Bei mir wird immer erhöht oder verringert wenn nicht im Toleranzbereich. Der Shelly ändert das auch nix, lediglich zum schnellen herunterregeln und zum eingrenzen, dass nicht "bis zur obergrenze" ständig erhöht wird.

reserve85, 2023-03-30:
ok, hab ich erweitert und noch ein Bug gefunden bei der Auswahl von dem Zwischenzähler -> da wurde immer der Typ vom "normalen" Zähler genommen. @Ollipop030  du hast beides mit Tasmota oder? daher ist das nicht aufgefallen...

ich erstelle gleich die DEV und es wäre super, wenn @Leviathan09 den Shelly 1PM testen könntest als "Intermediate" zähler.

Ollipop030, 2023-03-30:
@reserve85 Ja habe ich.

Ui, schon Version 1.16. Da kommt man kaum hinterher. 😄

Leviathan09, 2023-03-30:
Dev eingespielt und die Daten eingetragen
Bisher sieht alles gut aus. Ob sich jetzt am Verhalten was geändert hat sehe ich jetzt bisher noch nicht aber ich hab halt auch gerade genau mein Tracking Point von nahe 0.

Muss nur dem lumapu für die neuste ahoyDTU Version ein issue eröffnen, seit dem Update stürzt das Teil nämlich immer wieder ab.

Leviathan09, 2023-03-30:
> Echt? Bei mir läuft das Release richtig gut (V0.6.0). Du solltest im Log irgendwann dann sowas hier sehen: `intermediate meter: xxx Watt` das ist dann der Wert vom Shelly.

Tauchte bisher nicht auf bei mir.

Leviathan09, 2023-03-30:
Scheint noch nicht zu funktionieren

2023-03-30 15:34:14,892 INFO     Inverter 0: setting new limit from 0 Watt to 600 Watt
2023-03-30 15:34:24,914 INFO     Intermediate Meter Test:
2023-03-30 15:34:24,940 INFO     powermeter: 153  Watt
2023-03-30 15:34:24,959 INFO     Inverter 0 reachable: True

reserve85, 2023-03-30:
ich depp, geht mit 
```
GetHoymilesActualPower()
```

famlac9805, 2023-03-30:
Sorry wenn ich da so reinplatze. Wenn ich nun auch auf die neueste Version updaten will, dann mache ich die Schritte

**wget https://github.com/reserve85/HoymilesZeroExport/archive/refs/heads/main.zip
unzip main.zip
rm main.zip
mv HoymilesZeroExport-main/ HoymilesZeroExport/
cd HoymilesZeroExport/**

 einfach nochmal und kopiere mir die neue 

**sudo nano HoymilesZeroExport_Config.ini**

rein, passe sie wieder an und bin auf aktuellem Stand oder?

Ollipop030, 2023-03-30:
@famlac9805 

Du brauchst nur die *.ini und die *.py zu ersetzen, vorher aber den Dienst abschalten. Die *.ini musst du dann logischweise neu anpassen.

reserve85, 2023-03-30:
Kannst auch komplett neu installieren. ich habe seit heute das Installscript angepasst dass das funktioniert. Also du müsstest dann von der ini eine sicherheitskopie anlegen damit du die Werte noch hast und das Verzeichnis löschen. Dann vorgehen wie Schritt 1 in der readme

Ollipop030, 2023-03-30:
@reserve85 

Wird der alte Dienst damit deinstalliert? Gibt es irgendwelche Vorteile den Dienst neu zu installieren?

reserve85, 2023-03-30:
Könnte ja sein dass das Verzeichnis wechselt vom Script, dann wäre der Service dann tot. Also besser entfernen und hinzufügen. Geht ja automatisch

reserve85, 2023-03-30:
Öh gute Frage bei mir gehts

Ollipop030, 2023-03-30:
Bei mir auf dem Pi3 nicht, nur als pip3. Bin noch auf Buster.

Ollipop030, 2023-03-30:
Anscheinend ist pip für python2 und pip3 für python3.

Ollipop030, 2023-03-30:
Hier wird der Unterschied erklärt:

https://www.pythonpool.com/pip-vs-pip3/

pip3 macht dann wohl mehr Sinn, python2 hat ja gar keinen Support mehr.

famlac9805, 2023-03-30:
so. ich habs verkackt. bitte nochmals um entschuldigung das ich dieses issue kapere. wie kann ich das service komplett löschen. mit sudo ./uninstall_service.sh komm ich nicht weiter. 
ich würd gerne nochmals von vorne beginnen

pi@phoscon:~/HoymilesZeroExport $ sudo ./uninstall_service.sh
sudo: ./uninstall_service.sh: command not found
pi@phoscon:~/HoymilesZeroExport $ cd
pi@phoscon:~ $ sudo ./uninstall_service.sh
sudo: ./uninstall_service.sh: command not found

reserve85, 2023-03-30:
mach mal sudo chmod +x uninstall_service.sh

famlac9805, 2023-03-30:
ich habs im verzeichniss auch gemacht

pi@phoscon:~ $ cd HoymilesZeroExport/
pi@phoscon:~/HoymilesZeroExport $ sudo chmod +x uninstall_service.sh
chmod: cannot access 'uninstall_service.sh': No such file or directory
pi@phoscon:~/HoymilesZeroExport $ sudo ./uninstall_service.sh
sudo: ./uninstall_service.sh: command not found
pi@phoscon:~/HoymilesZeroExport $

famlac9805, 2023-03-30:
ich bedanke mich!!!!!

Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     Log write to file: False
Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     Author: Tobias Kraft / Script Version: 1.16
Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     Config file V 1.16
Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     Inverter 0 reachable: True
Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     setting new limit to 1500 Watt
Mar 30 17:32:13 phoscon python3[25741]: 2023-03-30 17:32:13 INFO     Inverter 0: setting new limit from 0 Watt to 1500 Watt
Mar 30 17:32:23 phoscon python3[25741]: 2023-03-30 17:32:23 INFO     Inverter 0 reachable: True
Mar 30 17:32:24 phoscon python3[25741]: 2023-03-30 17:32:24 INFO     powermeter: 246  Watt

closed 2023-03-30 by reserve85