reserve85/HoymilesZeroExport #28: Eure Config Parameter?

reserve85, 2023-03-30:
Wie sind eure Erfahrungen mit dem Script?
Ich stehe auf das Jump to Max Limit, damit ich möglichst sofort reagiere wenn ein Verbraucher dazukommt.
Mit Ahoy V0.6.0 konnte ich die Timings im Vergleich zu den Default-Values ganz schön redizieren und bekomme wirklich Top-Ergebnisse:

```
[COMMON]
INVERTER_COUNT = 1
SLOW_APPROX_LIMIT_IN_PERCENT = 25
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 10
SET_LIMIT_DELAY_IN_SECONDS = 3
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
POLL_INTERVAL_IN_SECONDS = 1
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = true
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 15
ENABLE_LOG_TO_FILE = false
LOG_BACKUP_COUNT = 30
SET_LIMIT_RETRY = -1

[CONTROL]
POWERMETER_TARGET_POINT = -75
POWERMETER_TOLERANCE = 5
POWERMETER_MAX_POINT = 0
```

Leviathan09, 2023-03-30:
Also ich hab dein Skript derzeit mit folgenden Einstellungen laufen und muss sagen es funktionierte heute echt gut (Gestern war gabs hier kein Sonne)

Muss aber sagen so 100% verstanden was welche Option macht habe ich noch nicht, daher sind die Einstellungen so auf "Gut Glück" gewählt 😄 

```
[COMMON]
INVERTER_COUNT = 1
SLOW_APPROX_LIMIT_IN_PERCENT = 10
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 20
SET_LIMIT_DELAY_IN_SECONDS = 5
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
POLL_INTERVAL_IN_SECONDS = 1
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = False
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
ENABLE_LOG_TO_FILE = true
LOG_BACKUP_COUNT = 5
SET_LIMIT_RETRY = 3

[CONTROL]
POWERMETER_TARGET_POINT = 0
POWERMETER_TOLERANCE = 15
POWERMETER_MAX_POINT = 550
```

reserve85, 2023-03-31:
Ja, einfach Fragen, ist ziemlich viel ich weiß. 

SLOW_APPROX_LIMIT_IN_PERCENT = 10: das ist etwas wenig wenn du ein 1500W Inverter betreibst, dann wird ab > 150W (10%) Limitänderung das ganze dann langsamer nach unten geregelt. Wenn du z.B. 200W überproduzierst: Dann greift der SLOW_APPROX_FACTOR_IN_PERCENT: anstelle der 200W wird dann 160W geregelt. Und im nächsten Schritt (Loop Interval in Seconds) die restlichen 40W. Das flacht die Regelung etwas ab bei großen Änderungen am Limit.

Mit Ahoy kannst du locker LOOP_INTERVAL_IN_SECONDS  = 10s und SET_LIMIT_DELAY_IN_SECONDS = 3 einstellen, das beschleunigt enorm.

Wenn du ohne Jump eingestellt hast, würde ich MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER aktivieren (so 10%-15%), das begrenzt das Erhöhen des Limits obwohl nicht mehr produziert werden kann (zu wenig Sonne). Sonst läuft das Limit dauerhaft in Richtung 1500W, aber der Inverter dümpelt bei 300W herum.

Ollipop030, 2023-03-31:
Das sind meine Werte:

```
[COMMON]
INVERTER_COUNT = 2
SLOW_APPROX_LIMIT_IN_PERCENT = 20
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 11
SET_LIMIT_DELAY_IN_SECONDS = 1
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 1
POLL_INTERVAL_IN_SECONDS = 1
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = false
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 10
ENABLE_LOG_TO_FILE = true
LOG_BACKUP_COUNT = 7
SET_LIMIT_RETRY = 10

POWERMETER_TARGET_POINT = -60
POWERMETER_TOLERANCE = 35
POWERMETER_MAX_POINT = 0
```

Alles funktioniert wunderbar. Auch hält die DTU das aus. Bei mir das Problem: Der Zähler gibt mir immer nur alle 10 Sekunden ein Wert. Das stört aber grundsätzlich die Funktion nicht. Wo es ein wenig stört: Bei 3D Druck oder wenn die Waschmaschine läuft. In dem Moment, in dem ein Motor anläuft werden ganz kurz mal knapp 300 Watt gezogen. Wenn genau dann vom Zähler wieder ein Wert kommt wird für 11 Sekunden das Limit demensprechend erhöht.

So sah der Mittag gestern aus (es war aber auch sehr bewölkt):

[image]

Rot=Netzbezug
Grün=Einspeisung

Bis kurz nach 12 lief der Geschirrspüler, man kann gut sehen wann der Heizstab anspringt. Um ca. halb 1 wurde gekocht, da heizt das Ceranfeld. Hier sieht man dann auch kurz die zu hohe Einspeisung. Aber ansonsten dümpelt der Einspeisewert hier bei 60-80 Watt rum.

@reserve85 Hast du Lust meinen Zähler (emlog) offiziell mit aufzunehmen? Dann brauche ich nicht bei jedem Update die Domain und Prefixes ändern.

Leviathan09, 2023-03-31:
> Ja, einfach Fragen, ist ziemlich viel ich weiß.
>
> SLOW_APPROX_LIMIT_IN_PERCENT = 10: das ist etwas wenig wenn du ein 1500W Inverter betreibst, dann wird ab > 150W (10%) Limitänderung das ganze dann [...]

Also ich hab nen 600W WR.
Schau mir die Einstellungen nachher nochmal an.

reserve85, 2023-03-31:
> @reserve85 Hast du Lust meinen Zähler (emlog) offiziell mit aufzunehmen? Dann brauche ich nicht bei jedem Update die Domain und Prefixes ändern.

ja sehr gerne, mach einfach ein neues Issue auf und lass mir die Werte darin zukommen die ich da brauche, bzw. deine Implementierung.

Leviathan09, 2023-03-31:
Ich hab meine Werte jetzt nochmal etwas angepasst und da ist mir aufgefallen das da wohl iwas noch nicht so ganz passt.
Die Funktion die ich in dem issue #26 angesprochen hatte hast du ja schon drin gehabt, kam nur bei mir wohl nie zum tragen.
Aber wenn jetzt der Fall eintritt das mein Limit höher ist als die Produktion schreibt er folgendes raus:

```
2023-03-31 14:21:37,503 INFO     Cut limit to 302 Watt, limit was higher than 15 percent of live-production
2023-03-31 14:21:37,503 INFO     Not enough energy producing: increasing limit
2023-03-31 14:21:37,503 INFO     setting new limit to 353 Watt
2023-03-31 14:21:37,503 INFO     Inverter 0: setting new limit from 352 Watt to 353 Watt
```
```
2023-03-31 14:21:24,085 INFO     Inverter 0 power producing: 212  Watt
2023-03-31 14:21:24,085 INFO     Cut limit to 302 Watt, limit was higher than 15 percent of live-production
2023-03-31 14:21:24,088 INFO     Not enough energy producing: increasing limit
2023-03-31 14:21:24,088 INFO     setting new limit to 352 Watt
2023-03-31 14:21:24,089 INFO     Inverter 0: setting new limit from 356 Watt to 352 Watt
```
```
```
Laut output gibt er an das Limit zu reduzieren, zb auf 302 Watt. Im nächsten Schritt kommt dann aber wieder das increasing limit und mischt sich ein. Dadurch wird das Limit zwar reduziert oder bleibt annähernd gleich aber es wird nicht auf das reduziert was er bei "Cut limit to XXX Watt" angibt

Ollipop030, 2023-03-31:
@Leviathan09 Du sollest die Toleranz nicht so eng einstellen, ruhig 30 Watt. Und der Targetpoint sollte auch weiter im Negativen liegen, versuch mal -50 Watt. Bei 15 Watt Einspeisung muss das Limit ja schon wieder verringert werden, bei 15 Netzbezug wieder erhöht. So kleine Sprünge wirst du aber wahrschlich gar nicht hinbekommen.

mark82436, 2023-03-31:
Das sind meine Werte mit den ich im Moment teste ist etwas mehr Toleranz bei Bewölkung damit mit die ganze Zeit rauf und runter geregelt wird.

[COMMON]
# Number of Inverters
INVERTER_COUNT = 1
# max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
# interval time for setting limit to Hoymiles
LOOP_INTERVAL_IN_SECONDS = 20
# delay time after sending limit to Hoymiles Inverter
SET_LIMIT_DELAY_IN_SECONDS = 5
# delay time after sending limit to Hoymiles Inverter when using more than one Inverter
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
# polling interval for powermeter (must be < LOOP_INTERVAL_IN_SECONDS)
POLL_INTERVAL_IN_SECONDS = 1
# when powermeter > 0: (True): always jump to maxLimit of inverter; (False): increase limit based on previous limit
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = true
# max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
# enable logging to file
ENABLE_LOG_TO_FILE = false
# how many logfiles you wish to keep
LOG_BACKUP_COUNT = 30

[CONTROL]
# --- global defines for control behaviour ---
# this is the target power for powermeter in watts
POWERMETER_TARGET_POINT = -100
# this is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 50
# this is the max power to regulate the limit. if your powermeter is above this point, the limit jumps to 100% (when JUMP_TO_MAX_LIMIT_ON_GRID_>
POWERMETER_MAX_POINT = 200

mark82436, 2023-03-31:
Ich habe es bei mir so laufen, damit er nicht ständig hin und her regelt
somit ist die Toleranz größer

`[COMMON]
INVERTER_COUNT = 1
SLOW_APPROX_LIMIT_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 20
SET_LIMIT_DELAY_IN_SECONDS = 5
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
POLL_INTERVAL_IN_SECONDS = 1
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = true
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
ENABLE_LOG_TO_FILE = false
LOG_BACKUP_COUNT = 30

[CONTROL]
POWERMETER_TARGET_POINT = -100
POWERMETER_TOLERANCE = 50
JUMP_TO_MAX_LIMIT_ON_GRID_>
POWERMETER_MAX_POINT = 200`

Leviathan09, 2023-04-01:
> @Leviathan09 Du sollest die Toleranz nicht so eng einstellen, ruhig 30 Watt. Und der Targetpoint sollte auch weiter im Negativen liegen, versuch mal -50 Watt. Bei 15 Watt Einspeisung muss das Limit [...]

Werde die Toleranz mal anpassen.
Aber die 0 Watt als Target-Point sind ja so gewollt, ich will bei um die 0 liegen.
Nicht bei -50 oder so

Ollipop030, 2023-04-01:
> Aber die 0 Watt als Target-Point sind ja so gewollt, ich will bei um die 0 liegen.

Dann wirst du aber mehr Netzbezug haben, denn die Toleranz lässt bei dir dann ja auch positive Werte zu, das Script wird in dem Fall gar nicht eingreifen. 

Lieber ein bisschen einspeisen als Puffer für Schwankungen beim Verbrauch. Genau die 0 zu treffen und zu halten wird wohl nicht möglich sein. Es wird ja auch nur alle 10 Sekunden angepasst.

Soll nur ein Tipp sein.

Leviathan09, 2023-04-01:
> Dann wirst du aber mehr Netzbezug haben, denn die Toleranz lässt bei dir dann ja auch positive Werte zu, das Script wird in dem Fall gar nicht eingreifen.
>
> Lieber ein bisschen einspeisen als Puffer [...]

Ich werde die Tage mal beobachten und sehen wie sich das ganze verhält und mal testen was mit unterschiedlichen Parametern passiert 👍

Gestern zumindest lief es ganz gut um den 0 Punkt, mal so -15 bis +15 war die übliche Schwankung.
Heute wird aber nix mit testen, Maximum der Anlage heute waren 10W, hier ist es mega bewölkt, dunkel und nur am regnen.

Aber danke für die Tipps. Werde die Ergebnisse demnächst dann hier mal mitteilen

Sposch123, 2023-04-15:
Hi! 
Erst einmal herzlichen Danke für das tolle Skript - auch als echter Amateur bin ich gut zurecht gekommen und habe es ans laufen bekommen. Ich habe allerdings eine Weile gebraucht, um das Format meines Hichi herauszubekommen. 
Für weitere Anfänger wie mich, die mit einem Tasmota-Leser arbeiten, schlage ich deshalb eine Ergänzung in der config.ini vor:

# the following three constants describes how to navigate through the Tasmota-JSON
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
Vorgeschlagene Ergänzung
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010

Herzlichen Dank noch einmal und viele Grüße

feeti1, 2023-04-17:
Hallo,

Ich habe das Script jetzt seit gestern mit mit ahoydtu und einem HM-1500 auf 48V Batterie am laufen.
Die Funktion ist wirklich sehr ordentlich.
Alles andere, das Ich vorher probiert habe war nicht so gut, oder hat überhaupt nicht funktioniert.
Als absoluter Neuling auf diesem Gebiet, eine Frage an die Profis, wäre es möglich den vom Inverter für die Strings gemessen Spannungswert für ein Änderung des Limit bzw. die Abschaltung des Inverters zu verwenden?
Zum Beispiel möchte Ich ab 48V am String das Powerlimit auf 300W setzen und dann vielleicht ab 47V den Inverter ausschalten.
Könnte man so etwas integrieren?
Und wenn ja, wie in etwa müsste Ich da vorgehen?

mfg Dirk

keinprofi, 2023-05-01:
Was ist in Sachen Einspeisung reralistisch möglich? Ich finde meine Einspeisewerte ein wenig hoch:

[image]

So sind mir gestern etwa 7kW übrig geblieben und 1,85kW eingespeist.
AM 28.04. habe ich rund 3kW und am 25.04. knapp 3,2kW erzeugt

Meine Config:
[image]

Lässt sich das noch optimieren?

reserve85, 2023-05-01:
Kannst du das visualisiert Darstellen? Das würde eine Menge bringen.

Prinzipiell kommt das auch immer auf deine Verbraucher drauf an. Wenn du z.B. einen Ofen anhast der alle 5 sekunden für 2 sekunden 2000W zieht, dann wird die Regelung nicht gut funktionieren.

Du hast -50W als Target eingestellt, sobald du halt eine 60W Birne einschaltest oder du im Energieverbrauch schwankungen von >50W hast, dann wird dein Powermeter positiv unds somit dein Inverter direkt auf 100% ausgeregelt weil du "jump_to_max_limit_on_grid_usage" auf TRUE hast.
Diese Option verursacht zwar eine höhere Einspeisung, dafür reagiert dein Inverter sofort wenn du Strom kaufen müsstest. Das musst du halt abwägen was für dich wichtiger ist. 

Mit der Option probiere mal 75 Watt oder sogar 100 Watt, dann wird der Inverter seltener auf 100% gepusht.

SlowApproxLimitInPercent könntest du auch etwas erhöhen, bei mir ist 25 eigentlich ziemlich gut. 

Hoffe das Hilft dir, eine Grafik wäre super.

keinprofi, 2023-05-01:
Visualisiert... würde ich gerne, wenn ich wüsste wie. 

Kann ich irgendwie eine Datenbank einsetzen um alles aufzuzeichnen und dann auszugeben? Ich habe ein anderes Projekt gesehen da wird das u.a. mit Grafana gemacht. Aber wie mein Username schon sagt: Ich bin alles andere als ein Profi. Dieses Projekt sind meine allerersten Erfahrungen mit Linux. Ich bin wissbegierig und lernfähig, aber ich brauche echt Hilfe dabei! Wenn Du das machen kannst, bzw. möchtest(?).

Zur Config:
Ich werde die Option mal auf 100 Watt anpassen und das SlowApproxLimitInPercent auf 25% (was 575 Watt wären) einstellen und beobachten. 

Danke bis hierher!

reserve85, 2023-05-01:
Naja, am einfachsten wäre es IoBroker oder HomeAssistant zu installieren. Damit kannst du deine Zähler ohne großen Aufwand hinzufügen und Ahoy/OpenDTU sowieso. Du musst dich dennoch reinarbeiten, da gibt es aber tausende Tutorials und Beispiele. Welchen Zähler hast du?

keinprofi, 2023-05-02:
Reinarbeiten ist kein Ding. Hast Du ein gutes Tut? 
Ich habe einen Logarex lk13be803039.

keinprofi, 2023-05-15:
Ist es möglich den "POWERMETER_TARGET_POINT" automatisiert zu bestimmten Tageszeiten anzupassen?
Mir geht es dabei um eine Nachteinspeisung, da würde ich den Wert gerne automatisch deutlich reduzieren, da Nachts keine Spitzen zu erwarten sind, ich aber möglichst noch weniger als tagsüber einspeisen möchte.

Was den Wert tagsüber angeht, da habe ich mit -100 erste Erfolge erzielt. Soll heißen das ich nun schon weniger einspeise und mich nun an einen "idealen" Wert herantaste. Ich nehme an, dass dieser tatsächlich bei 75 Watt sein wird.

reserve85, 2023-05-15:
@keinprofi: bei Batteriebetrieb würde ich dir generell empfehlen: TargetPoint = 0 und JumpToMaxLimitOnGridUsage = false

Die Lösung für unterschiedliches Tages- und Nachtlimit müsstest du selbst realisieren, ggf. mit einem automatisierten CronJob der dir den Wert in der *.ini anpasst und das Script dann neustartet. In meinem Script würde ich das ungern implementieren.

keinprofi, 2023-05-15:
Ja, bei Batteriebetrieb ist 0 definitiv das Ziel, aber tagsüber ist das keine brauchbare Option.

Na gut, dann muss ich mal sehen ob ich da was hinbekomme.........

reserve85, 2023-05-15:
Ja dann würde ich wirklich die Linie fahren für Tag- und Nachtmodus jeweils eine eigene ini zu hinterlegen (z.B. HoymilesZeroExport_Day.ini und HoymilesZeroExport_Night.ini) und über ein Script zu bestimmter Zeit die Datei in den neuen Dateinamen HoymilesZeroExport.ini zu kopieren (mit automatischen Überschreiben). Dann muss nur noch ./restart.sh ausgeführt werden.

keinprofi, 2023-05-15:
Gute Idee. Ich werde mich da mal dran machen.

Danke!

keinprofi, 2023-05-25:
Kann ich das so umsetzen?
[image]

reserve85, 2023-05-25:
Hi, ich würde noch einfacher vorgehen:

Erstelle Day_Config.ini + Night_Config.ini in deinem HoymilesZeroExport Verzeichnis.

Erstelle einen CronJob:

1. um 05:55 Uhr die Datei /etc/HoymilesZeroExport/Day_Config.ini nach /etc/HoymilesZeroExport/HoymilesZeroExport_Config.ini kopieren und überschreiben
2. um 06:00 Uhr führe Script "sudo /etc/HoymilesZeroExport/restart.sh" aus
3. um 19:55 Uhr die Datei /etc/HoymilesZeroExport/Night_Config.ini nach /etc/HoymilesZeroExport/HoymilesZeroExport_Config.ini kopieren und überschreiben
4. um 20:00 Uhr führe Script "sudo /etc/HoymilesZeroExport/restart.sh" aus

Dazu hab ich mal ChatGPT gefragt (Danke dafür 🥇):

> Öffne ein Terminal und gib den folgenden Befehl ein, um die Crontab-Datei zu bearbeiten:
> 
> `crontab -e`
> Füge die folgenden Zeilen am Ende der Datei hinzu, um die gewünschten CronJobs einzurichten:
> 
> ```
> 55 5 * * * cp -f /etc/HoymilesZeroExport/Day_Config.ini /etc/HoymilesZeroExport/HoymilesZeroExport_Config.ini
> 0 6 * * * sudo /etc/HoymilesZeroExport/restart.sh
> 55 19 * * * cp -f /etc/HoymilesZeroExport/Night_Config.ini /etc/HoymilesZeroExport/HoymilesZeroExport_Config.ini
> 0 20 * * * sudo /etc/HoymilesZeroExport/restart.sh
> ```

Damit sollte das so laufen wie du es vorhast. Die Zeiten und den Ordner musst du noch anpassen, so würde ich das jedenfalls lösen.

Gruß Tobias

keinprofi, 2023-05-25:
Meine Ergebnisse stammen auch von ChatGPT... :-D

reserve85, 2023-05-25:
Also ob das so funktioniert kann ich dir nicht sagen, ich kenn das so nicht. Bin leider auch nicht zu Hause sonst würde ich es mal schnell eintippen. 
Jedenfalls: du musst das Script trotzdem neu starten, die Config wird nur ein mal geladen (beim Starten des Scripts).

Zynus, 2023-06-01:
Hallo zusammen,
hat von euch er Erfahrung mit einem HM an einem 24V Akku, der nur Nachts an ist.
Nachts schwankt der Strombedarf nicht so stark, maximal schaltet der Kühlschrank oder die Gefriertruhe ein, die dann für 10/15 min laufen.
Da mein Grundbedarf nachts zwischen 130 und 220W schwankt, habe ich das Skript auf 250W begrenzt.

Wie geht ihr mit diesen 2 Parameter um?

- max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
- if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20

Zynus, 2023-06-01:
Startet das Skript automatisch nach einem Neustart meines Raspberry?

reserve85, 2023-06-02:
Wenn du es als Service installiert hast dann ja.

keinprofi, 2023-06-14:
So, ich denke ich habe das derzeit beste für mich eingestellt.
Ich habe 3 WR: 400, 800 und 1500er. 
Meine Konfig habe ich folgendermaßen angepasst:
[COMMON]
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = false

[CONTROL]
POWERMETER_TARGET_POINT = -50
POWERMETER_TOLERANCE = 25
POWERMETER_MAX_POINT = 0

Damit speise ich zwischen 400 und 700 Wh pro Tag ein. Ich beziehe (wenn nicht gerade der Herd läuft oder alles gleichzeitig) tagsüber ca. 1 kWh aus dem Netz (wenn ich nicht da bin weniger).

Beispiel gestern (Klima lief ab Mittag):
[image]

Ich rechne damit, dass om August mein Speicher kommt. Dann wird es nochmal spannend.

Aber: seit Montag steigt das Scrippt bis zu zweimal am Tag aus, bleibt einfach stehen. zu unregelmäßigen Zeiten morgens und abends. Ich muss dann neustarten. Hat das von euch auch schon jemand gehabt, bzw. jemand eine Idee warum das passiert? Ich habe an meiner Konfig nichts geändert und der Rest (Influx, ioroker, Grafan etc.) laufen ja stabil weiter.

keinprofi, 2023-06-14:
Da ich den Raspi heute früh neu gestartet habe bekomme ich nur das Log ab 8.48 Uhr. Ich werde es mal in der Konfig aktivieren und es Dir schicken wenn es wieder passiert. Heute morgen blieb es gegen halb 7 stehen meine ich. Aber wie gesagt, bisher gibt es kein genaues Muster, nur morgens und abends.

delacor, 2023-06-14:
Ist bei mir auch noch nie stehen geblieben.. Häuft sich aber irgendwie in letzter Zeit, die Meldungen.. mal sehen..

keinprofi, 2023-06-14:
Hier das Log. Ausstieg erfolgte um 17.30 Uhr.

Die Ausfälle des Tasmota (10.10.10.139) vor 17 Uhr habe ich provoziert, da ich mir eine Überwachung des Sensors gebastelt habe.
[log.txt](https://github.com/reserve85/HoymilesZeroExport/files/11748048/log.txt)

reserve85, 2023-06-14:
Poah, ohne Fehlermeldung… das konnte ich noch nie beobachten. Reagiert da der RasPi noch? Oder machst du power reset?

reserve85, 2023-06-14:
Version 1.24 stimmt?

keinprofi, 2023-06-14:
ja

keinprofi, 2023-06-14:
Das Script ist wieder ausgestiegen, diesmal 18.24 Uhr
[image]

Der Sensor ist (u.a.) auch um 18.24 Uhr ausgestiegen (wird alle 30 Sekunden gepingt und bei nichterreichen gibts ne Mail):
[image]

Wird damit zusammen hängen, oder was meinst du?

reserve85, 2023-06-14:
update mal (wenn du irgenddwann zeit hast) auf die neue dev version (1.41) - da habe ich ein timeout für die HTTP Abfragen hinzugefügt. Ich vermute dass die IP zwar erreichbar ist aber einfach keine Antwort zurück kommt und er "unendlich" in der Tasmota-Abfrage hängen bleibt.
https://github.com/reserve85/HoymilesZeroExport/tree/dev

keinprofi, 2023-06-18:
> 

Es wird in Kürze ernst mit meinem Speicher, denn der Versand wurde vorbereitet, sollte in ca. 2 Wochen ankommen.

Ich werde Deine Lösung nehmen, denn das was ich bisher probiert habe funktioniert nicht.
Zur Konfig:
POWERMETER_TARGET_POINT = 0 
nachts ist klar. Aber was mache am Besten mit 
POWERMETER_MAX_POINT?
Ich möchte ja sowohl möglichst viel Batteriestrom selbst nutzen, aber auch möglichst wenig Strom vom Netzbetreiber kaufen.
Hat da jemand Erfahrungswerte oder Vorschläge?

reserve85, 2023-06-19:
POWERMETER_MAX_POINT ist ja der Punkt an dem die normale Regelungsschleife unterbrochen wird und "sofort" nachgeregelt wird. 
Prinzipiell sollte der Punkt nur überschritten werden wenn du einen Verbraucher anschaltest damit halt sofort reagiert wird. Sollte immer nur die Ausnahme darstellen.
Ich würde aus dem Bauch raus POWERMETER_TARGET_POINT auf -20, Toleranz auf +20 und POWERMETER_MAX_POINT auf 75 oder 100 setzen.

reserve85, 2023-06-19:
ja probiere es einfach aus, muss jeder selbst wissen. Ich finde targetpoint = -20 ist ja nicht viel oder? 
Wenn du ihn auf 0 setzt und die Toleranz auf 20 lässt, dann kann es auch sein dass du eben "einkaufst" weil es halt um die 0 herumpendelt (+-20).

Wie auch immer, du musst "Jump to max limit" ausschalten und ich würde einfach 75 bis 100W dazu addieren und da dann den POWERMETER_MAX_POINT setzen.

keinprofi, 2023-06-19:
Ja -20 und 20 klingt nicht so unvernünftig. Ich werde mich da wohl vorsichtig rantasten müssen.
Danke!

feeti1, 2023-06-19:
Hallo,

also bei mir läuft das jetzt schon 2 Monate im 🔋 Betrieb.
Meine Config steht jetzt seit längerem 0 mit Toleranz 20.
Damit wird die Null ganz gut gehalten, solange niemand Zuhause ist.
Sobald meine Familie anwesend ist, fangen die Werte deutlich an zu schwanken.
In der Regel kommen häufige Wechsel um +-100Watt zustande.
Dafür ist die Regelung wahrscheinlich nicht schnell genug.
Ich sehe das als Systemgrenze und nicht als Fehler.
Allerdings ist bei mir gestern das Script einmal ausgestiegen.
Das ist mir nur aufgefallen, das ich immer wieder Mal in die Werte schaue.
Da waren meine Batterien (AGM) schon unter meiner Abschaltschwelle.
Da ich nicht Zuhause war, könnte ich das Log nicht sichern.
Habe aus der Ferne nur den Server neu gestartet.
Dann lief wieder alles.

mfg Dirk

reserve85, 2023-06-19:
Hi Dirk,
update bitte auf die neue Version, es gab ein Problem bei der HTTP Abfrage welches unter Umständen das Script abstürzen lassen konnte.

keinprofi, 2023-06-19:
Auf was hast Du den POWERMETER_MAX_POINT gestellt?

feeti1, 2023-06-19:
Bei mir steht der Maxpoint auf 1500

keinprofi, 2023-06-19:
1500? Wow, das ist mir zu hoch. Ich werde das mal mit -20 und 20  und... mal sehen testen und mich dann herantasten womit ich am besten fahre. 

Danke euch!

reserve85, 2023-06-19:
Das ist dann klar dass deine Regelung recht träge ist. Stelle das besser auf 75 oder 100 Watt dann wird die Regelung auch schneller reagieren wenn ein Verbraucher anspringt.

feeti1, 2023-06-19:
Werde ich Mal probieren. Ich bin der Meinung wenn ich das so niedrig eingestellt habe, ist der Hoymiles ab einer erforderlichen Einspeisung über dem Wert immer auf Maximum gegangen.
Kann aber auch sein ,dass ich da etwas übersehen oder missinterpretiert habe.

reserve85, 2023-06-19:
Ja, deshalb JumpToMaxWattsOnGridUsage auf false setzen. Dann springt er nicht auf Maximum.

feeti1, 2023-06-19:
Hallo,

ist genauso wie ich gedacht habe, sobald ich den Wert runter setze wird das gesetzte Limit nicht mehr eingehalten.
Der Hoymiles geht dann bis auf Anschlag.
Das möchte ich nicht.

reserve85, 2023-06-19:
Dann hast du den Wert 
`[COMMON]JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = false`
angepasst?

feeti1, 2023-06-19:
Der Wert steht auf false.
Sobald ich bei Maxpoint einen Wert unterhalb meines Maximal Limit eingebe, wird das Limit nicht mehr berücksichtigt.
Auch mit der 1.40 nicht.

Von: reserve85
Gesendet: Montag, 19. Juni 2023 20:04
An: reserve85/HoymilesZeroExport
Cc: feeti1; Comment
Betreff: Re: [reserve85/HoymilesZeroExport] Eure Config Parameter? (Issue #28)

Dann hast du den Wert
[COMMON]JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = false
angepasst?

reserve85, 2023-06-19:
Das muss ich mir morgen mal anschauen.

reserve85, 2023-06-19:
@feeti1 schickst du mir mal deine config?  Oder lädst sie hier im Anhang hoch?

reserve85, 2023-06-22:
Klingt gut, hab es gerade in den DEV Zweig https://github.com/reserve85/HoymilesZeroExport/tree/dev eingepflegt. 
Ungetestet.

160Wh klingt auch gut würde ich sagen. 
Der Max Point mit 10 ist aber glaube ich etwas zu niedrig? Kommt natürlich immer auf deine Schwankungen drauf an.

keinprofi, 2023-06-22:
Ich hatte den Max Point vorher auch höher. Aber Nachts gibt es Schwankungen in der Regel nur von Kühl- oder Gefrierschrank und die liegen über 70 Watt. Nachtlicht ist bei 3 Watt wenn es schaltet. Von daher mein Test mit 10. Ich beobachte das natürlich noch und experimentiere weiter. Aber wenn es bei 160 Wh bleibt, dann ist das - wie Du auch sagst - schon okay.

Ich kann die Dev heute Abend mal testen, wird aber spät...

keinprofi, 2023-06-24:
Also, wie oben erwähnt, eigentlich funktioniert es. Leider springt er immer wieder auf das eingestellte Limit, auch dann, wenn der Verbrauch deutlich höher ist! Dann springt er wieder RUNTER zum Limit.

[111 more replies]