reserve85/HoymilesZeroExport #29: emlog Zähler Integration

Ollipop030, 2023-03-31:
Diese Werte werden benötigt:

Erstmal zum Verständnis: emlog läuft auf einem PI, an dem man unterschiedlichste Zähler anschließt, Strom und Gas. Als Stromzähler werden alle möglichen Geräte unterstützt, Impulszähler, RS485 und IR. Zwischen Zähler und PI werkelt aber immer extra HW, damit der PI überhaupt damit arbeiten kann.
Ist eine out of the box Lösung und kostet ein wenig Geld. Ist aber in Minuten eingerichtet und alles ohne HA, ioBroker, Grafana usw.

Dadurch erreicht man alle Zähler über eine IP, nur die letzte Ziffer der Domain verweist dann auf den entsprechenden Zähler

{emlogIP}/pages/getinformation.php?heute&meterindex=1 (beim Powermeter)
{emlogIP}/pages/getinformation.php?heute&meterindex=3 (bei intermediate)

Also brauchst du in der *.ini

Für Powermeter
   IP Adresse
   Meterindex (die Zählernummer in emlog)

Für Einspeisezähler
   IP Adresse
   Meterindex (die Zählernummer in emlog)

Damit kann dann die Domain gebaut werden.

Die Input und Outputlabel heißen immer gleich, die brauchen nicht variabel angegeben werden:
Leistung170 und Leistung270 bei powermeter, Leistung170 bei Intermediate.

Hier ein Auszug:

`{"Leistung170":139,"Leistung171":0,"Leistung172":0,"Leistung173":0,"Leistung270":0,"Leistung271":0,"Leistung272":0,"Leistung273":0,"Waehrung":"EUR","Stand300":null,"M300":0,"Stand180":0,"Stand181":0,"Stand182":0,"Stand280":0,"Stand281":0,"Stand282":0,"Kwh180":0,"Kwh181":0,"Kwh182":0,"Kwh280":0,"Kwh281":0,"Kwh282":0,"Kwh300":0,"Betrag180":0,"Betrag181":0,"Betrag182":0,"Betrag280":0,"Betrag281":0,"Betrag282":0,"Betrag300":0,"Metertype":"Strom"}`

Außerdem muss hier Einspeisung - Netzbezug gerechnet werden, denn die Leistung wird nicht negativ dargestellt.

reserve85, 2023-03-31:
kannst du es mal probieren:

https://github.com/reserve85/HoymilesZeroExport/tree/dev_emlog

Ollipop030, 2023-03-31:
Inder ini fehlt noch USE_EMLOG = true unter [SELECT_POWERMETER], danach bekomme ich aber diesen Fehler:

Mär 31 13:16:50 raspberrypi python3[23350]: 2023-03-31 13:16:50 INFO     Config file V 1.17
Mär 31 13:16:50 raspberrypi python3[23350]: 2023-03-31 13:16:50 INFO     Ahoy: Inverter 0 reachable: True
Mär 31 13:16:51 raspberrypi python3[23350]: 2023-03-31 13:16:51 INFO     Ahoy: Inverter 1 reachable: True
Mär 31 13:16:51 raspberrypi python3[23350]: 2023-03-31 13:16:51 INFO     setting new limit to 1800 Watt
Mär 31 13:16:51 raspberrypi python3[23350]: 2023-03-31 13:16:51 INFO     Ahoy: Inverter 0: setting new limit from 0 Watt to 1500 Watt
Mär 31 13:16:52 raspberrypi python3[23350]: 2023-03-31 13:16:52 INFO     Ahoy: Inverter 1: setting new limit from 0 Watt to 300 Watt
Mär 31 13:16:54 raspberrypi python3[23350]: 2023-03-31 13:16:54 INFO     Ahoy: Inverter 0 reachable: True
Mär 31 13:16:54 raspberrypi python3[23350]: 2023-03-31 13:16:54 INFO     Ahoy: Inverter 1 reachable: True
Mär 31 13:16:54 raspberrypi python3[23350]: 2023-03-31 13:16:54 ERROR    Exception at GetPowermeterWatts
Mär 31 13:16:54 raspberrypi python3[23350]: 2023-03-31 13:16:54 ERROR    Expecting value: line 1 column 1 (char 0)

Ollipop030, 2023-03-31:
Blödsinn, läuft. Habe vergessen die alten Sachen unter Tasmota zu löschen.

reserve85, 2023-03-31:
ok super, sehe hier aber noch: 
`url = f'http://{EMLOG_IP}//pages/getinformation.php?heute&meterindex={EMLOG_METERINDEX}'`
sollte das nicht
`url = f'http://{EMLOG_IP}/pages/getinformation.php?heute&meterindex={EMLOG_METERINDEX}'` 
sein?

Ollipop030, 2023-03-31:
Auch grad gesehen. Ein / zuviel. Teste gerade...

reserve85, 2023-03-31:
ok,
USE_EMLOG = true unter [SELECT_POWERMETER] ist eingebaut, hatte es vergessen zu speichern.

Ich mache beim Start des Scripts noch eine Init Routine rein, dass alles mal abgefragt wird.

Ollipop030, 2023-03-31:
Irgendwie vermisse ich intermediate meter EMLOG:

Das wird mir komischerweise nie angezeigt bei Einspeisung.

Ollipop030, 2023-03-31:
Ah gefunden. Der / zuviel ist beide male zuviel, bei Powermeter und intermediate. Sieht jetzt alles aus, wie es soll, alle Werte werden korrekt angezeigt und berechnet.

reserve85, 2023-03-31:
Top, ich hab es nochmal in die DEV gespielt, wäre super wenn du nochmal die *.py datei überschreiben könntest und out of the box testen könntest. Da solltest du ja nix mehr anpassen müssen...

Ollipop030, 2023-03-31:
Funktioniert super.

```
Config file V 1.17
Mär 31 13:52:01 raspberrypi python3[540]: 2023-03-31 13:52:01 INFO     ---Init---
Mär 31 13:52:01 raspberrypi python3[540]: 2023-03-31 13:52:01 INFO     Ahoy: Inverter 0 reachable: True
Mär 31 13:52:01 raspberrypi python3[540]: 2023-03-31 13:52:01 INFO     Ahoy: Inverter 1 reachable: True
Mär 31 13:52:01 raspberrypi python3[540]: 2023-03-31 13:52:01 INFO     intermediate meter EMLOG: 300  Watt
Mär 31 13:52:01 raspberrypi python3[540]: 2023-03-31 13:52:01 INFO     setting new limit to 1800 Watt
Mär 31 13:52:01 raspberrypi python3[540]: 2023-03-31 13:52:01 INFO     Ahoy: Inverter 0: setting new limit from 0 Watt to 1500 Watt
Mär 31 13:52:02 raspberrypi python3[540]: 2023-03-31 13:52:02 INFO     Ahoy: Inverter 1: setting new limit from 0 Watt to 300 Watt
Mär 31 13:52:03 raspberrypi python3[540]: 2023-03-31 13:52:03 INFO     powermeter EMLOG: -66  Watt
Mär 31 13:52:04 raspberrypi python3[540]: 2023-03-31 13:52:04 INFO     ---Start Zero Export---
Mär 31 13:52:04 raspberrypi python3[540]: 2023-03-31 13:52:04 INFO     Ahoy: Inverter 0 reachable: True
Mär 31 13:52:04 raspberrypi python3[540]: 2023-03-31 13:52:04 INFO     Ahoy: Inverter 1 reachable: True
Mär 31 13:52:04 raspberrypi python3[540]: 2023-03-31 13:52:04 INFO     powermeter EMLOG: -66  Watt
Mär 31 13:52:05 raspberrypi python3[540]: 2023-03-31 13:52:05 INFO     powermeter EMLOG: -66  Watt
Mär 31 13:52:06 raspberrypi python3[540]: 2023-03-31 13:52:06 INFO     powermeter EMLOG: -66  Watt

```

Finde übrigens gut, das der Name des Zähler jetzt mit erwähnt wird. Und die Abfrage am Anfang macht eine Fehlersuche auch einfacher. Von meiner Seite ist das Ding "Ready for main".

Besten Dank

closed 2023-03-31 by reserve85