reserve85/HoymilesZeroExport #31: OpenDTU Script läuft nicht bei geschützter Live Ansicht

mark82436, 2023-03-31:
Wenn man in den Sicherheitseinstellungen den Schalter
Berechtigung
Nur-Lese-Zugriff auf die Weboberfläche ohne Passwort zulassen ausschaltet, hat das Script keinen Zugriff mehr auf die DTU
Passwort und User sind doch hinterlegt, dann sollte das doch gehen oder?

Mar 31 16:42:05 zeroexport python3[272]: 2023-03-31 16:42:05 INFO     Exception at GetHoymilesAvailable, Inverter not available
Mar 31 16:42:05 zeroexport python3[272]: 2023-03-31 16:42:05 ERROR    Expecting value: line 1 column 1 (char 0)

[image]

reserve85, 2023-03-31:
Ok schau ich mir mal an wenn ich dazu komme, bisher war nur das Limit setzen passwortgeschützt.

Weiß zufällig jemand wie ich das über die API mitsenden kann?

reserve85, 2023-04-01:
sollte gefixt sein in der aktuellen DEV
https://github.com/reserve85/HoymilesZeroExport/tree/dev

Wäre super wenn du es testen könntest.

mark82436, 2023-04-01:
Im ersten Moment schaut es gut aus, ich beobachte es einmal

```
Apr 01 12:28:54 zeroexport systemd[1]: Started HoymilesZeroExport Service.
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     Log write to file: False
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     Author: Tobias Kraft / Script Version: 1.18
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     read config file: /root/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     Config file V 1.17
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     ---Init---
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     OpenDTU: Inverter 0 power producing: 84  Watt
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     setting new limit to 1500 Watt
Apr 01 12:28:55 zeroexport python3[453]: 2023-04-01 12:28:55 INFO     OpenDTU: Inverter 0: setting new limit from 0 Watt to 1500 Watt
Apr 01 12:29:00 zeroexport python3[453]: 2023-04-01 12:29:00 INFO     powermeter Shelly 3EM: 1301  Watt
Apr 01 12:29:05 zeroexport python3[453]: 2023-04-01 12:29:05 INFO     ---Start Zero Export---
Apr 01 12:29:05 zeroexport python3[453]: 2023-04-01 12:29:05 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:29:05 zeroexport python3[453]: 2023-04-01 12:29:05 INFO     powermeter Shelly 3EM: 1680  Watt
Apr 01 12:29:05 zeroexport python3[453]: 2023-04-01 12:29:05 INFO     setting new limit to 1500 Watt
Apr 01 12:29:05 zeroexport python3[453]: 2023-04-01 12:29:05 INFO     OpenDTU: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
Apr 01 12:29:25 zeroexport python3[453]: 2023-04-01 12:29:25 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:29:25 zeroexport python3[453]: 2023-04-01 12:29:25 INFO     powermeter Shelly 3EM: 1261  Watt
Apr 01 12:29:25 zeroexport python3[453]: 2023-04-01 12:29:25 INFO     setting new limit to 1500 Watt
Apr 01 12:29:25 zeroexport python3[453]: 2023-04-01 12:29:25 INFO     OpenDTU: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
Apr 01 12:29:45 zeroexport python3[453]: 2023-04-01 12:29:45 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:29:45 zeroexport python3[453]: 2023-04-01 12:29:45 INFO     powermeter Shelly 3EM: 1589  Watt
Apr 01 12:29:45 zeroexport python3[453]: 2023-04-01 12:29:45 INFO     setting new limit to 1500 Watt
Apr 01 12:29:45 zeroexport python3[453]: 2023-04-01 12:29:45 INFO     OpenDTU: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
```

reserve85, 2023-04-01:
Top, ich hab noch die 1.18 draus gemacht, hatte ich vergessen

mark82436, 2023-04-01:
Ich habe auch noch einen Shelly Plug mit Tasmota zwischen WR und Netz soll ich den mal unter SELECT_INTERMEDIATE_METER configurieren?

reserve85, 2023-04-01:
Ist definitiv besser als die Powerdaten von der DTU zu nehmen, da die ja im besten Falle nur alle 5 Sekunden übertragen werden

mark82436, 2023-04-01:
da habe ich nur keinen Playload oder sehe ich den nicht vor lauter Bäumen

```
13:39:36.397 MQT: tele/tasmota_D7301C/SENSOR = {"Time":"2023-04-01T13:39:36","ANALOG":{"Temperature":16.5},"ENERGY":{"TotalStartTime":"2022-12-23T16:11:58","Total":91.573,"Yesterday":2.036,"Today":1.165,"Period":0,"Power":82,"ApparentPower":161,"ReactivePower":138,"Factor":0.51,"Voltage":236,"Current":0.680},"TempUnit":"C"}
```

reserve85, 2023-04-01:
Power:82 oder? Wie kommst du da dran mit welchem Befehl?

mark82436, 2023-04-01:
das ist aus der Konsole von Tasmota

dann müsste es ja so gehen

```
[INTERMEDIATE_TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP_INTERMEDIATE = xxx.xxx.xxx.xxx
# the following three constants describes how to navigate through the Tasmota-JSON
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"curr_w":500}}}
TASMOTA_JSON_STATUS_INTERMEDIATE = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX_INTERMEDIATE = ANALOG
# Power-MQTT label (the current solar-power in Watt).
TASMOTA_JSON_POWER_MQTT_LABEL_INTERMEDIATE = Power
```

mark82436, 2023-04-01:
ah ich glaube das müsste dann ENERGY sein als Playload

reserve85, 2023-04-01:
Gib das mal im Browser ein:
http://{TASMOTA_IP_INTERMEDIATE}/cm?cmnd=status%2010

dann siehst du das JSON wie es zurück kommt

mark82436, 2023-04-01:
da kommt das zurück

```
{"StatusSNS":{"Time":"2023-04-01T13:55:01","ANALOG":{"Temperature":17.5},"ENERGY":{"TotalStartTime":"2022-12-23T16:11:58","Total":91.593,"Yesterday":2.036,"Today":1.186,"Power":80,"ApparentPower":158,"ReactivePower":137,"Factor":0.50,"Voltage":238,"Current":0.665},"TempUnit":"C"}}
```

mark82436, 2023-04-01:
ich habe es jetzt mal so drin

```
[INTERMEDIATE_TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP_INTERMEDIATE = 192.168.1.208 
# the following three constants describes how to navigate through the Tasmota-JSON
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"curr_w":500}}}
TASMOTA_JSON_STATUS_INTERMEDIATE = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX_INTERMEDIATE = ENERGY
# Power-MQTT label (the current solar-power in Watt).
TASMOTA_JSON_POWER_MQTT_LABEL_INTERMEDIATE = Power
``` 

dann kommt das raus

```
Apr 01 12:52:49 zeroexport systemd[1]: Started HoymilesZeroExport Service.
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     Log write to file: False
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     Author: Tobias Kraft / Script Version: 1.18
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     read config file: /root/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     Config file V 1.18
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     ---Init---
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     intermediate meter Tasmota: 80  Watt
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     setting new limit to 1500 Watt
Apr 01 12:52:49 zeroexport python3[471]: 2023-04-01 12:52:49 INFO     OpenDTU: Inverter 0: setting new limit from 0 Watt to 1500 Watt
Apr 01 12:52:54 zeroexport python3[471]: 2023-04-01 12:52:54 INFO     powermeter Shelly 3EM: 719  Watt
Apr 01 12:52:59 zeroexport python3[471]: 2023-04-01 12:52:59 INFO     ---Start Zero Export---
Apr 01 12:52:59 zeroexport python3[471]: 2023-04-01 12:52:59 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:52:59 zeroexport python3[471]: 2023-04-01 12:52:59 INFO     powermeter Shelly 3EM: 685  Watt
Apr 01 12:52:59 zeroexport python3[471]: 2023-04-01 12:52:59 INFO     setting new limit to 1500 Watt
Apr 01 12:52:59 zeroexport python3[471]: 2023-04-01 12:52:59 INFO     OpenDTU: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
Apr 01 12:53:19 zeroexport python3[471]: 2023-04-01 12:53:19 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:53:20 zeroexport python3[471]: 2023-04-01 12:53:20 INFO     powermeter Shelly 3EM: 740  Watt
Apr 01 12:53:20 zeroexport python3[471]: 2023-04-01 12:53:20 INFO     setting new limit to 1500 Watt
Apr 01 12:53:20 zeroexport python3[471]: 2023-04-01 12:53:20 INFO     OpenDTU: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
Apr 01 12:53:40 zeroexport python3[471]: 2023-04-01 12:53:40 INFO     OpenDTU: Inverter 0 reachable: True
Apr 01 12:53:40 zeroexport python3[471]: 2023-04-01 12:53:40 INFO     powermeter Shelly 3EM: 674  Watt
Apr 01 12:53:40 zeroexport python3[471]: 2023-04-01 12:53:40 INFO     setting new limit to 1500 Watt
Apr 01 12:53:40 zeroexport python3[471]: 2023-04-01 12:53:40 INFO     OpenDTU: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
```

reserve85, 2023-04-01:
Ja dann StatusSNS, ENERGY und Power

mark82436, 2023-04-01:
mehr testen kann ich heute nicht bei dem sch.. Wetter
dafür Drucke ich ein paar DTU Gehäuse

reserve85, 2023-04-01:
Ich hab meine aus Holz gelasert.
[image]

closed 2023-04-01 by reserve85

mark82436, 2023-04-01:
Ich habe meinen aus Bambus PLA gedruckt
[image]
[image]
[image]