reserve85/HoymilesZeroExport #32: Support von Shelly Pro 3EM

onceuponatime78, 2023-03-31:
Hi,
ich habe bei mir einen Shelly Pro 3EM verbaut und den Code folgendermaßen erweitert um es verwenden zu können:

`import websocket
import json

....

def GetPowermeterWattsShellyPro3EM():
 payload = {"id": 1,"method": "Shelly.GetStatus"}
 url = f"ws://{SHELLY_PRO3EM_IP}/rpc"
 ws = websocket.create_connection(url)
 ws.send(json.dumps(payload))
 result = ws.recv()
 ws.close()
 Watts = json.loads(result)["result"]["em:0"]["total_act_power"]
 logger.info("powermeter: %s %s",Watts," Watt")
 return int(Watts)`

Wäre toll, wenn Du den support offiziellen dazu geben könntest.

reserve85, 2023-04-01:
Danke für den Code, mir war nicht bewusst dass es da zwei Versionen gibt, auch vom 1PM. 
Ich habe den Code noch etwas umgeschrieben (analog zu den anderen Funktionen) und implementiert.
Könntest du es bitte vorab testen bevor ich es in die MAIN pushe?
https://github.com/reserve85/HoymilesZeroExport/tree/dev

reserve85, 2023-04-01:
ich nutze das hier gerade nochmal:
@Leviathan09: bei der aktuellen Version war noch ein Bug drin mit dem 1PM - ich hatte da zum einen die Abfrage vergessen und 2. war die Benamung zu deinem Shelly falsch. Du müsstest dann in deiner INI auf die USE_SHELLY_PLUS_1PM_INTERMEDIATE Variante wechseln.

onceuponatime78, 2023-04-01:
Danke fürs übernehmen - auslesen vom Pro 3EM funktioniert so bei mir

closed 2023-04-01 by reserve85