reserve85/HoymilesZeroExport #34: "Inverter reachable" nach Sonnenuntergang

Ollipop030, 2023-04-03:
Habe gerade noch was entdeckt:

Nach Sonnenuntergang sind die Inverter logischerweise nicht mehr erreichbar, es wird aber nur ein Inverter abgefragt:

```
2023-04-03 21:27:15,726 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 21:27:26,787 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 21:27:37,846 INFO     Ahoy: Inverter 0 reachable: False
```
Morgens um 7 startet WLAN wieder, dann werden auch beide Inverter abgefragt:
```

2023-04-03 07:00:15,606 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 07:00:26,712 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 07:00:37,771 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 07:00:48,831 INFO     Ahoy: Inverter 0 reachable: True
2023-04-03 07:00:48,859 INFO     Ahoy: Inverter 1 reachable: True
2023-04-03 07:00:48,886 INFO     powermeter EMLOG: 114  Watt
2023-04-03 07:00:48,887 INFO     setting new limit to 354 Watt
```

Und das hier ist der letzte Kontakt zum Inverter von heute:

```
2023-04-03 20:08:33,083 INFO     Ahoy: Inverter 0 reachable: True
2023-04-03 20:08:33,110 INFO     Ahoy: Inverter 1 reachable: True
2023-04-03 20:08:33,134 INFO     powermeter EMLOG: 1470  Watt
2023-04-03 20:08:33,135 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1800 Watt
2023-04-03 20:08:43,220 INFO     Ahoy: Inverter 0 reachable: True
2023-04-03 20:08:43,247 INFO     Ahoy: Inverter 1 reachable: False
2023-04-03 20:08:54,382 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 20:09:05,441 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 20:09:16,502 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 20:09:27,561 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 20:09:38,621 INFO     Ahoy: Inverter 0 reachable: False
2023-04-03 20:09:49,679 INFO     Ahoy: Inverter 0 reachable: False
```

Wie du siehst wird Inverter 1 gar nicht abgefragt, wenn Inverter 0=false ist. Ist wohl nur ein Schönheitsfehler, kommt nämlich nur vor wenns dunkel ist.

reserve85, 2023-04-04:
Ja das Script regelt nur wenn alle eingetragen Inverter erreichbar sind. Von daher spare ich mir den zweiten „Erreichbar“ Call. Evtl könnte ich das noch erweitern, dass auch geregelt wird wenn nur einer erreichbar ist. Hab bisher aber noch nicht die Notwendigkeit dafür gesehen.

Ollipop030, 2023-04-04:
Das wäre vielleicht sogar besser, gerade bei verschiedenen Ausrichtungen. Bei mir geht der HM300 auch bei richtig starker Bewölkung mal offline, der HM1500 produziert aber weiter. An dem HM300 hängt noch ein altes Poly Modul, das reagiert extremer auf Verschattungen. 
Bei mir ist der HM1500 Inverter 0, dann wird auch produziert wenn Inverter 1 nicht da ist. Umgekehrt wäre dann aber problematisch.

reserve85, 2023-04-04:
Ja hab’s auf der ToDo, ist denke ich nicht viel Aufwand.

reserve85, 2023-04-04:
https://github.com/reserve85/HoymilesZeroExport/tree/dev

schau es mal an, ich kann das mit meinem einem Inverter nicht testen.

Ollipop030, 2023-04-04:
Sieht erstmal gut aus:

```
2023-04-04 22:09:43,576 INFO     Ahoy: Inverter 0 reachable: False
2023-04-04 22:09:43,602 INFO     Ahoy: Inverter 1 reachable: False
```

Ich melde mich morgen nochmal, wenn die Inverter wieder aufwachen.

Ollipop030, 2023-04-05:
Alles läuft wie gewohnt, auch nach dem Aufwachen der Inverter. 
Von mir aus ist das hier erledigt.

Wie immer, Besten Dank für deine Hilfe!

reserve85, 2023-04-05:
Top, sind die gleichzeitig aufgewacht? Die MaxWatts sollten nämlich dann individuell berechnet werden, also wenn dein 300er offline ist sollte dann 1500 eingestellt werden als Maximum.

Ollipop030, 2023-04-05:
Ja sind beide gleichzetig aufgewacht. Wetter ist hier aber seit ein paar Tagen auch hammermäßig. 

```
2023-04-05 06:53:40,789 INFO     Ahoy: Inverter 1 reachable: False
2023-04-05 06:53:52,020 INFO     Ahoy: Inverter 0 reachable: False
2023-04-05 06:53:52,051 INFO     Ahoy: Inverter 1 reachable: False
2023-04-05 06:54:03,284 INFO     Ahoy: Inverter 0 reachable: True
2023-04-05 06:54:03,324 INFO     Ahoy: Inverter 1 reachable: True
2023-04-05 06:54:03,351 INFO     powermeter EMLOG: 125  Watt
2023-04-05 06:54:03,352 INFO     setting new limit to 185 Watt
2023-04-05 06:54:03,353 INFO     Ahoy: Inverter 0: setting new limit from 0 Watt to 154 Watt
2023-04-05 06:54:04,383 INFO     Ahoy: Inverter 1: setting new limit from 0 Watt to 30 Watt
2023-04-05 06:54:15,549 INFO     intermediate meter EMLOG: 4  Watt
2023-04-05 06:54:15,550 INFO     Cut limit to 184 Watt, limit was higher than 10 percent of live-production
2023-04-05 06:54:15,783 INFO     Ahoy: Inverter 0 reachable: True
2023-04-05 06:54:15,812 INFO     Ahoy: Inverter 1 reachable: True
2023-04-05 06:54:15,838 INFO     powermeter EMLOG: 125  Watt
2023-04-05 06:54:15,839 INFO     setting new limit to 369 Watt
2023-04-05 06:54:15,839 INFO     Ahoy: Inverter 0: setting new limit from 154 Watt to 307 Watt
```

Aber beim Aufwachen wird das Limit nicht auf Maximum gesetzt.

reserve85, 2023-04-05:
Ja hab ich gesehen, schau ich mir dann noch mal an. Interessant wäre auch wenn du die nachtabschaltung mal deaktivierst ob dann auch gleichzeitig beide aktiv werden

Ollipop030, 2023-04-05:
Hab ich mal rausgenommen. Auf das Resultat müssen wir aber noch ein wenig warten. 😄

reserve85, 2023-04-05:
Nochwas, das Limit auf Maximum nach dem aufwachen ist eigentlich Quatsch? So ist das doch besser, ist ja der „Normalbetrieb“. War das bisher morgens auch auf Maximum? Normal sollte das nicht so gewesen sein, außer es passierte ein Neustart abends oder nachts?

Ollipop030, 2023-04-05:
Ne wurde nie direkt auf Max gesetzt. Dachte aber das sollte es tun, weil du das heute morgen geschrieben hattest.

reserve85, 2023-04-05:
Ja bei der Jump to Max wäre das dann so

Ollipop030, 2023-04-05:
Also Abends klappt das schonmal:

```
2023-04-05 20:13:49,495 INFO     Ahoy: Inverter 0 reachable: True
2023-04-05 20:13:49,525 INFO     Ahoy: Inverter 1 reachable: True
2023-04-05 20:13:49,549 INFO     powermeter EMLOG: 1352  Watt
2023-04-05 20:13:49,550 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1800 Watt
2023-04-05 20:13:59,630 INFO     Ahoy: Inverter 0 reachable: True
2023-04-05 20:13:59,665 INFO     Ahoy: Inverter 1 reachable: False
2023-04-05 20:13:59,691 INFO     powermeter EMLOG: 1344  Watt
2023-04-05 20:13:59,693 INFO     setting new limit to 1500 Watt
2023-04-05 20:13:59,694 INFO     Ahoy: Inverter 0: setting new limit from 1500 Watt to 1500 Watt
```

reserve85, 2023-04-05:
Ja perfekt so sollte es sein.

Ollipop030, 2023-04-06:
Morgens läuft es auch:

```
2023-04-06 06:35:01,911 INFO     Ahoy: Inverter 0 reachable: False
2023-04-06 06:35:01,948 INFO     Ahoy: Inverter 1 reachable: False
2023-04-06 06:35:13,068 INFO     Ahoy: Inverter 0 reachable: True
2023-04-06 06:35:13,100 INFO     Ahoy: Inverter 1 reachable: False
2023-04-06 06:35:13,127 INFO     powermeter EMLOG: 186  Watt
2023-04-06 06:35:13,128 INFO     Set Limit Retry Counter exceeded: Inverterlimit already at 1500 Watt
```

closed 2023-04-06 by reserve85