reserve85/HoymilesZeroExport #35: Fehler seit Update auf 1.18

Andre0ber, 2023-04-05:
Hallo,

Version 1.14 hat bei mir funktioniert. Habe heute ein Update auf 1.18 gemacht (HoymilesZeroExport_Config.ini angepasst). Jedoch bekomme ich immer folgende Meldung:

`Apr 05 14:14:52 pi1b python3[4136]: 2023-04-05 14:14:52 INFO     Error on reading ENABLE_LOG_TO_FILE, set it to DISABLED
Apr 05 14:14:52 pi1b python3[4136]: 2023-04-05 14:14:52 ERROR    File contains no section headers.
Apr 05 14:14:52 pi1b python3[4136]: file: '/home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini', line: 1
Apr 05 14:14:52 pi1b python3[4136]: '.# HoymilesZeroExport - https://github.com/reserve85/HoymilesZeroExport\n'
Apr 05 14:14:52 pi1b python3[4136]: 2023-04-05 14:14:52 INFO     Log write to file: False
Apr 05 14:14:52 pi1b python3[4136]: 2023-04-05 14:14:52 INFO     Author: Tobias Kraft / Script Version: 1.18
Apr 05 14:14:52 pi1b python3[4136]: 2023-04-05 14:14:52 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Apr 05 14:14:52 pi1b python3[4136]: Traceback (most recent call last):
Apr 05 14:14:52 pi1b python3[4136]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 351, in <module>
Apr 05 14:14:52 pi1b python3[4136]:     VERSION = config.get('VERSION', 'VERSION')
Apr 05 14:14:52 pi1b python3[4136]:   File "/usr/lib/python3.9/configparser.py", line 781, in get
Apr 05 14:14:52 pi1b python3[4136]:     d = self._unify_values(section, vars)
Apr 05 14:14:52 pi1b python3[4136]:   File "/usr/lib/python3.9/configparser.py", line 1149, in _unify_values
Apr 05 14:14:52 pi1b python3[4136]:     raise NoSectionError(section) from None
Apr 05 14:14:52 pi1b python3[4136]: configparser.NoSectionError: No section: 'VERSION'
Apr 05 14:14:52 pi1b systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Apr 05 14:14:52 pi1b systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 05 14:14:52 pi1b systemd[1]: HoymilesZeroExport.service: Consumed 4.328s CPU time.
Apr 05 14:14:52 pi1b systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 3.
Apr 05 14:14:52 pi1b systemd[1]: Stopped HoymilesZeroExport Service.
Apr 05 14:14:52 pi1b systemd[1]: HoymilesZeroExport.service: Consumed 4.328s CPU time.
Apr 05 14:14:52 pi1b systemd[1]: Started HoymilesZeroExport Service.
`

Kann allerdings nicht erkennen wieso? Nutze OpenDTU und Tasmota zum Zähler auslesen.

Danke.

reserve85, 2023-04-05:
Hi,
Du musst die config auch ersetzen oder ergänzen, bei dir fehlen noch die neuen Einträge.

Andre0ber, 2023-04-05:
Hab den Fehler gerade gefunden. Es war ein falsches Zeichen in einer Zeile von mir eingegeben worden. Sorry. Bitte schließen.

reserve85, 2023-04-05:
Super, kein ding