reserve85/HoymilesZeroExport #37: Ansteuerung zusätzlicher Verbraucher vor Limitierung

tqma1, 2023-04-07:
Mir ist da so eine Idee gekommen:
Könnte man nicht Verbraucher definieren welche zuerst zugeschalten werden bevor die Limitierung beginnt?

Beispiel:
Ich habe ein Aquarium mit 800L Inhalt wo ich ja die Energie für spätere Stunden schon speichern könnte.
Einen zusätzlichen Heizstab der 2-3 Grad höher eingestellt ist als der Standard-Heizstab an einem Tasmota Zwischenstecker einbauen.
Im Script definieren der Zusatz-Heizstab hat eine Leistung von z.Bsp. 200W und diesen entsprechend freischalten.

Würde natürlich auch für viele andere Dinge funktionieren...

Ollipop030, 2023-04-07:
Ich denke, das sollte eher eine Hausautomatisierung übernehmen. Das Projekt hier soll ja nur eine Nulleinspeisung regeln. Fürs Steckdosen schalten nehme immer Zeitschaltuhren, oder eben manuell über eine Handyapp wenn die Sonne scheint.

Ist aber nur meine Meinung, vielleicht sieht @reserve85 das ja anders?

Wie auch immer, euch ein frohes Osterfest.

reserve85, 2023-04-07:
Servus und euch auch frohe Ostern.

Das hat in meinen Augen nix mit Limitreduzierung zu tun. Du bräuchtest eine Steuerung, die den Zähler ausliest und wenn er negativ ist (bei Überproduktion) via PWM den Heizstab regelt. Gibts ja schon für Heizsysteme auch fertig zu kaufen, die dann in den WW Behälter eingeschraubt werden.

tqma1, 2023-04-11:
Grundsätzlich bin ich da ja auch bei Euch, nur wer ist dann schneller?
Ist die Limitierung schneller wird der Verbraucher nie freigegeben - daher mein Gedanke.

tqma1, 2023-04-11:
Alles klar, kann geschlossen werden.