reserve85/HoymilesZeroExport #38: Inverternamen/Temperatur im Log anzeigen

Ollipop030, 2023-04-07:
Kann man den Inverter über das Script beim Namen nennen? In Ahoy vergibt man ja Namen für die Inverter, diesen Namen könnte man ja auch im Script anzeigen lassen. Sieht ein bisschen hübscher aus als Inverter 0, Inverter 1...

Des Weiteren:
Der HM1500 wird sehr warm bei mir. Ist außen montiert, Nordseite an der Wand, auf 100% Limit komme ich aber schnell an die 45 Grad. Kann man die Temperatur mit abrufen, so dass die mit geloggt wird? Meine Idee für die Zukunft wäre dann vielleicht eine Art Sicherheitsfunktion, das bei Überschreiten einer bestimmten Temperatur die Leistung gedrosselt wird. Das funktioniert auch tatsächlich, sobald ich manuell die Leistung limitiere wird der WR auch wieder kühler. Nur hätte ich gern die Möglichkeit überhaupt die Temperatur mitzuschreiben, die hohen Außentemperaturen kommen ja erst noch, wobei der HM1500 glaub ich bis 60 Grad ab kann.

BTW:
Eigentlich wollte ich die Log Dateien nur 5 Tage behalten, ich habe aber jetzt schon 8 Files. Kann es sein, dass der Counter dafür wieder neu startet, sobald auch der Dienst neu gestartet wird?

reserve85, 2023-04-07:
Servus, den Namen könnte ich aus Ahoy / openDTU auslesen. Schau ich mir mal an. Die Temperatur dann ebenfalls, wobei ich das nur optional machen würde, da extra Abfrage. Ich nehm es mal mit wenn ich Zeit habe. 

Bzgl Log müsste ich das mal testen, kann aber durchaus sein dass das so ist. Wenn das die benutzte Library nicht hergibt dann lass ich es auch so, selber will ich da nix programmieren.

Ollipop030, 2023-04-07:
> Allerdings hat der Hoymiles selbst eine „Notdrosselfunktion“ bei Überhitzung. Habe ich irgendwo mal gelesen.

Das stimmt. Aber ich dachte, er schaltet dann komplett ab. Naja, wie auch immer, erstmal das lange Wochenende genießen.

reserve85, 2023-04-08:
https://github.com/reserve85/HoymilesZeroExport/tree/dev

Ollipop030, 2023-04-08:
Hi, grad installiert bekomme noch folgendes im Log:

```
Apr 08 16:38:49 raspberrypi python3[13615]: 2023-04-08 16:38:49 INFO     Ahoy: Inverter "HM-300" reachable: True
Apr 08 16:38:49 raspberrypi python3[13615]: [74B blob data]
Apr 08 16:38:49 raspberrypi python3[13615]: [73B blob data]
```

Wird wahrscheinlich die Temperatur Abfrage sein?

reserve85, 2023-04-08:
Äh im logfile oder im Live log? Bei mir kommt das nicht?

Ollipop030, 2023-04-08:
Komisch. Nur im Live Log. Im Logfile steht die Temperatur.

reserve85, 2023-04-08:
Hab eine Vermutung: kannst du mal im Code die „°C“ in „degC“ ändern? Ggf. kann das Terminal das bei dir nicht ausgeben

Ollipop030, 2023-04-08:
Ja kommt was:

`Ahoy: Inverter "HM-1500" temperature: 25.0 degC`

reserve85, 2023-04-08:
Ok dann ändere ich das noch, danke

Ollipop030, 2023-04-08:
Unter Windows funktioniert das übrigens in der ursprünglichen Form. Anscheinend mag Linux das Gradzeichen nicht.

reserve85, 2023-04-08:
Ja komisch, bei mir im raspi ging es auch, debian system

Ollipop030, 2023-04-08:
Bin auf Raspian Buster, aber basiert ja auf Debian. Eigenartig.

reserve85, 2023-04-08:
OK, habs in der MAIN angepasst. Die Logfiles musste ich als Default-Benamung lassen, dann werden die automatisch gelöscht.

Ollipop030, 2023-04-09:
Vielen Dank. Das Gradzeichen wird wohl bei mir nicht funktionieren, da die Shell bei mir deutsch ausgibt. Wird wohl ein anderer Zeichensatz installiert sein. War damals ein offizielles Raspi Image, nur eben auf deusch.

closed 2023-04-11 by reserve85