reserve85/HoymilesZeroExport #41: Service startet nicht

costa2, 2023-04-09:
Mein System: 2 HM-600, Ahoy und Hichi Lesekopf.
Im HoymilesZeroExport.py ist scheinbar ein Fehler.
Kann es daran liegen?

Apr 09 14:41:29 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Apr 09 14:41:29 raspberrypi python3[7912]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 69
Apr 09 14:41:29 raspberrypi python3[7912]: url=f"http://{OPENDTU_IP}/api/limit/config"
Apr 09 14:41:29 raspberrypi python3[7912]: ^
Apr 09 14:41:29 raspberrypi python3[7912]: SyntaxError: invalid syntax
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service: main process exited, code=exited, status=1/FAILURE
Apr 09 14:41:29 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 14:41:29 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Apr 09 14:41:29 raspberrypi python3[7918]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 69
Apr 09 14:41:29 raspberrypi python3[7918]: url=f"http://{OPENDTU_IP}/api/limit/config"
Apr 09 14:41:29 raspberrypi python3[7918]: ^
Apr 09 14:41:29 raspberrypi python3[7918]: SyntaxError: invalid syntax
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service: main process exited, code=exited, status=1/FAILURE
Apr 09 14:41:29 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 14:41:29 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Apr 09 14:41:29 raspberrypi python3[7920]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 69
Apr 09 14:41:29 raspberrypi python3[7920]: url=f"http://{OPENDTU_IP}/api/limit/config"
Apr 09 14:41:29 raspberrypi python3[7920]: ^
Apr 09 14:41:29 raspberrypi python3[7920]: SyntaxError: invalid syntax
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service: main process exited, code=exited, status=1/FAILURE
Apr 09 14:41:29 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 14:41:29 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Apr 09 14:41:29 raspberrypi python3[7923]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 69
Apr 09 14:41:29 raspberrypi python3[7923]: url=f"http://{OPENDTU_IP}/api/limit/config"
Apr 09 14:41:29 raspberrypi python3[7923]: ^
Apr 09 14:41:29 raspberrypi python3[7923]: SyntaxError: invalid syntax
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service: main process exited, code=exited, status=1/FAILURE
Apr 09 14:41:29 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.
Apr 09 14:41:29 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 14:41:29 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 14:41:29 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Apr 09 14:41:30 raspberrypi python3[7925]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 69
Apr 09 14:41:30 raspberrypi python3[7925]: url=f"http://{OPENDTU_IP}/api/limit/config"
Apr 09 14:41:30 raspberrypi python3[7925]: ^
Apr 09 14:41:30 raspberrypi python3[7925]: SyntaxError: invalid syntax
Apr 09 14:41:30 raspberrypi systemd[1]: HoymilesZeroExport.service: main process exited, code=exited, status=1/FAILURE
Apr 09 14:41:30 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.
Apr 09 14:41:30 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 14:41:30 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 14:41:30 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 14:41:30 raspberrypi systemd[1]: HoymilesZeroExport.service start request repeated too quickly, refusing to start.
Apr 09 14:41:30 raspberrypi systemd[1]: Failed to start HoymilesZeroExport Service.
Apr 09 14:41:30 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.

reserve85, 2023-04-09:
Hallo,
das kann ich irgendwie nicht wirklich nachvollziehen - bei mir funktioniert der code, unter Windows und unter Linux mit Python 3.
Kannst du evtl. testweise in der Zeile 69 die " durch ' ersetzen?
also:
`url=f'http://{OPENDTU_IP}/api/limit/config'`

costa2, 2023-04-09:
Stimmt, Python 3.4.2
Dann werde ich es Morgen mal updaten.
Danke erst einmal, ich melde mich dann wieder.

costa2, 2023-04-09:
An Python lag es nicht, es ist nun 3.9.9

Apr 09 22:21:05 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 22:21:05 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 22:21:05 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 22:21:05 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Apr 09 22:21:05 raspberrypi python3[7496]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 69
Apr 09 22:21:05 raspberrypi python3[7496]: url=f"http://{OPENDTU_IP}/api/limit/config"
Apr 09 22:21:05 raspberrypi python3[7496]: ^
Apr 09 22:21:05 raspberrypi python3[7496]: SyntaxError: invalid syntax
Apr 09 22:21:05 raspberrypi systemd[1]: HoymilesZeroExport.service: main process exited, code=exited, status=1/FAILURE
Apr 09 22:21:05 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.
Apr 09 22:21:05 raspberrypi systemd[1]: HoymilesZeroExport.service holdoff time over, scheduling restart.
Apr 09 22:21:05 raspberrypi systemd[1]: Stopping HoymilesZeroExport Service...
Apr 09 22:21:05 raspberrypi systemd[1]: Starting HoymilesZeroExport Service...
Apr 09 22:21:05 raspberrypi systemd[1]: HoymilesZeroExport.service start request repeated too quickly, refusing to start.
Apr 09 22:21:05 raspberrypi systemd[1]: Failed to start HoymilesZeroExport Service.
Apr 09 22:21:05 raspberrypi systemd[1]: Unit HoymilesZeroExport.service entered failed state.

reserve85, 2023-04-09:
Bist du sicher, dass der Dienst auch mit der neuen Python Version ausgeführt wird? Ich glaube das muss man explizit noch aktivieren.
Kommt bei python3 --version jetzt 3.9.9?

costa2, 2023-04-09:
pi@raspberrypi:/usr/bin $ python --version
Python 3.9.9

reserve85, 2023-04-09:
was ist wenn du mit dem Befehl hier mal probierst das Script zu starten?
`python3.9  HoymilesZeroExport/HoymilesZeroExport.py`

costa2, 2023-04-09:
pi@raspberrypi:/ $ sudo python3.9: can't open file '/home/pi/HoymilesZeroExport/HoymilesZeroExport/HoymilesZeroExport.py': [Errno 2] No such file or directory

Es ist ja Python 3.9.9, aber auch das geht nicht.

pi@raspberrypi:/ $ sudo python3.9.9 HoymilesZeroExport/HoymilesZeroExport.py
sudo: python3.9.9: Kommando nicht gefunden

reserve85, 2023-04-09:
Oben ist der Pfad falsch, da findet er die Datei nicht.

`python3.9 /home/pi/HoymilesZeroExport/HoymilesZeroExport.py`
Das müsste korrekt sein

costa2, 2023-04-10:
Traceback (most recent call last):
  File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 20, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

reserve85, 2023-04-10:
`pip3 install requests`

costa2, 2023-04-10:
Leider funktioniert es nicht, irgendwas ist mit meinem Fhem nicht i.O..
Die selbe Fehlermeldung bekomme ich auch, wenn ich ein Upgrade auf Bullseye machen möchte.

 sudo apt-get install python3-requests
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
E: Das Paket fhem muss neu installiert werden, es kann jedoch kein Archiv dafür gefunden werden.

Nun habe ich folgendes dazu gefunden:

𝗲𝘀 𝘀𝗰𝗵𝗲𝗶𝗻𝘁 𝗮𝗹𝘀 𝘄ü𝗿𝗱𝗲 𝗲𝗶𝗻 𝗮𝗹𝘁𝗲𝘀 𝗽𝗮𝗸𝗲𝘁 𝗻𝗼𝗰𝗵 𝘄𝗮𝘀 𝗯𝗲𝗶 𝗱𝗶𝗿 𝗯𝗹𝗼𝗰𝗸𝗶𝗲𝗿𝗲𝗻: 𝗶𝗰𝗵 𝘄ü𝗿𝗱𝗲 𝗱𝗶𝗿 𝗳ü𝗿 𝗮𝗹𝗹𝗲 𝗳𝗮𝗹𝘀𝗰𝗵𝗲𝗻 𝘂𝗻𝗱 𝗻𝗶𝗰𝗵𝘁 𝘇𝘂𝗺 𝗼𝗳𝗳𝗶𝘇𝗶𝗲𝗹𝗹𝗲𝗻 𝗱𝗲𝗯𝗶𝗮𝗻 𝘁𝗿𝗲𝗲 𝗴𝗲𝗵ö𝗿𝗲𝗻𝗱𝗲 𝗽𝗮𝗸𝗲𝘁𝗲 𝗲𝗺𝗽𝗳𝗲𝗵𝗹𝗲𝗻 𝗱𝗶𝗲𝘀𝗲 𝗺𝗶𝘁 𝗱𝗽𝗸𝗴 --𝗽𝘂𝗿𝗴𝗲 --𝗳𝗼𝗿𝗰𝗲-𝗮𝗹𝗹 𝘇𝘂 𝗹ö𝘀𝗰𝗵𝗲𝗻. 𝗺𝗶𝘁 𝗱𝗲𝗿 𝗼𝗽𝘁𝗶𝗼𝗻 --𝗳𝗼𝗿𝗰𝗲-𝗿𝗲𝗺𝗼𝘃𝗲-𝗿𝗲𝗶𝗻𝘀𝘁𝗿𝗲𝗾 𝘀𝗼𝗹𝗹𝘁𝗲 𝘀𝗶𝗰𝗵 𝗱𝗮𝘀 𝗼𝗯𝗲𝗻 𝗴𝗲𝘀𝗰𝗵𝗶𝗹𝗱𝗲𝗿𝘁𝗲 𝗣𝗿𝗼𝗯𝗹𝗲𝗺 𝗹ö𝘀𝗲𝗻 𝗹𝗮𝘀𝘀𝗲𝗻.

Ich befürchte aber, dass ich meine Fhem Installation damit schrotte.

reserve85, 2023-04-10:
Ja irgendwas ist da total faul.
Du kannst noch das hier probieren:
python3.9 -m pip3 install requests
Oder
python3.9 -m pip install requests

costa2, 2023-04-10:
Nun habe ich das Script auf einem zweiten Raspi installiert, welcher nur als Bluetooth Repeater genutzt wird, da läuft das Script, aaaber:

Apr 10 16:59:27 raspi2 python3[15109]: 2023-04-10 16:59:27 ERROR    Exception at GetPowermeterWatts
Apr 10 16:59:27 raspi2 python3[15109]: 2023-04-10 16:59:27 ERROR    'SM1'

Der Hichi hat das MQTT Topic SM1.
In der ini ist es so eingestellt:

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = true

TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = 192.168.0.86
# the following three constants describes how to navigate through the Tasmota-JSON
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SM1
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = SM_16_7_0
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
TASMOTA_JSON_POWER_INPUT_MQTT_LABEL =
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
TASMOTA_JSON_POWER_OUTPUT_MQTT_LABEL =

Was fehlt da?

reserve85, 2023-04-10:
Gib mal 
`http://192.168.0.86/cm?cmnd=status%2010`
Im Browser ein und poste mal das was zurück kommt

costa2, 2023-04-10:
// 20230410173233
// http://192.168.0.86/cm?cmnd=status%2010

{
  "StatusSNS": {
    "Time": "2023-04-10T17:32:10",
    "SM": {
      "1_8_0": 143.54,
      "2_8_0": 35.45,
      "16_7_0": -53.00,
      "36_7_0": 27.35,
      "56_7_0": 179.22,
      "76_7_0": -259.57,
      "32_7_0": 225.7,
      "52_7_0": 227.6,
      "72_7_0": 228.0,
      "96_1_0": "1EBZ0101985476"
    }
  }
}

reserve85, 2023-04-10:
Dann musst du einstellen:
```
TASMOTA_JSON_STATUS = StatusSNS
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SM
TASMOTA_JSON_POWER_MQTT_LABEL = 16_7_0
```

costa2, 2023-04-11:
> Dann musst du einstellen:
> 
> ```
> TASMOTA_JSON_STATUS = StatusSNS
> TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SM
> TASMOTA_JSON_POWER_MQTT_LABEL = 16_7_0
> ```

Vielen Dank für Deine Bemühungen, jetzt läuft es.
Nun muss ich mich nur noch mit dem Feintuning beschäftigen.

reserve85, 2023-04-11:
Super, dass es läuft und keine ahnung was mit dem Python auf deinem anderen Raspi los ist.

closed 2023-04-11 by reserve85