reserve85/HoymilesZeroExport #42: Hichi IR Lesekopf über USB an Raspberry Pi

raffy1234, 2023-04-09:
Ich habe den Lesekopf über USB direkt am Raspberry Pi stecken und über ioBroker sehe ich auch die Werte des Zählers.
Wie muss ich denn die Config.ini anpassen, um die Nulleinspeisung zu realisieren?

reserve85, 2023-04-09:
Hi,
ist so out of the Box nicht vorgesehen direkt auf USB zuzugreifen. 
Kommst du über eine REST-API an die aktuellen Zählerwerte von ioBroker? Sollte laut kurzer Recherche irgendwie funktionieren, schau dir mal https://github.com/ioBroker/ioBroker.simple-api an.

Wenn das funktioniert dann kann ich die Abfrage-URL in den Code übernehmen. Du müsstest mich da allerdings unterstützen und mir den JSON-String geben sowie die URL die aufgerufen werden muss. Ich würde das dann irgendwie variabel einbinden falls möglich.

raffy1234, 2023-04-09:
Hi, 
habe es grad probiert. Auf diesem Weg komme ich leider nicht an die aktuellen Werte. Scheint also nicht zu funktionieren, schade.

reserve85, 2023-04-09:
Doch, das muss mit dem Plugin funktionieren. Wahrscheinlich musst du das erst installieren. Dann kommst du an alle Werte.

Gibt das hier eine Rückgabe im Browser?: http://ipaddress:8087/help

raffy1234, 2023-04-09:
Gibt leider keine Rückgabe.

D3R-ST3FAN, 2023-04-13:
Danke für den Hinweis mit dem Adapter.
Nach Hinzufügen des SimpleAPI Adapters (auf Deutsch: Einfache RESTful API) läuft es mit den Standardeinstellungen.

[image]

Ein CURL gibt auch direkt den Wert aus.

z.B.

`curl http://10.xxx.xxx.71:8087/getPlainValue/alias.0.Zaehler.Zaehler_Netzbezug
435`

Kannst Du das nicht für alle Zähler einbauen? Also Quasi eine URL die den jeweiligen Wert zurückgibt.
Das sollte doch ziemlich flexibel sein.

reserve85, 2023-04-13:
Ja geht halt nicht immer, aber das würde ich dann als Universallösung so einbauen.
Hast du es schon laufen?

D3R-ST3FAN, 2023-04-13:
Ich wollte mir das am WE mal genauer ansehen.

Script läuft im Moment nicht so gut, da ich zu Testzwecken mehrere WR habe, die aber nicht komplett mit Modulen bestückt sind (1x1500er mit 2x Modul, 1x1500er mit 1x Modul, 1x400er). Außerdem ist die Sonne nicht sehr beständig im Moment.

Mein Setup:
Ich einen IR Leser am Zähler des VNB via Tasmota und mehrere Zwischenzähler, die ich per RS485/Modbus anbinden möchte. (Ich hab im Moment aus historischen Gründen einen ziemlichen Zoo, Volkszähler, FHEM, Cerbo....)

Da der Tasmota ja schon direkt bei Werteänderung per MQTT an den IoBRoker Daten schiebt, wollte ich das arme Kerlchen nicht unnötig abfragen. 
Der Modbus Zähler vor den WR wird auch an IoBroker liefern. (Tut er aber im Moment noch nicht.)
Den "Energiebedarf" Rechne ich trotzdem im IoBroker aus (Netzbezug + Summe PV Erzeugung aus OpenDTU).

Übrigens gibt es "getPlainValue"  - ohne parsen- scheinbar nur für Einzelwerte.

Man kann auch via "getBulk" mehrere Werte in einer Abfrage auslesen. Dann muss man wieder parsen.

`curl http://10.xxx.xxx.71:8087/getBulk/alias.0.Zaehler.Zaehler_Netzbezug,alias.0.Zaehler.Zaehler_Energiebedarf`
`[{"id":"alias.0.Zaehler.Zaehler_Netzbezug","val":340,"ts":1681379835195,"ack":true},{"id":"alias.0.Zaehler.Zaehler_Energiebedarf","val":1258,"ts":1681379835195,"ack":true}]`

Macht ja vielleicht auch Sinn, um die Anzahl der Abfragen zu reduzieren.

D3R-ST3FAN, 2023-04-13:
Da hab ich wohl quatsch geschrieben.

kann sehr wohl mehrere Werte

` curl http://10.xxx.xxx.71:8087/getPlainValue/alias.0.Zaehler.Zaehler_Netzbezug,alias.0.Zaehler.Zaehler_Energiebedarf`

ergibt
`450
729`

Es würde also eine Basis URL reichen und dann die jeweiligen Objektkbezeichner.

reserve85, 2023-04-14:
Moin,
ich brauche ja nur einen Wert und das ist die aktuelle Leistung vom Hauptzähler und optional noch die aktuelle Gesamtleistung der Inverter (intermediate meter). 

Bei dem Zwischenzähler wird vom Script aktuell nur ein Zähler unterstützt (das könnte ich in Zukunft noch ausbauen, aber da wirds für ein Script ganz schnell kompliziert), von daher ist das ja ganz gut wenn du alles über den Broker zusammenführst. 

Am Ende müsste ich also jeweils einen Wert abfragen für GridUsage(W) und optional SolarPower(W).
Ich mache das in Python ohne CURL Aufruf, sondern über Requests. Daher würde ich dann über den GetBulk Aufruf gehen.

reserve85, 2023-04-14:
Habe es mal in den Code übernommen, kannst du das testen?
https://github.com/reserve85/HoymilesZeroExport/tree/dev_iobroker

D3R-ST3FAN, 2023-04-14:
Hey. danke,

scheint aber nicht 100Pro zu klappen.

"intermediate meter" wird erkannt, mit dem "Haupzähler" schein was krumm zu sein.

`Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     Log write to file: True
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     Author: Tobias Kraft / Script Version: 1.24
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     read config file: /opt/HoymilesZeroExport-dev_iobroker/HoymilesZeroExport_Config.ini
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     Config file V 1.24
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     ---Init---
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "01-HM1500-01-Dach" / serial number "1161XXXXXXXX" / temperature 11.4 degC
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "01-HM1500-01-Dach" / serial number "1161XXXXXXXX" / temperature 11.4 degC
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "02-HM1500-02-Unten" / serial number "1161XXXXXXXX" / temperature 13.1 degC
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "01-HM1500-01-Dach" / serial number "1161XXXXXXXX" / temperature 11.4 degC
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "02-HM1500-02-Unten" / serial number "116XXXXXXXX" / temperature 13.1 degC
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "03-ZaunPV400" / serial number "112XXXXXXXX" / temperature 14.4 degC
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     intermediate meter IOBROKER: 126  Watt
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     setting new limit to 3400 Watt
Apr 14 11:18:02 zeroexport python3[14069]: 2023-04-14 11:18:02 INFO     OpenDTU: Inverter "01-HM1500-01-Dach": setting new limit from 0 Watt to 1500 Watt
Apr 14 11:18:04 zeroexport python3[14069]: 2023-04-14 11:18:04 INFO     OpenDTU: Inverter "02-HM1500-02-Unten": setting new limit from 0 Watt to 1500 Watt
Apr 14 11:18:06 zeroexport python3[14069]: 2023-04-14 11:18:06 INFO     OpenDTU: Inverter "03-ZaunPV400": setting new limit from 0 Watt to 400 Watt
Apr 14 11:18:11 zeroexport python3[14069]: 2023-04-14 11:18:11 ERROR    Exception at GetPowermeterWatts
Apr 14 11:18:11 zeroexport python3[14069]: 2023-04-14 11:18:11 ERROR    local variable 'input' referenced before assignment
Apr 14 11:18:11 zeroexport python3[14069]: 2023-04-14 11:18:11 ERROR    local variable 'input' referenced before assignment
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 INFO     ---Start Zero Export---
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 INFO     OpenDTU: Inverter "01-HM1500-01-Dach" reachable: True
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 INFO     OpenDTU: Inverter "02-HM1500-02-Unten" reachable: True
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 INFO     OpenDTU: Inverter "03-ZaunPV400" reachable: True
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 ERROR    Exception at GetPowermeterWatts
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 ERROR    local variable 'input' referenced before assignment
Apr 14 11:18:26 zeroexport python3[14069]: 2023-04-14 11:18:26 ERROR    local variable 'input' referenced before assignment`

reserve85, 2023-04-14:
ok, kannst du mal noch die sektion [IOBROKER] aus der Ini posten?
Wie ich das sehe hast du IOBROKER_POWER_CALCULATE auf true stehen?

D3R-ST3FAN, 2023-04-14:
irgendwie komme ich mit dem github markup nicht klar

Alias, Port und IP stimmen aber.

mit curl bekomme ich auch den Wert.

`curl http://10.1.120.71:8087/getPlainValue/alias.0.Zaehler.Zaehler_Netzbezug`

`602`

reserve85, 2023-04-14:
in [SELECT_POWERMETER] steht USE_IOBROKER auf true und alle anderen auf false?

D3R-ST3FAN, 2023-04-14:
> in [SELECT_POWERMETER] steht USE_IOBROKER auf true und alle anderen auf false?

ja

reserve85, 2023-04-14:
ah ja, fehler gefunden
kannst du diese zeile ändern:

`IOBROKER_POWER_CALCULATE = config.get('IOBROKER', 'IOBROKER_POWER_CALCULATE')`
in
`IOBROKER_POWER_CALCULATE = config.getboolean('IOBROKER', 'IOBROKER_POWER_CALCULATE')`

D3R-ST3FAN, 2023-04-14:
ja, das hat es gebracht!

Der PowerMeter Wert wird jetzt ausgelesen.

`powermeter IOBROKER: 526  Watt`

Die eigentliche Regelung kann ich Wetterbedingt leider nicht testen.

Zwischenzähler und Hauptzähler können also aus IoBroker gelesen werden TOP!

Apropos Zwischenzähler, da bei mir nicht alle WR an einer "Leitung" hängen gibt es mehrere.
Nur falls du dich fragst warum macht man sowas. 
Die müssten man dann im Zweifel addieren oder als virtuellen Zähler im IoBroker anlegen und dort addieren.

reserve85, 2023-04-14:
ja genau, die musst du dann im ioBroker addieren und als einen Gesamtwert bereitstellen.

Dann nehm ich das so nachher mit in die MAIN

raffy1234, 2023-04-15:
> > Gibt leider keine Rückgabe.
> 
> Wie hast Du den den simple API Adapter installiert und konfiguriert?

> > Gibt leider keine Rückgabe.
> 
> Wie hast Du den den simple API Adapter installiert und konfiguriert?

Das wird bei mir angezeigt, leider kann ich damit überhaupt nichts anfangen.

{ "getPlainValue": "http://192.168.178.45:8087/getPlainValue/stateID", "get": "http://192.168.178.45:8087/get/stateID/?prettyPrint", "getBulk": "http://192.168.178.45:8087/getBulk/stateID1,stateID2/?prettyPrint", "set": "http://192.168.178.45:8087/set/stateID?value=1&prettyPrint", "toggle": "http://192.168.178.45:8087/toggle/stateID&prettyPrint", "setBulk": "http://192.168.178.45:8087/setBulk?stateID1=0.7&stateID2=0&prettyPrint", "setValueFromBody": "http://192.168.178.45:8087/setValueFromBody?stateID1", "objects": "http://192.168.178.45:8087/objects?pattern=system.adapter.admin.0*&prettyPrint", "states": "http://192.168.178.45:8087/states?pattern=system.adapter.admin.0*&prettyPrint", "search": "http://192.168.178.45:8087/search?pattern=system.adapter.admin.0*&prettyPrint", "query": "http://192.168.178.45:8087/query/stateID1,stateID2/?dateFrom=2019-06-06T12:00:00.000Z&dateTo=2019-06-06T12:00:00.000Z&noHistory=false&aggregate=minmax&count=3000&prettyPrint" }

D3R-ST3FAN, 2023-04-15:
Na dann läuft es doch offenbar schon.

Du musst dir aus der Objektliste die stateid der Leistung in W raussuchen.

Bei mir wäre das z.B.

`sonoff.0.tasmopwr.PWR_power-live`

wird bei Dir anders sein, vielleicht ja ein Alias.

In die URL 

http://192.168.178.45:8087/getPlainValue/stateID

Setzt Du zum testen mal dein StateID ein.

Sieht dann so aus:

http://192.168.178.45:8087/getPlainValue/hier.deine.stateid.power

hier.deine.stateid.power  musst du natürlich anpassen.

Die URL testest Du (rufst sie einfach im Browser auf). 
Wenn der Wert Leistungswert kommt hast Du alle Infos für die Konfigurationsdatei.

HoymilesZeroExport_Config.ini

in der [IOBROKER] Sektion

IOBROKER_IP = 192.168.178.45
IOBROKER_PORT = 8087
IOBROKER_CURRENT_POWER_ALIAS = hier.deine.stateid.power

raffy1234, 2023-04-16:
Super, es läuft. Jetzt muss nur och besseres Wetter werden ;-) 
Vielen Dank für Eure Hilfe und diese tolle Lösung.

closed 2023-04-20 by reserve85

raffy1234, 2024-03-27:
Ich bekomme es wieder mal nicht hin, was mache ich verkehrt?

Mar 27 00:05:21 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:24 raspberrypi python3[819]: Traceback (most recent call last):
Mar 27 00:05:24 raspberrypi python3[819]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:24 raspberrypi python3[819]:     from packaging import version
Mar 27 00:05:24 raspberrypi python3[819]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:24 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:24 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:24 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 1.
Mar 27 00:05:24 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:24 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:25 raspberrypi python3[834]: Traceback (most recent call last):
Mar 27 00:05:25 raspberrypi python3[834]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:25 raspberrypi python3[834]:     from packaging import version
Mar 27 00:05:25 raspberrypi python3[834]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:25 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:25 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:25 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 2.
Mar 27 00:05:25 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:25 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:26 raspberrypi python3[835]: Traceback (most recent call last):
Mar 27 00:05:26 raspberrypi python3[835]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:26 raspberrypi python3[835]:     from packaging import version
Mar 27 00:05:26 raspberrypi python3[835]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:26 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:26 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:26 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 3.
Mar 27 00:05:26 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:26 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:26 raspberrypi python3[837]: Traceback (most recent call last):
Mar 27 00:05:26 raspberrypi python3[837]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:26 raspberrypi python3[837]:     from packaging import version
Mar 27 00:05:26 raspberrypi python3[837]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Mar 27 00:05:27 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:27 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:27 raspberrypi python3[839]: Traceback (most recent call last):
Mar 27 00:05:27 raspberrypi python3[839]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:27 raspberrypi python3[839]:     from packaging import version
Mar 27 00:05:27 raspberrypi python3[839]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:28 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Mar 27 00:05:28 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:28 raspberrypi systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Mar 27 00:05:28 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:28 raspberrypi systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:43 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:44 raspberrypi python3[3052]: Traceback (most recent call last):
Mar 27 20:35:44 raspberrypi python3[3052]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 20:35:44 raspberrypi python3[3052]:     from packaging import version
Mar 27 20:35:44 raspberrypi python3[3052]: ModuleNotFoundError: No module named 'packaging'
[... cut ...]
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Mar 27 22:01:55 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 22:01:55 raspberrypi systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.