reserve85/HoymilesZeroExport #43: ERROR    Exception at GetHoymilesActualPower

Leviathan09, 2023-04-14:
Moin @reserve85 ,
habe seit ich gestern von V1.19 auf V1.21 geupdatet habe ziemlich viele ERROR Meldungen im Log.
Habe soweit nur die HoymilesZeroExport.py  und die HoymilesZeroExport_Config.ini ausgetauscht und in der .ini meine Werte aus der alten übernommen.

```
2023-04-13 08:19:58,830 ERROR    Exception at GetHoymilesActualPower
2023-04-13 08:19:58,831 ERROR    Expecting value: line 1 column 1 (char 0)
2023-04-13 10:14:40,652 ERROR    Exception at GetHoymilesActualPower
2023-04-13 10:14:40,653 ERROR    Expecting value: line 1 column 1 (char 0)
2023-04-13 10:15:01,095 ERROR    Exception at GetHoymilesActualPower
2023-04-13 10:15:01,096 ERROR    Expecting value: line 1 column 1 (char 0)
2023-04-13 10:15:21,526 ERROR    Exception at GetHoymilesActualPower
2023-04-13 10:15:21,526 ERROR    Expecting value: line 1 column 1 (char 0)
2023-04-13 10:15:42,052 ERROR    Exception at GetHoymilesActualPower
2023-04-13 10:15:42,052 ERROR    Expecting value: line 1 column 1 (char 0)
2023-04-13 10:16:13,020 ERROR    Exception at GetHoymilesActualPower
2023-04-13 10:16:13,020 ERROR    Expecting value: line 1 column 1 (char 0)
2023-04-13 10:16:33,610 ERROR    Exception at GetHoymilesActualPower
```

reserve85, 2023-04-14:
Servus, kannst du die INI posten oder mir schicken? Hast du einen Zwischenzähler definiert (Intermediate)?

Leviathan09, 2023-04-14:
Mein Fehler, ich habe die ini jetzt beim pasten nochmal gecheckt und gesehen das ich bei Tasmota nen true drin vergessen hatte aber kein Tasmota nutze und beim Intermediate den normalen 1PM statt 1PM PLUS ausgewählt.

closed 2023-04-14 by Leviathan09