reserve85/HoymilesZeroExport #45: Batteriebetrieb

reserve85, 2023-04-18:
Hallo,

Ich habe das Script jetzt seit gestern mit mit ahoydtu und einem HM-1500 auf 48V Batterie am laufen.
Die Funktion ist wirklich sehr ordentlich.
Alles andere, das Ich vorher probiert habe war nicht so gut, oder hat überhaupt nicht funktioniert.
Als absoluter Neuling auf diesem Gebiet, eine Frage an die Profis, wäre es möglich den vom Inverter für die Strings gemessen Spannungswert für ein Änderung des Limit bzw. die Abschaltung des Inverters zu verwenden?
Zum Beispiel möchte Ich ab 48V am String das Powerlimit auf 300W setzen und dann vielleicht ab 47V den Inverter ausschalten.
Könnte man so etwas integrieren?
Und wenn ja, wie in etwa müsste Ich da vorgehen?

mfg Dirk

_Originally posted by @feeti1 in https://github.com/reserve85/HoymilesZeroExport/issues/28#issuecomment-1511724697_ _edit reserve85: ich lagere das mal aus.-

reserve85, 2023-04-18:
Ja wäre theoretisch möglich.
Wären das dann diese Werte?
[image]
Wenn ein Eingang auf 48V steht -> max. inverter limit auf 300?
Wenn ein Eingang auf 47V steht -> inverter ausschalten?

feeti1, 2023-04-18:
> [https://user-images.githubusercontent.com/111107925/232700732-15a9d42d-a4f1-42d3-9097-247e6a4cc2c1.jpg][2023-04-18 09_14_52-Live - Chromium][https://user-images.githubusercontent.com/111107925/232700732-15a9d42d-a4f1-42d3-9097-247e6a4cc2c1.jpg]

Hallo,

danke für deine Nachricht.
Es ist schon Mal ermutigend, das man so etwas umsetzen könnte.
Für mich ist es sehr spannend, mich damit auseinander zu setzen.
Allerdings habe ich bisher noch gar keine Erfahrungen mit Python und Mqtt gemacht.
Die von Dir markierten Werte sind genau die, auf die ich mich beziehe.
Genau darüber möchte ich gerne den Inverter ab 48V zuerst Herunterregeln, und dann ab 47V abschalten.
Da es sich um normale AGM Batterien handelt.
Sonst könnte sich die Zyklenzahl zu stark reduzieren.

mfg Dirk
[cid:eu.faircode.email.849]

feeti1, 2023-04-18:
[image]
Mal sehen ob es jetzt mit dem Screenshot klappt.

mfg Dirk

reserve85, 2023-04-18:
Ja ich schau mir das mal an, komme aber wohl erst nächste Woche irgendwann dazu. Im Prinzip müsste wahrscheinlich nur die Min-Spannung aller Panels ausgelesen werden und dann die Max-Watts begrenzt werden wenn unter Grenzwert 1 bzw. der Inverter ausgeschaltet werden wenn unter Grenzwert 2. Ich würde das allerdings ungern in mein Repository übernehmen, vielleicht wäre es sinnvoll wenn du davon ein fork erstellst?

feeti1, 2023-04-18:
Hallo,

danke das Du dir darüber Gedanken machen magst.  Ich werde versuchen mich da auch reinzudenken, allerdings waren Scripte für mich bisher copy and paste.
Das empfinde Ich an dieser Stelle als eher unbefriedigend. Unter diesen Voraussetzungen ein Fork zu eröffnen, ist vielleicht
eine sehr sportliche Sichtweise. ; )
Wenn es hilft, lasse ich auch gerne ein Kasten Bier springen.

lg

reserve85, 2023-04-19:
noch eine Frage: Was ist wenn der Inverter ausgeschaltet ist? Wird die Spannung weitergemessen? Wann wird wieder eingeschaltet?

feeti1, 2023-04-19:
Der Inverter gibt auch nach Abschaltung für jeden String die Werte für die Spannung weiter aus.
Da sich die Batterie nach Entladung erst etwas erholen muss, sollte ab 52V Eingangsspannung am Inverter wieder eingeschaltet werden.

reserve85, 2023-04-21:
@feeti1: schaust du dir das mal an? https://github.com/reserve85/HoymilesZeroExport/tree/dev_battery_powered
@peff74: das könnte dann auch für dich interessant sein.

ich kann das nicht mit Akku testen, daher bin ich auf euch angewiesen 👍

reserve85, 2023-04-21:
> Sag aber, warum machst Du Dir so einen Aufwand mit dem Skript für Nulleinspeisung, wenn Du keinen Akku nutzt?

Weil mir der Netzbetreiber meinen Zähler austauscht, ich dafür auch noch mehr zahlen muss und ausschließlich er davon provitiert. Ohne mich.

peff74, 2023-04-21:
Ansonsten scheint hier schon alle implementiert zu sein:
https://github.com/helgeerbe/OpenDTU-OnBattery

reserve85, 2023-04-21:
Jupp, das ist auf ESP32 Basis - wird wahrscheinlich sogar besser sein weil getestet und erprobt :)

feeti1, 2023-04-21:
Hallo,

das habe Ich nicht besonders gut ans laufen bekommen.
Diese Lösung hat Out of the Box wirklich gut funktioniert.
Und wenn jetzt die Ab.- und Zuschaltung über das Script funktioniert, brauche Ich keine andere Lösung mehr.

lg

peff74, 2023-04-21:
> Hallo, das habe Ich nicht besonders gut ans laufen bekommen. Diese Lösung hat Out of the Box wirklich gut funktioniert. Und wenn jetzt die Ab.- und Zuschaltung über das Script funktioniert, brauche Ich keine andere Lösung mehr. lg

Gut zu wissen

feeti1, 2023-04-22:
Hallo,

Ich habe die Änderungen der config gestern übernommen.
Es hat soweit alles funktioniert. Bei 48V wurde auf das gewünschte Limit von 300W gewechselt.
Die Abschaltung wurde nicht erreicht, da die Spannung über Nacht nicht soweit gefallen ist.
Aus diesem Grund wurde auch die Wiedereinschaltschwelle nicht berücksichtigt.
Der Wechselrichter ist damit auf der 300W Begrenzung geblieben.
Hier müsste wohl eine zusätzliche Zeile für diesen Fall eingebaut werden.

lg

peff74, 2023-04-22:
Was ich als Info beitragen kann ist, dass das Skript einen ESP8266 mit Ahoy überfordert.
Das Skript ist gestern mehrfach einfach stehen geblieben, an der Stelle, wo er den Inverter abfragt.
Bei genauer Betrachtung, kam es zu der Zeit auch zu einem Restart des ESPs.
Habe heute Morgen auf ESP32 umgesattelt, das läuft jetzt seit 5h stabil

reserve85, 2023-04-22:
Ah ja ok, hab auch noch was bei openDTU entdeckt ich bin heute jedoch leider nicht zu Hause. Würde dann ein unteres Volt-Limit und noch ein zusätzliches oberes Volt-Limit einführen, damit die 300W und 1500Watt MaxPower nicht hin und herspringen. Also so ne Art Hysterese.

feeti1, 2023-04-22:
Also, Ich habe von Anfang an einen ESP32 benutzt.
Da darauf vorher auch openDTU lief.
Aus diesem Grund ist mir in dieser Richtung wohl nichts aufgefallen.

feeti1, 2023-04-22:
Wenn du unterwegs bist, genieße die Zeit und mach dir keine Gedanken um solche Dinge.
Ich hebe die Begrenzung einfach durch einen Neustart des Script auf, solange bis der Zusatz drin ist.
Heute werde Ich bestimmt unter die Abschaltschwelle fallen.
Dann weiss ich morgen ob "reenable" funktioniert.

reserve85, 2023-04-22:
@feeti1: neue Version ist online

feeti1, 2023-04-23:
Hallo,

habe ich gerade erst eingespielt. War mir erst jetzt möglich.

reserve85, 2023-04-23:
Gerne und danke für deine Spende, hab’s eben gesehen 🥇

peff74, 2023-04-25:
Habe nun auch einen Test Akku dran.
Hat allerdings nur ~0.8kWh und hängt nur an einem Eingang des Inverters. --> HM700
Da wenn ich das korrekt sehe aber in Deinem Skript nach allen U_DC im http://{AHOY_IP}/api/record/live gesucht wird...
Findet er am Eingang wo nichts angeschlossen ist quasi Null und schaltet ab wegen Unterspannung :-(

Ich bin aber zu blöde das Skrip so umzubauen, dass es einfach nur den ersten Wert nimmt.

Das hier habe ich zusammengebastelt, aber es läuft in einen Fehler:
`def GetHoymilesPanelMinVoltageAhoy(pInverterId):`
`    url = f'http://{AHOY_IP}/api/record/live'`
`    ParsedData = requests.get(url).json()`
`    PanelVDC = []`
`    for item in ParsedData['inverter'][pInverterId]:`
`        if item['fld'] == 'U_DC':`
`            PanelVDC.append(float(item['val']))`
`            break`
`    minVdc = float("inf")`
`    if PanelVDC:`
`       minVdc = PanelVDC[0]`
`    return minVdc`

[image]

reserve85, 2023-04-25:
eigentlich nicht, der sollte nur Spannungen > 1V berücksichtigen:
```
...
    for item in ParsedData['inverter'][pInverterId]:
        if item['fld'] == 'U_DC':
            PanelVDC.append(float(item['val']))
    minVdc = float("inf")
    for i in range(len(PanelVDC)):
    // für Mindestspannung nur Eingänge >1V berücksichtigen
        if (minVdc > PanelVDC[i]) and (PanelVDC[i] > 1):
            minVdc = PanelVDC[i]
    return minVdc
...
```

peff74, 2023-04-25:
O.K. ich habe nun > 1 in > 2  geändert und er greift nicht mehr auf die ungenutzten Eingänge zu.
Diese haben bei mir immer so zwischen 1-3V ;-)

Der Fehler kommt aber immer noch.

[image]

reserve85, 2023-04-25:
ahso, das kommt bei meinem unveränderten Code?
Hab das so verstanden als wäre das in der Funktion die du programmiert hast, ich hatte es noch nicht angeschaut...

update: Ich habe noch was beim Init gefunden und zwei "+=" operatoren die ggf. ein Problem verursachen, vielleicht lag es daran. Ich checke es ein.

peff74, 2023-04-25:
ich hatte nur mit meinem Skript getestet.
Nachdem ich verstanden hatte was die >1 bedeutet bin ich wieder zu Deinem zurück.

Nun mit der neuen Version gibt's keine Fehler mehr.

Was natürlich jetzt auch nicht funktioniert, ist die 0 Regelung.
Da ich nur einen Port am Inverter angeschlossen habe...
So sind 400W im Skript nun 200W am Inverter.

Kann man das vielleicht ohne grossen Aufwand anpassen, oder einfach die Regelung deaktivieren?

reserve85, 2023-04-25:
Oh, ich glaube das ist nicht so ohne weiteres Möglich. Die Berechnungen gehen immer Prozentual auf die 100%, aber er sollte sich dann doch einpendeln oder?

feeti1, 2023-04-25:
Hallo,

also bei mir klappt die Regelung meines HM-1500 im 🔋 Betrieb sehr gut.
Ich habe eine Begrenzung auf 600Watt DC drin, um das erstmal so laufen zu lassen.
Über den Tag gesehen arbeitet das System in Verbindung mit 8x100AH AGM Batterien in 4x Reihe +4xParallel so 12 bist 20 Stunden unterbrechungsfrei.
Es deckt damit zwischen 20 -50% meines täglichen Verbrauch.
Was leider nicht funktioniert ist die Abschaltung des Inverters bei ca. 47 Volt Spannung.
Ich habe bereits alle Zeilen außer der für die Abschaltschwelle und die für das Wiedereinschalten gelöscht.
Allerdings hat auch das keinen Unterschied gemacht.
Was kann Ich für die Fehlerdiagnose machen?

lg Dirk

reserve85, 2023-04-25:
Hi Dirk,

spiele nochmal die neue Version von heute ein (nur die *.py, an der config hat sich nix geändert), da bekommst du im Log die "kleinste Spannung" in jeder Runde vom Inverter angezeigt. Ich hatte Testhalber bei mir in der config die Spannungen mal so eingestellt wie es ahoy angezeigt hat und der Inverter wurde korrekt aus- und wieder eingeschaltet.
Poste bitte auch mal aus deiner Config die Settings für den Inverter.

feeti1, 2023-04-25:
[image]

reserve85, 2023-04-25:
Ah, du hast glaube ich nicht die Version aus dem DEV-Pfad gezogen, oder?
https://github.com/reserve85/HoymilesZeroExport/tree/dev_battery_powered

In der INI fehlen dir auch noch ein paar Einträge zur aktuellen Version.
-> *.ini und die *.py müsstest du auf den letzten Stand bringen, dann sollte es gehen.

peff74, 2023-04-26:
Also bei mir klappt mit der Steuerung über die Spannung einwandfrei.

[image]

Man sieht wie ich bis 16Uhr getestet habe und ab da das System selbst laufen lassen habe.
Gegen 20Uhr ist dann reduziert worden und gegen kurz vor 22Uhr abgeschaltet worden.
Heute Morgen gegen 10:30Uhr war dann der Akku wieder voll und es ging wieder los.

Mein Akku ist recht schwach, daher bricht die Spannung schon ziemlich ein, wenn der Inverter ein paar Ampere zieht.
Aber zum Testen reicht es erstmal.
TOP!
[image]

reserve85, 2023-04-26:
Das klingt ja schon mal gut, so sollte es sein. Danke für dein Feedback.

feeti1, 2023-04-26:
[image]

So jetzt ist die Spannung da. Habe ein zweites Verzeichnis gehabt, in dem Ich die .py aktualisiert hatte.
Aus diesem Grund hatte das auch keine Wirkung. Blöd ich bin. 
Werde mal sehen, wie sich heute in der Nacht das System verhält.

peff74, 2023-04-26:
Wäre es ohne grossen Aufwand möglich weiter Leistungsdrosselungen anhand von Spannung einzubauen?
Käme mir bei meinem kleinen Akku entgegen.

@feeti1 
Freut mich, dass es bei Dir nun auch klappt!

feeti1, 2023-04-27:
Hallo,

die Regelung ist jetzt über Nacht durchgelaufen.
Meine Batteriespannung ist unter die 300w Begrenzung gefallen und das Script hat auch auf 300w begrenzt.
Die Schwelle für die Abschaltung wurde nicht unterschritten.
Heute morgen wurde dann die Schwelle für das aufheben der 300w Begrenzung ubersch

feeti1, 2023-04-27:
ritten. Und das Script hat auch wieder freigegeben.
Also, es hat alles super funktioniert. Mit der Feinjustierung für die Regelung um 0 habe Ich noch nicht das richtige Maß gefunden. 
Entweder sind die Sprünge etwas groß oder zu klein für meinen Geschmack.
Das kann aber auch eine Systemgrenze sein.

lg

reserve85, 2023-04-27:
> Wäre es ohne grossen Aufwand möglich weiter Leistungsdrosselungen anhand von Spannung einzubauen?
> Käme mir bei meinem kleinen Akku entgegen.

@peff74 wäre theoretisch möglich. Wird halt schnell komplex für den Anwender, da musst du dann wieder zwei neue Spannungen und das Limit angeben (reduzierspannung, erhöhungsspannung, limit). Würde das erstmal ungern implementieren. Ggf. wäre es sinnvoll das irgendwie dynamisch abzubilden, das kostet aber einiges an Zeit..

@feeti1 super, kannst du ggf. mal die Sprünge erklären oder visualisiert darstellen? Sieht eigentlich auf den ersten Blick nicht schlecht aus, wenn https://github.com/reserve85/HoymilesZeroExport/issues/45#issuecomment-1522276347 noch aktuell ist.

peff74, 2023-04-27:
@feeti1 
Mal eine Verständnisfrage, hast Du einfach den Akku nun an beide (bzw. bei Dir ja 4) Eingänge des Wechselrichters gehängt?
Also quasi ein Y-Y Kabel gebaut?

reserve85, 2023-04-27:
Ich hab da auch ne frage: wie lange halten die Akkus (Lebensdauer)? Sind normale Gelbatterien aus dem Auto?

peff74, 2023-04-27:
Ich arbeite mit Li Ion Akkus.
Mein Testakku ist ein defekter Rucksackakku für Gartengeräte.
Defektes BMS rausgeschmissen, JKBMS reingebaut.

Der große wird auf "recycelten" E-Bike / Gartengeräte / etc. Akkus gebaut 12x48 Stück 18650er. Da fehlen mir aber noch ~50 Stück.

peff74, 2023-04-27:
Wichtig bei Li nie 'Voll' oder 'Leer' laden. Das erhöht die Lebensdauer enorm.

feeti1, 2023-04-27:
@peff74 
Ich habe an den 4 Eingängen die Batterien hängen. Und parallel zu den Batterien ein Victron MPPT Ladegerät an dem die 4
Panels hängen. Das funktioniert wirklich gut und die Akkus werden in der Regel über den Tag voll geladen.
@reserve85 
Ich verwende normale Gel Autobatterien, da diese für mich "kostenlos" zur Verfügung stehen. Man geht dabei von einer
Zyklenfestigkeit in einer Größe von 600-800 Zyklen aus.
Diese erhöht sich, wenn man die Akkus nicht komplett entleert. Aus diesem Grund entlade Ich bis max. 47,5 unter Last.
Sobald sich die Anlage ausschaltet sind dann wieder mehr als 48Volt auf dem Batteriesystem.
Außerdem sind diese Batterien in meinen Augen sehr sicher.
Ich komme aus der KFZ Branche und habe mit verschiedenen Batterietypen Erfahrungen. 
Ich arbeite auch an HV-Systemen. Und Ich bleibe lieber im Bereich mit 48V und konventionellen Batterien.
Das reicht meiner Meinung nach für so ein Balkonkraftwerk.
Außerdem mag LiFEPO keine Minustemperaturen und nimmt dann schnell schaden.
Li Ion sind von Hause aus eher fragil und  Sie reagieren sporadisch unvorhersehbar. Außerdem ist das mit der Ladung schwieriger in meiner Kombi umzusetzen.

Ich110, 2023-04-27:
Guten Abend.
Ich betreibe zur Zeit einen Hoymiles HM300 an einem 24V 100Ah LiFePo Akku mit openDTU fest begrenzt auf 80 Watt für eine reine Grundlastdeckung über Nacht. Als Zähler ist ein Shelly 3EM verbaut. Ist es möglich den Hoymiles so zu regeln, dass man diese feste Begrenzung (80 Watt) vielleicht variabel gestallten könnte z.B. in einem Bereich zwischen 80 und 150 Watt um eventuelle Lasten mit abzudecken?

 
mfg Yves

feeti1, 2023-04-27:
Hallo,

reserve85 hat das hier alles programmiert.
Mit dieser Konstellation kann man das denke ich 1:1 nach deinem Wunsch umsetzen.
Allerdings besser auf Basis von ahoydtu.
Mit openDTU hat das bei mir nicht so gefunzt.
Sofern du einen ESP32 als Basis hast, ist das umflashen allerdings problemlos möglich.

lg

reserve85, 2023-04-27:
Ja, das ist jetzt schon möglich. Ich habe in der Version 1.28 auch die Option "HOY_BATTERY_NORMAL_WATT" mit eingepflegt, das ist genau dasd was du brauchst. Damit kannst du deinen Inverter im Batterie-Modus reduzieren (z.B. wenn du einen 1500Watt Inverter hat kannst du im Batteriebetrieb max. 500 Watt einstellen).

Die neue Version habe ich eben in die MAIN gemerged. Somit ist das jetzt Live.

feeti1, 2023-04-27:
[image]
Hier ist mein HM-1500 begrenzt. Sofern du die Spannungen und Watt nach deinen Wünschen anpasst, sollte es problemlos mit der von reserve85 geleisteten Arbeit umzusetzen sein.

feeti1, 2023-04-27:
[image]
@reserve85 
Hier nochmal die aktuelle Regelung.
Und meine Einstellungen in der config.
[image]

feeti1, 2023-04-27:
Vor allen Dingen bei großen Sprüngen nach unten, dauert es mir gefühlt zu lange, bis es sich eingeregelt hat.
Allerdings bin Ich mir mit der Wirkung von "slow_approx" auch nicht ganz klar, wie die Auswirkung ist.
Wenn du dort für meinen Fall andere Einstellungen vorschlägst, werde Ich gerne probieren.

peff74, 2023-04-27:
@Ich110 
Bei 24v Systemen regelt der Hoymiles nicht so sauber. Habe ich schon häufiger gelesen. Also besser in Richtung 36 oder 48 Volt

Ich110, 2023-04-27:
[image]
Zur Zeit Sieht mein Verbrauch über Nacht so aus. Wollte eigentlich nicht über die Batterie-Voltage gehen. Mir würde die Begrenzung des HM300 auf 150 Watt reichen. So das nur die Spitzen in der Nacht noch etwas abgesenkt werden. Ich setz mich Morgen ran. Das Script läuft bei mir noch auf 1.14.
Ich danke Euch und werde berichten.

reserve85, 2023-04-27:
> Vor allen Dingen bei großen Sprüngen nach unten, dauert es mir gefühlt zu lange, bis es sich eingeregelt hat.
> Allerdings bin Ich mir mit der Wirkung von "slow_approx" auch nicht ganz klar, wie die Auswirkung ist.

mach mal Slow Approx auf 0, dann springt er immer direkt zum Zielwert. Für Akkubetrieb brauchst du das nicht. Und JUMP_TO_MAX_LIMIT ist im batteriebetrieb auch nicht notwendig, würde sagen sogar unerwünscht.

feeti1, 2023-04-28:
Danke für diese Erweiterung und Erklärung. Werde Ich heute am Abend mit einpflegen.
Mein HM-1500 mag die Einstellung auf 300W nicht, er regelt dann maximal bis 218W DC. Und das auch nur nach Lust und
Laune.
Ich habe den Wert jetzt auf 400W gesetzt, damit regelt er jetzt bis 318W DC.
Dieses Verhalten war mir allerdings schon bei openDTU mit niedrigen Watt Werten aufgefallen.

peff74 hat ja schonmal nach weiteren Möglichkeiten der Leistungsdrosselung gefragt.
Nachdem jetzt alles so super läuft, wäre für mich auch die Frage, ob man über einer Spannung X zu Beispiel 52V die 
komplette Leistung freigeben könnte?
Wenn ich das einfach in der config eingebe, kann die .py damit nicht umgehen,oder?

reserve85, 2023-04-28:
> Mein HM-1500 mag die Einstellung auf 300W nicht, er regelt dann maximal bis 218W DC. Und das auch nur nach Lust und
> Laune.
> Ich habe den Wert jetzt auf 400W gesetzt, damit regelt er jetzt bis 318W DC.

klingt fast so, als wäre ein Eingang offen? ist ja immer 3/4 der Leistung...

> Nachdem jetzt alles so super läuft, wäre für mich auch die Frage, ob man über einer Spannung X zu Beispiel 52V die
> komplette Leistung freigeben könnte?
> Wenn ich das einfach in der config eingebe, kann die .py damit nicht umgehen,oder?

Das wäre denkbar. Also im Prinzip noch eine Regelstufe dazunehmen. Komme wahrscheinlich aber erst kommende Woche dazu.

peff74, 2023-04-28:
@feeti1 
> Ich verwende normale Gel Autobatterien, da diese für mich "kostenlos" zur Verfügung stehen.
Da hast Du nicht zufällig noch ein "paar" übrig^^
Dann kann ich mit das Zusammenbauen der 18650 sparen ;-)

feeti1, 2023-04-28:
@peff74 

Das könnte schon sein. Aber 26kg pro 100AH sind schlecht zu versenden. ; )

@reserve85 

[image]

Wenn Ich die 1500W eintrage, werden Sie auch sofort genutzt.
Und diesen HTTP Fehler habe Ich mit der 1.28 zum ersten Mal gesehen. Der kommt aber nur sporadisch.
Jetzt seit 10min nicht mehr.

peff74, 2023-04-28:
@feeti1 
Kommt darauf an, wo Du herkommst
Bin viel unterwegs und im Kofferraum ist Platz^^

reserve85, 2023-04-28:
> Wenn Ich die 1500W eintrage, werden Sie auch sofort genutzt. Und diesen HTTP Fehler habe Ich mit der 1.28 zum ersten Mal gesehen. Der kommt aber nur sporadisch. Jetzt seit 10min nicht mehr.

Tatsache, das liegt daran dass ich im Solarbetrieb beim Init hochregele. Ist nur beim Start des Scripts.
Um das zu verhindern ändere den Block mal (ab Zeile 725):

```
try:
    logger.info("---Init---")
    newLimitSetpoint = 0
    if GetHoymilesAvailable():
        for i in range(INVERTER_COUNT):
            SetHoymilesPowerStatus(i, True)
        newLimitSetpoint = GetMaxWattFromAllInverters()
        GetHoymilesActualPower()
        SetLimit(newLimitSetpoint)
    GetPowermeterWatts()
    time.sleep(SET_LIMIT_DELAY_IN_SECONDS)
```
in:
```
try:
    logger.info("---Init---")
    newLimitSetpoint = 0
    if GetHoymilesAvailable():
        for i in range(INVERTER_COUNT):
            SetHoymilesPowerStatus(i, True)
        GetCheckBattery()
        newLimitSetpoint = GetMinWattFromAllInverters()
        GetHoymilesActualPower()
        SetLimit(newLimitSetpoint)
    GetPowermeterWatts()
    time.sleep(SET_LIMIT_DELAY_IN_SECONDS)
```

Wenn das so passt, dann würde ich das so übernehmen.
Woher der http Fehler kommt ist eine gute Frage, ich konnte das noch nicht beobachten. Vermutlich hast du ein IntermediateMeter in der config angegeben und keine IP-Adresse in der section?

feeti1, 2023-04-28:
[image]
@reserve85 

Du bist echt ein Profi. Das läuft!!!

reserve85, 2023-04-28:
Was war mit dem http Fehler?

[13 more replies]