reserve85/HoymilesZeroExport #47: Implementierung JSON Abfrage

peff74, 2023-04-20:
Hi,

Du schreibst, dass es easy wäre eine neue JSON Abfrage einzubauen, leider habe ich mit Python null Erfahrungen.
Aber ich lese meinen Stromzähler mit einem ESP aus, der wiederum seine Website über JSON aktualisiert.
Heisst bei \\192.168.xxx.xxx\get_measurements schaufelt er folgenden JSONString zurück.

["10994.379","   36","47 days, 7 hours, 1 minutes, 14 seconds","37.16 KB","26.61 KB","22","32","-77","3C:37:12:D9:74:B1","ESP8266_SML_v17","Nov 23 2022","14:36:06","Software/System restart"]

Der zweite Wert ist dann der aktuelle Zählerwert bei Einspeisung mit Minus davor.
Machbar/Aufwand?

Gruß
Peff

reserve85, 2023-04-20:
Hi,
das ist machbar, quick and dirty sollte das so aussehen:

ersetze die Funktion `def GetPowermeterWattsTasmota():` durch:

```
def GetPowermeterWattsTasmota():
    url = f'http://{TASMOTA_IP}/get_measurements'
    ParsedData = requests.get(url).json()
    Watts = int(ParsedData[1])
    logger.info("powermeter: %s %s",Watts," Watt")
    return int(Watts)
```

Natürlich wäre es besser das noch umzubenennen oder als neue Variante zu ergänzen. Zum testen sollte das aber so bereits funktionieren. Ist das eine Sonderlösung von dir oder hast du da ein fertiges Modul/Software?

peff74, 2023-04-20:
Hi,

teste ich Morgen mal, habe heute schon auf die Schnelle im ioBroker über JS die Daten in den ioB gezogen um sie dann von dort in für Dein Tool zur Verfügung zu stellen. Von hinten, durch die Brust, ins Auge^^

Meine Lösung ist nen Eigenbau weils für meinen Zähler nix fertiges gab...
https://github.com/peff74/MT691_to_InfluxDB

Was noch genial wäre, wenn man den Akkufüllstand mit einbinden könnte.
So nach dem Motto, wenn Spannung unter X dann Inverter auf 0%.

closed 2023-05-03 by reserve85