reserve85/HoymilesZeroExport #48: Keine Daten vom Tasmota Sensor

keinprofi, 2023-04-22:
Hi,

geniales Projekt, nur für mich leider zu schwer....

Ich habe alles soweit auf meinem RasPi installiert und "einiges" funktioniert auch. Bis auf die Verbindung mit dem Tasmota Sensor (Hichi WLAN). 
Bilder sagen mehr als 1000 Worte:
Der Tasmots Hichi WLAN Sensor (unabhängig vom System im Browser):
[image]

Der Raspi:
[image]

Wie bekomme ich das Script nun dazu den aktuellen Verbrauch sauber auszulesen, damit die WR entsprechend angepasst werden können?

Beste Grüße
Ka.

reserve85, 2023-04-22:
Hi, das kriegen wir schon hin:
Gib mal bitte das hier im Browser ein und teile bitte das result: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010

Natürlich mit der korrekten IP von tasmota.
Und kannst du die IP auch von dem RasPi pingen?

keinprofi, 2023-04-22:
Raspi kann ich ohne weiteres erreichen. 
Beim Link oben bekomme ich File not found.

Muss ich die Raspi IP in der Config irgendwo angeben? Hab da nichts gesehen.

reserve85, 2023-04-22:
Die xxx hast du auch geändert? Zur tasmota IP?

keinprofi, 2023-04-22:
klar doch.
Aber ich habe mich vertippt, sorry...

[image]

keinprofi, 2023-04-22:
[image]

reserve85, 2023-04-22:
ok, aber ist ja aktiv wenn du da was zurückbekommst.
Du musst in der Config einstellen:

```
[TASMOTA]
...
TASMOTA_IP = xxx.xxx.xxx.xxx
TASMOTA_JSON_STATUS = StatusSNS
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = Strom
TASMOTA_JSON_POWER_MQTT_LABEL = current
...
```

keinprofi, 2023-04-22:
Die Daten bekommt er jetzt :-)
[image]
sind die verbliebenden Fehler relevant? Der Sendor ist erreichbar (10.10.10.139) aber trotzdem Fehler?

reserve85, 2023-04-22:
ja, das ist nicht so gut. Hast du am Tasmota eine schlechte WLAN Verbindung? kannst du mal über das Terminal über längere Zeit den Tasmota anpingen?

keinprofi, 2023-04-22:
naja, der hängt leider in der Nachbarwohnung bei meinem Vermieter. Allerdings ein Repeater direkt im Zählerschrank (aus Metall...).
[image]

keinprofi, 2023-04-22:
Ich bin gespannt auf Morgen wenn wieder Sonne da ist.

Dir auf jeden Fall mal tausend Dank!!!

reserve85, 2023-04-22:
ja da ist dennoch ein Problem, du hast teilweise über 100ms für einen einfachen Ping und auch da bereits zwei mal Zeitüberschreitung... Musst du einfach mal beobachten, eventuell kannst du den repeater ja noch etwas optimieren.

keinprofi, 2023-04-22:
OK, ich werde es mal versuchen. Mal sehen was geht.

reserve85, 2023-04-22:
vielleicht hast du auch durch die Zählerschranktür den schlechten Empfang

keinprofi, 2023-04-22:
da gehe ich von aus. ABer da der Repeater ja an meine Steckdose im Zählerschrank muss und außen kein Platz für den Repeater ist, könnte ich höchstens externe Antennen an den Repeater anbringen und die nach außen verlegen. Ist ein AVM 1200, muss ich erstmal nach Material gucken.

closed 2023-05-03 by reserve85