reserve85/HoymilesZeroExport #51: Einrichtung des Programms

Cedi00123, 2023-05-01:
Hallo, ich weis mir nicht mehr weiter.

Ich bin recht neu in dem Thema und versuche eine Nulleinspeisung mit meinem Balkonkraftwerk. 
Dafür ist dieses Programm wie gemacht für meine Komponenten.

Open DTU und Smart Meter (mit Tasmota) eingerichtet und läuft.

Python Version 3.11.1 
Banana Pi m2 zero
Ubuntu

Leider bekomm ich es das Programm nicht zum Laufen. Hier einmal was mir ausgespuckt wird.
[image]

Was mach ich falsch? Läuft das gar nicht auf Ubuntu?
Vielleicht kann jemand mir ja Helfen. Vielen Dank im voraus.

reserve85, 2023-05-01:
Hallo,
sowas hatten wir hier schonmal, bist du sicher, dass das Script mit Python 3.11 geöffnet wird? Wenn Python geupdatet wurde, dann musst du glaube ich die Default-Version noch aktivieren.

siehe auch:
https://github.com/reserve85/HoymilesZeroExport/issues/41

Kannst du das Script mal direkt mit Python 3.11 öffnen und testen?
`python3.11 HoymilesZeroExport.py`

karrenbauer, 2023-05-12:
Wenn ich den Skript unter Windows starte, wo finde ich denn das Log-File?

reserve85, 2023-05-12:
Du musst in der Config die Zeile aktivieren: (auf true setzen)
```# enable logging to file
ENABLE_LOG_TO_FILE = true```

Dann findest du das Log im Unterverzeichnis "Log" wo deine HoymilesZeroExport.py liegt.

Cedi00123, 2023-05-13:
Bei mir lag es am Raspberry. Ich hatte es versucht auf einem Banana Pi auszuführen. Dann hab ich einen Raspberry Pi 3 A+ gekauft und mit dem funktionierte es sofort.  

Ich habe nur noch ein kleines Problem.  Ich habe einen Wechselrichter mit 1500W Ausgangsleistung. Möchte aber eine Obergrenze von 600W haben. Wenn ich in der Config das Limit auf 600W setze und mehr verbraucht wird dann springt er automatisch auf 1500W.

[image]

Das Bild zeigt meine Config. 

Vielen Dank.

Cedi00123, 2023-05-13:
Ach und wenn es auf 600W eingestellt ist dann speist er oft ca 300 Watt mehr ein als verbraucht wird. Das funktioniert dann nur richtig wenn das Limit auf 1500W steht.

reserve85, 2023-05-13:
@Cedi00123 du musst HOY_MAX_WATT auf 1500 stellen. Das ist immer das absolute Inverterlimit. Die 600 hast du ja bereits unten eingestellt. Dann wird auch dein zweites Problem mit der Überproduktion verschwinden, da das Verhältnis falsch berechnet wurde.

reserve85, 2023-05-15:
Servus,

Ich habe das generell nur für den Batteriemodus vorgesehen -> HOY_BATTERY_NORMAL_WATT.
Wenn du keinen Batteriebetrieb nutzt kannst du auch HOY_MAX_WATT auf 600W limitieren, das funktioniert aber **ausschließlich mit AHOY!**
Bei OpenDTU funktioniert das nicht, da hier das Limit in % abhängig vom Inverter übertragen wird.

closed 2023-05-30 by reserve85