reserve85/HoymilesZeroExport #54: läuft nicht unter windows

tthnn, 2023-05-07:
Hallo seit 3 tagen versuche ich das skript zum laufen zu bekommen 
gedownloaded und die config ausgefüllt 
Allerdings kommen immer nur fehler meldungen .
Wie wird das skript richtig installiert ? 
Aktuellste windows und Phyton version sowie Requests sind drauf .
der skript von selbstbau pv läuft problemlos also wo liegt mein fehler ?
Gibt es eine schritt für schritt anleitung ?

reserve85, 2023-05-07:
Zeig mal dein Log und deine Config.

reserve85, 2023-05-24:
Hi,
postet doch mal was genau nicht funktioniert (logs, bilder oder sonst irgendwas), Hellsehen kann hier leider keiner. 
Ich helfe gerne.

Zynus, 2023-05-24:
>  Hellsehen kann hier leider keiner.
Das stimmt aber bei mir sind es so grundlegende Dinge.
Ich habe Python 3.11.3 installiert aber wenn ich dann die Datei HoymilesZeroExport.py öffnen will. passiert nichts.

Zynus, 2023-05-24:
Das stimmt aber bei mir sind es so grundlegende Dinge.
Ich habe Python 3.11.3 installiert aber wenn ich dann die Datei HoymilesZeroExport.py öffnen will. passiert nichts.

Zynus, 2023-05-24:
[image]

reserve85, 2023-05-25:
Ok er findet in der Sektion OPEN_DTU die IP nicht in der config. Bitte hänge hier mal deine config an, dann finden wir das Problem.

Zynus, 2023-05-25:
[VERSION]
VERSION = 1.33

[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = true
USE_OPENDTU = false

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_3EM = true
USE_SHELLY_3EM_PRO = false
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = 192.168.2.9

[SHELLY_3EM]
# --- defines for Shelly ---
SHELLY_IP = 192.168.2.15

[EMLOG]
# --- defines for EMLOG (electronic meter log) System ---
EMLOG_IP = 192.168.2.8
EMLOG_METERINDEX = 

[HOMEASSISTANT]
# --- defines for HOME ASSISTANT ---
HA_IP = 192.168.2.56
HA_PORT = 8123
# you need to create a Long-Lived Access Token in your Home Assistant: click on your profile, then scroll down to the bottom
HA_ACCESSTOKEN = GlXchdoyMZdQh7TXlByH8fyiNuJOaSjtpbI1rfLtzUc
HA_CURRENT_POWER_ENTITY = sensor.dtz541_sml_curr_w
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
HA_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
HA_POWER_INPUT_ALIAS = sensor.dtz541_sml_170
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
HA_POWER_OUTPUT_ALIAS = sensor.dtz541_sml_270

[SELECT_INTERMEDIATE_METER]
# if you have an intermediate meter ("Zwischenzähler") to measure the outputpower of your inverter you can set it here. It is faster than the DTU current_power value
# --- define your intermediate meter - if you don´t have one set the following defines to false to use the value from your DTU---
USE_TASMOTA_INTERMEDIATE = false
USE_SHELLY_3EM_INTERMEDIATE = false
USE_SHELLY_3EM_PRO_INTERMEDIATE = false
USE_SHELLY_1PM_INTERMEDIATE = false
USE_SHELLY_PLUS_1PM_INTERMEDIATE = false
USE_SHRDZM_INTERMEDIATE = false
USE_EMLOG_INTERMEDIATE = false
USE_IOBROKER_INTERMEDIATE = false
USE_HOMEASSISTANT_INTERMEDIATE = false

[INTERMEDIATE_SHELLY]
# --- defines for Shelly ---
SHELLY_IP_INTERMEDIATE = xxx.xxx.xxx.xxx 

[INTERMEDIATE_EMLOG]
# --- defines for EMLOG (electronic meter log) System ---
EMLOG_IP_INTERMEDIATE = xxx.xxx.xxx.xxx
EMLOG_METERINDEX_INTERMEDIATE = 

[INTERMEDIATE_HOMEASSISTANT]
# --- defines for HOME ASSISTANT (you need to create a Long-Lived Access Token in your profile) ---
[... 64 lines cut ...]
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 25.8

# grid power
#    ...
#     |
#     | -------> jump limit to HOY_MAX_WATT if (JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = TRUE), else: increasing limit <-------
#     |
#   [0W]      [POWERMETER_MAX_POINT]
#     |
#     | -------> increasing limit <-------
#     |
#  [-50W]     [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-75W]     [POWERMETER_TARGET_POINT]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-100W]    [POWERMETER_TARGET_POINT - POWERMETER_TOLERANCE]
#     |
#     | -------> decreasing limit <-------
#     |
#    ...

reserve85, 2023-05-25:
Ok, du hast doch einige Sektionen rausgelöscht, dann funktioniert es nicht. 
Ich habe es angepasst, siehe Anhang (musst die Dateiendung noch von *.txt in *.ini ändern).

[HoymilesZeroExport_Config.txt](https://github.com/reserve85/HoymilesZeroExport/files/11568989/HoymilesZeroExport_Config.txt)

Zynus, 2023-05-25:
Ok, ich teste es gleich. Ich habe nicht gewusst das ich nicht löschen darf.

Zynus, 2023-05-25:
[image]

Zynus, 2023-05-25:
Funktioniert und Ahoy regelt.
Nun muss ich die config noch anpassen. Zum Verständnis, ich habe einen HM 300 an einem 24V LifePo4  Akku hängen, der nur Nachts zwischen Sonnenuntergang und Sonnenaufgang an ist.
Wenn ich möchte, das Ahoy zwischen 50 und 275 W regeln darf, wo muss ich das in der config genau anpassen.
Das hat sich mir noch nicht ganz erschlossen.

Das Skript läuft soweit über python in der Win Eingabeaufforderung, ziel ist es doch, dass das Skript in HA aktiv ist, wenn der Hoymiles nachts an ist. Verstehe ich das richtig, das ich das Skript und die COnfig irgendwie in HA implementieren muss ?

Zynus, 2023-05-25:
Ok, das mit der Einspeisung habe ich in der Config hin bekommen.
Ist echt eine Mega Projekt. Danke für deine Hilfe.

[image]
[image]

Zynus, 2023-05-25:
Nur wie gesagt, bin ich etwas planlos wie ich das in mein HA implementieren kann.

reserve85, 2023-05-26:
Gar nicht, das ist ein eigenes Python Script welches du nicht im HA einbinden kannst.

Zynus, 2023-05-26:
aber ich kann es auf ein RasPi 4 schreiben, oder?

reserve85, 2023-05-26:
ja, das habe ich auch so gemacht. siehe https://github.com/reserve85/HoymilesZeroExport#linux-installation

Zynus, 2023-05-27:
[image]

Keine Ahnung warum das mit sudo nicht klappt.

Zynus, 2023-05-28:
Kann man das Skript auch ohne die Config verwenden. Falls das gehen würde, kann ich das Skript in HA implementieren.

reserve85, 2023-05-28:
Ohne Config geht das nicht so ohne weiteres.
Du musst ggf. Sudo noch installieren. Welches Betriebssystem hast du denn da drauf?
Siehe https://www.makeuseof.com/fix-sudo-command-not-found-on-linux/

Am besten ist halt HomeAssistant als Supervisor-installation, dann hast du noch ein vollwertiges Linux nebenher und könntest beispielsweise noch ein PiHole nebenher laufen lassen.
https://peyanski.com/how-to-install-home-assistant-supervised-official-way/#How_to_Install_Home_Assistant_Supervised

Zynus, 2023-05-28:
Ich habe auf meinem Raspberry nur HA drauf.

Zynus, 2023-06-01:
Ich habe deinen Rat befolgt und das ganze unter raspbian os lite zum laufen gebracht. Super skript, danke für deine Hilfe.

reserve85, 2023-06-01:
Freut mich, dass es läuft. Hast du dafür jetzt dein HA geopfert? Wie gesagt, du kannst auch beides laufen lassen mit der Supervised-Methode. Ansonsten mach ich hier mal zu.

closed 2023-06-01 by reserve85

Zynus, 2023-06-01:
Ich war etwas pragmatisch und lass es auf einen anreden Raspberry laufen :-)