reserve85/HoymilesZeroExport #55: Automatischer Scriptinterval durch Zeitmessung

Ollipop030, 2023-05-08:
Hi,

die nächste Idee:
Wäre es möglich beim Scriptstart den Aktualisierungsintervall des Stromzählers zu messen und als Abfrageintervall festzulegen? Dann müsste der Stromzähler nicht so oft abgefragt werden, sondern immer genau dann, wenn auch ein neuer Wert zu erwarten ist. 

So stelle ich mir das vor:
Das Script startet, und fragt den Zähler sekündlich ab. Bei der ersten Änderung des Wertes beginnt die Zählung, und bei der nächsten Änderung steht ein Wert fest (zB. 10 Sekunden). Diesen Wert merkt sich das Script und fragt immer nur alle 10 Sekunden den Zähler ab. 

Allerdings werden die Sekunden beim Limitsetzen (SET_LIMIT_DELAY_IN_SECONDS usw.) dazuaddiert, dass bringt das Ganze dann wieder durcheinander. Vielleicht könnte man auch mit der Uhrzeit arbeiten, wann ein neuer Wert zur Verfügung steht.

reserve85, 2023-05-08:
Hi,

mMn. ist das nicht möglich, das läuft dir über die Zeit einfach weg. Lass es nur wenige Millisekunden unterschied sein, am Ende des Tages hast du mehrere Sekunden Offset. Abtastrate muss immer hoch sein...

Ollipop030, 2023-05-08:
Aahrgh, das hatte ich befürchtet. Naja, war ein Versuch wert.

Danke dir trotzdem.

closed 2023-05-11 by reserve85