reserve85/HoymilesZeroExport #56: Config parameter

fatzo, 2023-05-12:
ich bekomme den script nicht zum laufen auf einen Raspi zero

log

Mai 12 09:46:01 raspberrypizero systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mai 12 09:46:01 raspberrypizero systemd[1]: HoymilesZeroExport.service: Consumed 3.694s CPU time.
Mai 12 09:46:01 raspberrypizero systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 11.
Mai 12 09:46:01 raspberrypizero systemd[1]: Stopped HoymilesZeroExport Service.
Mai 12 09:46:01 raspberrypizero systemd[1]: HoymilesZeroExport.service: Consumed 3.694s CPU time.
Mai 12 09:46:01 raspberrypizero systemd[1]: Started HoymilesZeroExport Service.
Mai 12 09:46:07 raspberrypizero python3[6463]: 2023-05-12 09:46:07 INFO     Log write to file: False
Mai 12 09:46:07 raspberrypizero python3[6463]: 2023-05-12 09:46:07 INFO     Author: Tobias Kraft / Script Version: 1.32
Mai 12 09:46:07 raspberrypizero python3[6463]: 2023-05-12 09:46:07 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Mai 12 09:46:07 raspberrypizero python3[6463]: 2023-05-12 09:46:07 INFO     Config file V 1.31
Mai 12 09:46:07 raspberrypizero python3[6463]: Traceback (most recent call last):
Mai 12 09:46:07 raspberrypizero python3[6463]:   File "/usr/lib/python3.9/configparser.py", line 789, in get
Mai 12 09:46:07 raspberrypizero python3[6463]:     value = d[option]
Mai 12 09:46:07 raspberrypizero python3[6463]:   File "/usr/lib/python3.9/collections/__init__.py", line 941, in __getitem__
Mai 12 09:46:07 raspberrypizero python3[6463]:     return self.__missing__(key)            # support subclasses that define __missing__
Mai 12 09:46:07 raspberrypizero python3[6463]:   File "/usr/lib/python3.9/collections/__init__.py", line 933, in __missing__
Mai 12 09:46:07 raspberrypizero python3[6463]:     raise KeyError(key)
Mai 12 09:46:07 raspberrypizero python3[6463]: KeyError: 'ahoy_ip'
Mai 12 09:46:07 raspberrypizero python3[6463]: During handling of the above exception, another exception occurred:
Mai 12 09:46:07 raspberrypizero python3[6463]: Traceback (most recent call last):
Mai 12 09:46:07 raspberrypizero python3[6463]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 675, in <module>
Mai 12 09:46:07 raspberrypizero python3[6463]:     AHOY_IP = config.get('AHOY_DTU', 'AHOY_IP')
Mai 12 09:46:07 raspberrypizero python3[6463]:   File "/usr/lib/python3.9/configparser.py", line 792, in get
Mai 12 09:46:07 raspberrypizero python3[6463]:     raise NoOptionError(option, section)
Mai 12 09:46:07 raspberrypizero python3[6463]: configparser.NoOptionError: No option 'ahoy_ip' in section: 'AHOY_DTU'
Mai 12 09:46:08 raspberrypizero systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mai 12 09:46:08 raspberrypizero systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mai 12 09:46:08 raspberrypizero systemd[1]: HoymilesZeroExport.service: Consumed 3.736s CPU time.
Mai 12 09:46:08 raspberrypizero systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 12.
Mai 12 09:46:08 raspberrypizero systemd[1]: Stopped HoymilesZeroExport Service.

ich denke das ich die Config.ini nicht richtig konfiguriert habe.

ich habe 2 Hoymiles HM-1500
könnte über 
tasmota Lesekopf
shelly 3em 
Homeassistent
steuern
Ich habe eine opendtu mit  aktuellen Firmwareversion am laufen

der raspi zero ist nur für den sript da

# HoymilesZeroExport - https://github.com/reserve85/HoymilesZeroExport
# Copyright (C) 2023, Tobias Kraft

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

[VERSION]
VERSION = 1.31

[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = false
USE_OPENDTU = true
[... 214 lines cut ...]
# grid power
#    ...
#     |
#     | -------> jump limit to HOY_MAX_WATT if (JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = TRUE), else: increasing limit <-------
#     |
#   [0W]      [POWERMETER_MAX_POINT]
#     |
#     | -------> increasing limit <-------
#     |
#  [-50W]     [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-75W]     [POWERMETER_TARGET_POINT]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-100W]    [POWERMETER_TARGET_POINT - POWERMETER_TOLERANCE]
#     |
#     | -------> decreasing limit <-------
#     |
#    ...

vieleicht kann mir jemand helfen
gruß fatzo

reserve85, 2023-05-12:
Jo, 

du hast ein # vor ahoy_ip gesetzt (wird zwar nicht benötigt aber trotzdem eingelesen) und 
```
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
```
muss bei inverter 2 rein.

also hier nochmal komplett:

```
# HoymilesZeroExport - https://github.com/reserve85/HoymilesZeroExport
# Copyright (C) 2023, Tobias Kraft

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

[VERSION]
VERSION = 1.31

[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = false
USE_OPENDTU = true

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = true
USE_SHELLY_3EM = false
USE_SHELLY_3EM_PRO = false
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = 192.168.10.57

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = 192.168.178.xx
OPENDTU_USER = admin
OPENDTU_PASS = xxxxxxxx

[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = 192.168.178.73
# the following three constants describes how to navigate through the Tasmota-JSON. 
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = curr_w
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
[... 179 lines cut ...]
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51

# grid power
#    ...
#     |
#     | -------> jump limit to HOY_MAX_WATT if (JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = TRUE), else: increasing limit <-------
#     |
#   [0W]      [POWERMETER_MAX_POINT]
#     |
#     | -------> increasing limit <-------
#     |
#  [-50W]     [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-75W]     [POWERMETER_TARGET_POINT]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-100W]    [POWERMETER_TARGET_POINT - POWERMETER_TOLERANCE]
#     |
#     | -------> decreasing limit <-------
#     |
#    ...
```

Ollipop030, 2023-05-12:
Folgendes ist mir aufgefallen:

Du hast die Open DTU nicht richtig eingetragen: OPENDTU_IP = 192.168.178.xx, und kein Benutzer/Passwort.

bei TASMOTA_IP_INTERMEDIATE = 192.68.178.73, da fehlt eine 1. Die Intermediate IP ist aber nicht die gleiche wie die Smartmeter IP.

Ollipop030, 2023-05-12:
Unter Homeassistent stehen auch allerlei Sachen eingetragen, aber USE_HOMEASSISTANT = false. Womit möchtest du denn den Zähler auslesen?

fatzo, 2023-05-12:
> Folgendes ist mir aufgefallen:
> 
> Du hast die Open DTU nicht richtig eingetragen: OPENDTU_IP = 192.168.178.xx, und kein Benutzer/Passwort.
> 
> bei TASMOTA_IP_INTERMEDIATE = 192.68.178.73, da fehlt eine 1. Die Intermediate IP ist aber nicht die gleiche wie die Smartmeter IP.

wo finde ich den die intermediate IP vom Tasmota?
bei der open dtu hatte ich die daten rausgenommen

Ollipop030, 2023-05-12:
> wo finde ich den die intermediate IP vom Tasmota?

Intermediate ist ein Zwischenzähler für die Einspeiseleistung. Das machst du aber doch über den Shelly oder?

reserve85, 2023-05-12:
Mal eins nach dem anderen:

stelle erstmal OpenDTU richtig ein: 
```
[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = 192.168.178.xx
OPENDTU_USER = admin
OPENDTU_PASS = xxxxxxxx
```

dann dein Tasmota. 
```
[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = 192.168.178.73
# the following three constants describes how to navigate through the Tasmota-JSON. 
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = curr_w
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
TASMOTA_JSON_POWER_INPUT_MQTT_LABEL =
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
TASMOTA_JSON_POWER_OUTPUT_MQTT_LABEL =
```

Wichtig hier: gib http://192.168.178.73/cm?cmnd=status%2010 in deinen Browser ein (mit der richtigen IP) und passe dann die drei Felder auf die Rückgabe an: `TASMOTA_JSON_STATUS , TASMOTA_JSON_PAYLOAD_MQTT_PREFIX , TASMOTA_JSON_POWER_MQTT_LABEL`
Du kannst auch gerne die Rückgabe hier posten, dann kann ich dir besser helfen.

Dann deaktiviere vorerst den Intermediate Zähler, den kannst du im Nachgang auch noch einrichten.
also `USE_TASMOTA_INTERMEDIATE = false`.

Zu guter letzt musst du noch die Inverter-Watts (in Sektion `[INVERTER_1]` und `[INVERTER_2]`) anpassen: `HOY_MAX_WATT` - wenn du zwei Inverter mit 1500W hast, dann passt das schon.

fatzo, 2023-05-12:
> Mal eins nach dem anderen:
> 
> stelle erstmal OpenDTU richtig ein:
> 
> ```
> [OPEN_DTU]
> # --- defines for OPEN-DTU ---
> OPENDTU_IP = 192.168.178.xx
> OPENDTU_USER = admin
> OPENDTU_PASS = xxxxxxxx
> ```
> 

erledigt

> dann dein Tasmota.
>
> ```
> [TASMOTA]
> # --- defines for Tasmota Smartmeter Modul---
> TASMOTA_IP = 192.168.178.73
> # the following three constants describes how to navigate through the Tasmota-JSON.
> # if [...]

mein Status {"StatusSNS":{"Time":"2023-05-12T21:47:59","M":{"counter_pos":4755.958,"power":493,"tariflos":176.72}}}

> Dann deaktiviere vorerst den Intermediate Zähler, den kannst du im Nachgang auch noch einrichten. also `USE_TASMOTA_INTERMEDIATE = false`.
> 
gemacht

> Zu guter letzt musst du noch die Inverter-Watts (in Sektion `[INVERTER_1]` und `[INVERTER_2]`) anpassen: `HOY_MAX_WATT` - wenn du zwei Inverter mit 1500W hast, dann passt das schon.

habe ich

danke schon mal für die hilfe

fatzo

reserve85, 2023-05-13:
Ok, dann musst du dein Tasmota so einstellen:

```
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = M
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = power
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
```

dann sollte es ohne Fehler laufen.

fatzo, 2023-05-13:
> Ok, dann musst du dein Tasmota so einstellen:
>
> ```
> TASMOTA_JSON_STATUS = StatusSNS
> # Prefix for Web UI and MQTT JSON payload
> TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = M
> # Power-MQTT label (the current [...]

ort/HoymilesZeroExport_Config.ini
Mai 13 09:10:06 raspberrypizero python3[31194]: 2023-05-13 09:10:06 INFO     Config file V 1.31
Mai 13 09:10:06 raspberrypizero python3[31194]: Traceback (most recent call last):
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/configparser.py", line 789, in get
Mai 13 09:10:06 raspberrypizero python3[31194]:     value = d[option]
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/collections/__init__.py", line 941, in __getitem__
Mai 13 09:10:06 raspberrypizero python3[31194]:     return self.__missing__(key)            # support subclasses that define __missing__
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/collections/__init__.py", line 933, in __missing__
Mai 13 09:10:06 raspberrypizero python3[31194]:     raise KeyError(key)
Mai 13 09:10:06 raspberrypizero python3[31194]: KeyError: 'hoy_compensate_watt_factor'
Mai 13 09:10:06 raspberrypizero python3[31194]: During handling of the above exception, another exception occurred:
Mai 13 09:10:06 raspberrypizero python3[31194]: Traceback (most recent call last):
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 781, in <module>
Mai 13 09:10:06 raspberrypizero python3[31194]:     HOY_COMPENSATE_WATT_FACTOR.append(config.getfloat('INVERTER_' + str(i + 1), 'HOY_COMPENSATE_WATT_FACTOR'))
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/configparser.py", line 823, in getfloat
Mai 13 09:10:06 raspberrypizero python3[31194]:     return self._get_conv(section, option, float, raw=raw, vars=vars,
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/configparser.py", line 808, in _get_conv
Mai 13 09:10:06 raspberrypizero python3[31194]:     return self._get(section, conv, option, raw=raw, vars=vars,
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/configparser.py", line 803, in _get
Mai 13 09:10:06 raspberrypizero python3[31194]:     return conv(self.get(section, option, **kwargs))
Mai 13 09:10:06 raspberrypizero python3[31194]:   File "/usr/lib/python3.9/configparser.py", line 792, in get
Mai 13 09:10:06 raspberrypizero python3[31194]:     raise NoOptionError(option, section)
Mai 13 09:10:06 raspberrypizero python3[31194]: configparser.NoOptionError: No option 'hoy_compensate_watt_factor' in section: 'INVERTER_2'
Mai 13 09:10:07 raspberrypizero systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mai 13 09:10:07 raspberrypizero systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mai 13 09:10:07 raspberrypizero systemd[1]: HoymilesZeroExport.service: Consumed 3.772s CPU time.
Mai 13 09:10:07 raspberrypizero systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 37.
Mai 13 09:10:07 raspberrypizero systemd[1]: Stopped HoymilesZeroExport Service.
Mai 13 09:10:07 raspberrypizero systemd[1]: HoymilesZeroExport.service: Consumed 3.772s CPU time.
Mai 13 09:10:07 raspberrypizero systemd[1]: Started HoymilesZeroExport Service.

reserve85, 2023-05-13:
Ich mache sie dir dann fertig wenn ich zu Hause bin

fatzo, 2023-05-13:
> Ich mache sie dir dann fertig wenn ich zu Hause bin

super 
ich bräuchte noch deinen namen auf facebook

reserve85, 2023-05-13:
Ich habe kein Facebook, geht auch hier.

Damit sollte es laufen. Kannst ja nochmal drüberschauen ob die Inverter-Werte passen (2 x 1500 Watt?).
Allerdings fehlten mir die Werte für die Sektion `[OPEN_DTU]`

Die musst du noch anpassen. Also:

```
[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = xxx.xxx.xxx.xxx
OPENDTU_USER = 
OPENDTU_PASS = 
```
musst du noch ergänzen.

[HoymilesZeroExport_Config.zip](https://github.com/reserve85/HoymilesZeroExport/files/11469253/HoymilesZeroExport_Config.zip)

fatzo, 2023-05-14:
Guten morgen

Der script scheint zu laufen ( wechselrichter passen sich an)
Aber immer noch eine fehlermeldung am anfang, später nicht mehr, kann man rausbekommen, woran es genau liegt?
In einem log oder so

Aktueller log

Mai 14 08:58:01 raspberrypizero python3[428]: 2023-05-14 08:58:01 INFO     powermeter Tasmota: -233  Watt
Mai 14 08:58:02 raspberrypizero python3[428]: 2023-05-14 08:58:02 INFO     powermeter Tasmota: -245  Watt
Mai 14 08:58:03 raspberrypizero python3[428]: 2023-05-14 08:58:03 INFO     powermeter Tasmota: -245  Watt
Mai 14 08:58:05 raspberrypizero python3[428]: 2023-05-14 08:58:05 INFO     powermeter Tasmota: -243  Watt
Mai 14 08:58:06 raspberrypizero python3[428]: 2023-05-14 08:58:06 INFO     powermeter Tasmota: -245  Watt
Mai 14 08:58:07 raspberrypizero python3[428]: 2023-05-14 08:58:07 INFO     powermeter Tasmota: -246  Watt
Mai 14 08:58:08 raspberrypizero python3[428]: 2023-05-14 08:58:08 INFO     powermeter Tasmota: -246  Watt
Mai 14 08:58:09 raspberrypizero python3[428]: 2023-05-14 08:58:09 INFO     powermeter Tasmota: -246  Watt
Mai 14 08:58:10 raspberrypizero python3[428]: 2023-05-14 08:58:10 INFO     powermeter Tasmota: 195  Watt
Mai 14 08:58:10 raspberrypizero python3[428]: 2023-05-14 08:58:10 INFO     setting new limit to 3000 Watt
Mai 14 08:58:10 raspberrypizero python3[428]: 2023-05-14 08:58:10 INFO     OpenDTU: Inverter "Hoymiles-HM1500": setting new limit from 75 Watt to 1500 Watt
Mai 14 08:58:13 raspberrypizero python3[428]: 2023-05-14 08:58:13 INFO     OpenDTU: Inverter "Hoymiles HM-1500G": setting new limit from 75 Watt to 1500 Watt
Mai 14 08:59:25 raspberrypizero python3[428]: 2023-05-14 08:59:25 INFO     OpenDTU: Inverter "Hoymiles-HM1500" reachable: True
Mai 14 08:59:25 raspberrypizero python3[428]: 2023-05-14 08:59:25 INFO     OpenDTU: Inverter "Hoymiles HM-1500G" reachable: False
Mai 14 08:59:25 raspberrypizero python3[428]: 2023-05-14 08:59:25 INFO     powermeter Tasmota: -413  Watt
Mai 14 08:59:26 raspberrypizero python3[428]: 2023-05-14 08:59:26 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:28 raspberrypizero python3[428]: 2023-05-14 08:59:28 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:29 raspberrypizero python3[428]: 2023-05-14 08:59:29 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:30 raspberrypizero python3[428]: 2023-05-14 08:59:30 INFO     powermeter Tasmota: -414  Watt
Mai 14 08:59:31 raspberrypizero python3[428]: 2023-05-14 08:59:31 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:32 raspberrypizero python3[428]: 2023-05-14 08:59:32 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:33 raspberrypizero python3[428]: 2023-05-14 08:59:33 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:34 raspberrypizero python3[428]: 2023-05-14 08:59:34 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:36 raspberrypizero python3[428]: 2023-05-14 08:59:36 INFO     powermeter Tasmota: -412  Watt
Mai 14 08:59:37 raspberrypizero python3[428]: 2023-05-14 08:59:37 INFO     powermeter Tasmota: -417  Watt
Mai 14 08:59:38 raspberrypizero python3[428]: 2023-05-14 08:59:38 INFO     powermeter Tasmota: -417  Watt
Mai 14 08:59:39 raspberrypizero python3[428]: 2023-05-14 08:59:39 INFO     powermeter Tasmota: -417  Watt
Mai 14 08:59:40 raspberrypizero python3[428]: 2023-05-14 08:59:40 INFO     powermeter Tasmota: -417  Watt
Mai 14 08:59:41 raspberrypizero python3[428]: 2023-05-14 08:59:41 INFO     powermeter Tasmota: -417  Watt
Mai 14 08:59:42 raspberrypizero python3[428]: 2023-05-14 08:59:42 INFO     powermeter Tasmota: -417  Watt
Mai 14 08:59:43 raspberrypizero python3[428]: 2023-05-14 08:59:43 INFO     powermeter Tasmota: -645  Watt
Mai 14 08:59:45 raspberrypizero python3[428]: 2023-05-14 08:59:45 INFO     powermeter Tasmota: -645  Watt
Mai 14 08:59:46 raspberrypizero python3[428]: 2023-05-14 08:59:46 INFO     powermeter Tasmota: -645  Watt
Mai 14 08:59:47 raspberrypizero python3[428]: 2023-05-14 08:59:47 INFO     powermeter Tasmota: -638  Watt

Kann man noch irgendwo einstellen, das er weniger ins netz einspeißt?

gruß
frank

fatzo, 2023-05-14:
Vielen dank schon mal an alle!
Ich muss zwar noch optimieren, aber erstmal läuft er, zwar ein wenig träge, kann aber auch am Zero liegen.
ich habe auch das Problem das OpenDtu den zweiten Wechselrichter nicht immer erreicht, obwohl beidenebeneinander stehen.

gruß an alle

fatzo

closed 2023-05-14 by fatzo

reserve85, 2023-05-14:
Welche Fehlermeldung am Anfang?