reserve85/HoymilesZeroExport #57: Homeassistent als Powermeter - Fehler

checkit99out, 2023-05-15:
Hallo! Zuerst meine Hardware-config (zum Verständnis: Ich nutze OpenDTU und als Messgeräte zwei Shelly 3EM. Einer in einer Unterverteilung, wo auch der WR angeschlossen ist. Der zweite Shelly im Sicherungskasten, allerdings hinter dem Abzweig ( er kann also keine negativen Werte!).
Aus dem GRunde rechntet HA in der Entität "sensor.power_ges" alles zusammen und gibt bei Einspeisung größer Bezug auch negative Werte aus (wie der Zähler).
Darum auch HA als Powermeter in der config.ini.
Doch leider gibt das Skript Fehlermeldungen, so dass es keine Regelung schafft. 

Auszug aus dem Log (Fehlermeldungen):

2023-05-14 20:28:07 ERROR    Exception at GetPowermeterWatts
2023-05-14 20:28:07 ERROR    invalid literal for int() with base 10: '1428.35'
2023-05-14 20:28:07 ERROR    invalid literal for int() with base 10: '1428.35'
2023-05-14 20:28:27 INFO     ---Start Zero Export---
2023-05-14 20:28:28 INFO     OpenDTU: Inverter "HM1500" reachable: True
2023-05-14 20:28:28 ERROR    Exception at GetPowermeterWatts
2023-05-14 20:28:28 ERROR    invalid literal for int() with base 10: '1121.91'
2023-05-14 20:28:28 ERROR    invalid literal for int() with base 10: '1121.91'

Man sieht also, DTU ist erreichbar, Er bekommt auch die bezogenen Watt (1428 Watt), aber gibt Fehlermeldungen und rechnet kein neues Limit aus.

Auszug meiner config.ini:

[HOMEASSISTANT]
# --- defines for HOME ASSISTANT (you need to create a Long-Lived Access Token in your profile) ---
HA_IP = 192.168.50.223
HA_PORT = 8123
HA_ACCESSTOKEN = eyJhbGciOiJIUzI1NiIsInR5cxxxxxxxxxx.eyJpc3MiOiI3MTFiNDgyMGFmN2E0NTZjYjhhYzg4YzQ3NWRkYmVlYyIsImlhdCI6MTY4NDA4NjMxMxxxxxxxTk5NDQ2MzEzfxxxxxDt_V99CsplPe6rb3jDjEhloEJ5rgXM7SUA
HA_CURRENT_POWER_ENTITY = sensor.power_ges
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
HA_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
HA_POWER_INPUT_ALIAS = sensor.gesamt_strom
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
HA_POWER_OUTPUT_ALIAS = sensor.gesamt_strom

Jemand ne Idee?

reserve85, 2023-05-15:
Hi,

kannst du mal folgende Zeile im Code ändern:

Zeile 572:
`Watts = int(ParsedData['state'])`
in 
`Watts = int(float(ParsedData['state']))`

Gruß

checkit99out, 2023-05-15:
ok, sieht besser aus.
023-05-15 10:36:57 INFO     OpenDTU: Inverter "HM1500" reachable: True
2023-05-15 10:36:57 INFO     powermeter HomeAssistant: 306  Watt
2023-05-15 10:36:57 INFO     setting new limit to 1500 Watt
2023-05-15 10:36:57 INFO     OpenDTU: Inverter "HM1500": setting new limit from 1500 Watt to 1500 Watt
2023-05-15 10:37:17 INFO     OpenDTU: Inverter "HM1500" reachable: True
2023-05-15 10:37:17 INFO     powermeter HomeAssistant: 304  Watt
2023-05-15 10:37:17 INFO     setting new limit to 1500 Watt
2023-05-15 10:37:17 INFO     OpenDTU: Inverter "HM1500": setting new limit from 1500 Watt to 1500 Watt
2023-05-15 10:37:37 INFO     OpenDTU: Inverter "HM1500" reachable: True
2023-05-15 10:37:37 INFO     powermeter HomeAssistant: 156  Watt
2023-05-15 10:37:37 INFO     setting new limit to 1500 Watt
2023-05-15 10:37:37 INFO     OpenDTU: Inverter "HM1500": setting new limit from 1500 Watt to 1500 Watt

nun ist "nur" noch das Limit zu hoch. Aber der Shelly im Außengebäude verliert heute auch öfter mal die Funkverbindung, vlt. führt das zur Fehlberechnung?
Genau! Zwischendurch diese Werte:
2023-05-15 10:42:43 INFO     powermeter HomeAssistant: -147  Watt
2023-05-15 10:42:44 INFO     powermeter HomeAssistant: -147  Watt
2023-05-15 10:42:45 INFO     powermeter HomeAssistant: -147  Watt
2023-05-15 10:42:46 INFO     overproducing: reduce limit based on previous limit setpoint by approximation
2023-05-15 10:42:46 INFO     setting new limit to 251 Watt
2023-05-15 10:42:46 INFO     OpenDTU: Inverter "HM1500": setting new limit from 309 Watt to 251 Watt
2023-05-15 10:42:51 INFO     OpenDTU: Inverter "HM1500" reachable: True
2023-05-15 10:42:51 INFO     powermeter HomeAssistant: -145  Watt
2023-05-15 10:42:52 INFO     powermeter HomeAssistant: -145  Watt
2023-05-15 10:42:53 INFO     powermeter HomeAssistant: -145  Watt

Bei guter Wlan-VErbindung scheint er also zu regulieren.

DANKE!

checkit99out, 2023-05-15:
Gerne!  Habe ja auch ein Interesse dran, dass es funktioniert!
Das einzige - für mich ungewöhnlich - ist, dass Dein Skript offensichtlich bei "unverproducing" auf 100% des WR stellt, um es dann wieder angepasst runterzuregeln.
Scheint vermutlich weniger rechenintensiv/datenintensiv zu sein, als langsam angepasst hoch zu steppen....
Dennoch guter Gedankengang!

checkit99out, 2023-05-15:
...ups, habe ich in der ini übersehen. Aber den Gedanken dahinter verstanden ;-)
Wie gesagt, die Anpassung zum HA als Powermeter läuft! Danke nochmals

closed 2023-05-22 by reserve85