reserve85/HoymilesZeroExport #58: Fehler mit zwei WR unter Homeassistent

checkit99out, 2023-05-18:
Nadem das Skript echt gut mit EINEM WR lief, habe ich mal den zweiten WR mit eingegben.

Resultat: Fehler.

Hier mal ein paar Auszüge, vielleicht hilft es - Ich habe weder Ahnung von Python, noch behalte ich den Überblick bei derart vielen Zeilen ;-)

Aber reserve85 möchte ja auch, dass sein Skript getestet wird - also tue ich es gerne ;-)

[COMMON]
# Number of Inverters
INVERTER_COUNT = 2
# max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
# if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20

[INVERTER_2]
# maximum limit in watts (100%)
HOY_MAX_WATT = 1500
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48

Log:
2023-05-18 11:51:39.124514 INFO AppDaemon: Loading App Module: /config/appdaemon/apps/HoymilesZeroExport.py
2023-05-18 11:51:39 INFO     Log write to file: False
2023-05-18 11:51:39 INFO     Author: Tobias Kraft / Script Version: 1.32
2023-05-18 11:51:39 INFO     read config file: /config/appdaemon/apps/HoymilesZeroExport_Config.ini
2023-05-18 11:51:39 INFO     Config file V 1.31
2023-05-18 11:51:39.207456 WARNING Error: ------------------------------------------------------------
2023-05-18 11:51:39.208025 WARNING Error: Unexpected error loading module: /config/appdaemon/apps/HoymilesZeroExport.py:
2023-05-18 11:51:39.208702 WARNING Error: ------------------------------------------------------------
2023-05-18 11:51:39.220116 WARNING Error: Traceback (most recent call last):
  File "/usr/lib/python3.11/configparser.py", line 805, in get
    value = d[option]
            ~^^^^^^^^
  File "/usr/lib/python3.11/collections/__init__.py", line 1004, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/collections/__init__.py", line 996, in __missing__
    raise KeyError(key)
KeyError: 'hoy_compensate_watt_factor'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 1000, in check_app_updates
    await utils.run_in_executor(self, self.read_app, mod["name"], mod["reload"])
  File "/usr/lib/python3.11/site-packages/appdaemon/utils.py", line 304, in run_in_executor
    response = future.result()
               ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/appdaemon/app_management.py", line 783, in read_app
    self.modules[module_name] = importlib.import_module(module_name)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/appdaemon/apps/HoymilesZeroExport.py", line 781, in <module>
    HOY_COMPENSATE_WATT_FACTOR.append(config.getfloat('INVERTER_' + str(i + 1), 'HOY_COMPENSATE_WATT_FACTOR'))
                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/configparser.py", line 839, in getfloat

DEr FEhler scheint also in Line 805 aufzutreten und setzt sich folglich fort?

ab Line 804:
while True:
    try:
        PreviousLimitSetpoint = newLimitSetpoint
        if GetHoymilesAvailable() and GetCheckBattery():
            if LOG_TEMPERATURE:
                GetHoymilesTemperature()
            for x in range(int(LOOP_INTERVAL_IN_SECONDS / POLL_INTERVAL_IN_SECONDS)):
                powermeterWatts = GetPowermeterWatts()
                if powermeterWatts > POWERMETER_MAX_POINT:
                    if JUMP_TO_MAX_LIMIT_ON_GRID_USAGE:
                        newLimitSetpoint = GetMaxWattFromAllInverters()
                    else:

reserve85, 2023-05-18:
Servus, da fehlt der Eintrag 
HOY_COMPENSATE_WATT_FACTOR = 1
in der Sektion [Inverter_2] in der config.

Ich hab das in der DEV schon ergänzt.

checkit99out, 2023-05-18:
....und so läuft es... Great job!

closed 2023-05-18 by reserve85