reserve85/HoymilesZeroExport #61: Shelly 3 EM mit Passwortabfrage in der Config

XpertXXL, 2023-05-25:
könnte man beim Shelly 3 EM auch noch die Option des Passwort Schutzes umsetzen. (Abfrage in der HoymilesZeroExport_Config.ini)
Ich habe alle meine Shelly's normalerweise Passwort geschützt.
Danke auch für das tolle Script!

XpertXXL

reserve85, 2023-05-25:
Hi,

gute Idee.
Habe es mal in der DEV implementiert: https://github.com/reserve85/HoymilesZeroExport/tree/dev

Ich konnte es mit einem Shelly 1PM testen, sollte also auch mit dem 3EM klappen.
Leider habe ich keinen Shelly der 2. Generation greifbar, (PLUS_1PM, 3EM_PRO) kann das ggf. jemand testen? Habe es zwar wie in der API beschrieben umgesetzt aber man weiß ja nie...

XpertXXL, 2023-05-25:
mit 3 EM getestet, klappt :-)
vielen Dank

reserve85, 2023-05-25:
top, @onceuponatime78 kannst du das ggf. auch mit deinem Shelly Pro 3EM testen (falls du das Script noch laufen hast...)

onceuponatime78, 2023-05-26:
hab es kurz ausprobiert und es funktioniert auf Anhieb nicht - der Unterschied den ich auf die schnelle zwischen den shelly generationen erkennen konnte ist, dass bei den Pro Geräten nur ein Passwort gesetzt wird, bei den älteren user+passwort

reserve85, 2023-05-26:
ja hab grad geschaut, der User ist immer "admin" bei den GEN2 Produkten.
Kannst du das dann mal im Browser probieren: http://admin:PASS@xxx.xxx.xxx.xxx/rpc/EM.GetStatus?id=0

Gruß Tobi

onceuponatime78, 2023-05-26:
mit aktivierter passwortauthentifizierung: {"code":404,"message":"No handler for EM.GetStatus/"}

reserve85, 2023-05-26:
Mit welchem Shelly hast du das getestet? Pro 3EM oder 1PM Plus?

onceuponatime78, 2023-05-26:
mit dem pro 3EM - mit deaktiviertem PW wird der Status zurückgeliefert

reserve85, 2023-05-26:
Ein Versuch hätte ich noch, falls du nochmal dazu kommst -> https://github.com/reserve85/HoymilesZeroExport/tree/dev
Wäre echt top. in der config dann "admin" und dein Passwort.

delacor, 2023-05-29:
Ich kriegs nicht ganz zum laufen, vermute aber der Fehler liegt bei mir.
habe auch den shelly 3EM Pro

```
INFO     OpenDTU: Inverter "HM-1500" reachable: True
INFO     OpenDTU: Inverter "HM-1500" temperature: 33.1 degC
ERROR    Exception at GetPowermeterWatts
ERROR    Expecting value: line 1 column 1 (char 0)
ERROR    Expecting value: line 1 column 1 (char 0)

``` 

``` along with this program.  If not, see <http://www.gnu.org/licenses/>.                                                                                                                                                                                                                                                                                      
                                                                                                                                                                                                                                                                                                                                                             
[VERSION]                                                                                                                                                                                                                                                                                                                                                    
VERSION = 1.33                                                                                                                                                                                                                                                                                                                                               
                                                                                                                                                                                                                                                                                                                                                             
[SELECT_DTU]                                                                                                                                                                                                                                                                                                                                                 
# --- define your DTU (only one) ---                                                                                                                                                                                                                                                                                                                         
USE_AHOY = false                                                                                                                                                                                                                                                                                                                                             
USE_OPENDTU = true                                                                                                                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                                                                                                             
[SELECT_POWERMETER]                                                                                                                                                                                                                                                                                                                                          
# --- define your Powermeter (only one) ---                                                                                                                                                                                                                                                                                                                  
USE_TASMOTA = false                                                                                                                                                                                                                                                                                                                                          
USE_SHELLY_3EM = false                                                                                                                                                                                                                                                                                                                                       
USE_SHELLY_3EM_PRO = true                                                                                                                                                                                                            [... cut ...]                                                                                                              
HOY_BATTERY_REDUCE_WATT = 300                                                                                                                                                 
# reenable inverter at this voltage                                                                                                                                           
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51

reserve85, 2023-05-29:
Hey,
Hast du ein Passwort eingestellt bei dem Shelly 3EM Pro? Es kann sein dass das nämlich noch nicht funktioniert, da ich keinen habe konnte es nicht testen.
-> Abhilfe: entferne das Passwort bei dem 3EM pro (und setze den Username / Passwort auf leer in der Ini), dann sollte es klappen. Die MAIN Version sollte ebenfalls funktionieren.

delacor, 2023-05-29:
Achso ja alles gut, mir ging es eher darum dir dabei zu helfen, leider ist das logfile nicht sehr aussagekräftig. Ich Versuche nachher genauer herauszufinden, wo es hakt.

reserve85, 2023-05-29:
Ahso ok - danke schonmal! 
scheint also nicht zu klappen? ich kapiere es nicht so richtig, ohne codebeispiel in Python bekomme ich es irgendwie nicht hin. Siehe https://shelly-api-docs.shelly.cloud/gen2/0.14/General/Authentication#authentication

delacor, 2023-05-29:
Jau, das ist etwas komplizierter, da die das ganzen hashen wollen. Ich habe [hier](https://github.com/home-assistant-libs/aioshelly/pull/209/commits/ed6b9a14c96c9cbf208321153142a0a89c0bb1e5) was gefunden, ist auch für Gen2, könnte man so kopieren.

reserve85, 2023-05-29:
Jop das hab ich auch vorhin gesehen. Halte das aber etwas oversized, dann verzichte ich lieber auf die authentication.

delacor, 2023-05-29:
Got it.
```python
import requests
>>> from requests.auth import HTTPDigestAuth
>>> url = 'http://10.4.20.9/rpc/EM.GetStatus?id=0'
>>> requests.get(url, auth=HTTPDigestAuth('admin', '8chars1number'))
<Response [200]>

```

reserve85, 2023-05-29:
Leck am arsch, bin heute unterwegs ich implementiere es heute Abend! 1000 Dank.

delacor, 2023-05-29:
hab dir https://github.com/reserve85/HoymilesZeroExport/pull/64 schon reingedrückt :-)

reserve85, 2023-05-30:
ist in der MAIN, vielen Dank!

closed 2023-05-30 by reserve85