reserve85/HoymilesZeroExport #65: intermediate meter

fatzo, 2023-05-30:
Moin zusammen
Damit sind ja die zwischenzähler gemeint.

Ich habe pro Solarstrang je einen Shelly 1PM dazwischen,

wie bringe ich beide in die Config?

reserve85, 2023-05-30:
Hi,
geht so erstmal nicht - es ist nur ein Gesamtzähler als Zwischenzähler vorgesehen. 
Als option könntest du die Leistung der 1PM Shellys mittels HomeAssistant / IO Broker aufaddieren und den Gesamtwert dann als Intermediate meter auszulesen.

fatzo, 2023-05-31:
> Hi, geht so erstmal nicht - es ist nur ein Gesamtzähler als Zwischenzähler vorgesehen. Als option könntest du die Leistung der 1PM Shellys mittels HomeAssistant / IO Broker aufaddieren und den Gesamtwert dann als Intermediate meter auszulesen.

fatzo, 2023-05-31:
Vielen dank das hat mir geholfen.

closed 2023-05-31 by fatzo