reserve85/HoymilesZeroExport #67: Schritt für Schritt Anleitung für Windows

tthnn, 2023-06-01:
Hi . Wäre es dir möglich mal eine Schritt für Schritt Anleitung für Windows zu machen ? 
leider krieg ich das Script nach wie vor nicht zum laufen . Er findet immer die config nicht zumindest versucht er sich mit tasmota zu verbinden obwohl ich dort false und
Bei Shelly True geschrieben habe . 
toll wäre mal eine Anleitung

tthnn, 2023-06-01:
[image]
[image]
[image]
[image]
[image]
[image]

tthnn, 2023-06-01:
Beim Start des Skriptes schließt es sich sofort wieder .

tthnn, 2023-06-01:
Und editieren tuh ich die Konfiguration Datei mit dem Edition

reserve85, 2023-06-01:
Du hast bei [COMMON] / INVERTER_COUNT = 2 eingegeben aber unten nur ein Inverter konfiguriert. Ändere das auf 1 oder füge den zweiten Inverter am Ende hinzu.

tthnn, 2023-06-01:
Ahh jetz gehts . Aber den zweiten Inverter will ich natürlich zufügen . Wie meinst du jetz den zweiten unten zu fügen ?

reserve85, 2023-06-01:
ahso, jetzt sehe ich es.
ganz unten sind die ";" vor den Zeilen, das bedeutet "auskommentiert" - entferne die Semikolons und dann sollte es laufen.

closed 2023-06-12 by reserve85