reserve85/HoymilesZeroExport #68: Shelly EM einbinden

robson1311, 2023-06-02:
Hallo Tobias, vorerst mal Danke für deine Arbeit und das dies jedermann nutzen darf!

Bin selbst Neuling auf dem Gebiet, und deshalb leider nicht Sattelfest.
Ich habe dein Script in Verbindung mit einem HM-1200, einer OpenDTU und einem ShellyEm auf einem Proxmox am laufen.
Die Kommunikation mit der OpenDTU dürfte funktionieren, doch ich vermute am Shelly scheitert es.
Bekomme diese Fehlermeldung:
Jun 01 21:51:18 Nulleinspeisung python3[443]: 2023-06-01 21:51:18 INFO OpenDTU: Inverter "HM-1200" reachable: True
Jun 01 21:51:18 Nulleinspeisung python3[443]: 2023-06-01 21:51:18 ERROR Exception at GetPowermeterWatts
Jun 01 21:51:18 Nulleinspeisung python3[443]: 2023-06-01 21:51:18 ERROR 'total_power'
Jun 01 21:51:18 Nulleinspeisung python3[443]: 2023-06-01 21:51:18 ERROR 'total_power'
und das wiederholt sich ständig.
Dabei wird der HM auf 5% (60 Watt) runtergeregelt.

Wollte den ShellyEM über die .ini irgendwie einbinden, doch fehlt mir hier leider das Wissen.
Oder funktioniert der EM generell für dieses Vorhaben nicht?

Hättest du bitte eine Idee bzw. Lösung?!

Vielen Dank!

reserve85, 2023-06-02:
Hi Robson,

sieht ja erstmal ganz gut aus. Könntest du bitte die API Abfrage im Browser eintippen und posten was da zurückkommt:

http://SHELLY3EM_IP/status
SHELLY3EM_IP natürlich durch die Shelly IP Adresse ersetzen

robson1311, 2023-06-02:
Ohh, vielen Dank!
Bin noch in der Arbeit, werde das Script schnellstmöglich ersetzen.
Ich berichte dann.

LG Robert

reserve85, 2023-06-03:
beste Qualität :-)

robson1311, 2023-06-04:
So, schaut gut aus, das Script läuft und der ShellyEM den du implementiert hast, wird perfekt gelesen! Danke!
Leider hab ich da doch noch ein paar Fragen zum Script, bzw. wenn ich in der ini Datei Einstellungen verändern möchte.

Unter Windows würde ich mir den Ordner suchen in dem das Programm läuft und könnte dabei die ini verändern, aber unter Linux?!
Habe auf deiner Seite gestöbert und einige Kommentare durchgelesen und konnte dabei von dir folgendes finden:

>Terminal öffnen
cd HoymilesZeroExport
sudo nano HoymilesZeroExport.py
in der 1. Zeile 1. Buchstabe springen
ALT+A
danach mit
ALT+DOWN
bis zum Ende springen und alles dadurch markieren,
dann mit
STRG+K
alles ausschneiden,
dann den Inhalt der neuen Datei kopieren und mit Rechtsklick ins Terminal einfügen,
speichern mit
STRG+S
beenden mit
STRG+X
Für die Konfiguration das gleiche, die Werte vorher schnell und schmutzig abfotografieren.
Zum Schluss dann mit
sudo ./restart
dem Script neu 

Also so komme ich vermutlich in den HoymilesZeroExport Ordner im Linux.
 
Nur**Zur Zeit läuft das Ausgabeprotokoll**
sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f

Es klingt komisch, doch wie kann ich dieses beenden um die obigen Befehle einzugeben, um am Script etwas zu ändern?
Im Protokoll kann ich schreiben was ich will, da tut sich nix.

LG Robert

reserve85, 2023-06-04:
Super dass es läuft, dann kann ich es nach Main schieben.

Du kannst die Logausgabe mit Strg + c beenden und bist dann wieder im Terminal.

robson1311, 2023-06-04:
Oh Mann, so einfach würde es gehen.
Vielen Dank!!!! :-)

closed 2023-06-05 by reserve85