reserve85/HoymilesZeroExport #70: Tool zum Erstellen der config.ini

Ollipop030, 2023-06-05:
Moin Tobias,

ich lese hier immer fleissig mit, einige Leute scheinen öfter mal Probleme beim Einrichten zu haben. Das Script ist auch mittlerweile echt gewachsen, als Neueinsteiger ist man vielleicht etwas überfordert.

Hier ne Idee:
Wäre ein Tool zum Erstellen der Config vielleicht hilfreich? Ein kleines Script mit einer Art Interview-Modus:

_**Welche DTU:**_
**_1= Ahoy
2= OpenDTU_**
1

_**Zwischenzähler vorhanden ((j/n)?:**_
j

_**IP der DTU:**_
192.168.1.1

**_IP des Zwischenzählers:_**
192.168.1.2

_**Wie viele Inverter:**_
2

usw.

Nur für die Basics, für Intervalle oder Ähnlichen können die Leute ja immer noch die *.ini selbst editieren.

Sieh das hier bitte nur als Idee, programiertechnisch kann ich leider keine Hilfe leisten.

delacor, 2023-06-05:
Ich kann mir das Mal ansehen, was mich viel mehr nervt, ist dass ich config Inhalte nicht löschen kann, weil er danach sucht, selbst wenn intermediate z.b. auf OFF steht. Das bläht die config einfach extrem auf.

reserve85, 2023-06-05:
Ja beides natürlich möglich, ein Tool wäre mir persönlich erstmal egal - ggf. könnte man die Kommentare ja aussagekräftiger gestalten. Denke daran scheitert es am häufigsten. Manche Fragen nach "Handbuch" - ich wüsste gar nicht was ich da reinschreiben soll.
Es ist halt ein Script welches händisch konfiguriert werden muss, jeder der mit einem Raspi oder im entferntesten mit programmieren umgehen kann sollte das hinbekommen?! 

Dass alles in der Ini vorhanden sein muss hab ich absichtlich erstmal so vorausgesetzt - ich weiss genau, dass dann die ersten User kommen und sagen "ich habe seit heute einen Shelly 1PM, was muss ich einstellen". 

Keine Ahnung was am besten ist. 🌝

fatzo, 2023-06-07:
> Ja beides natürlich möglich, ein Tool wäre mir persönlich erstmal egal - ggf. könnte man die Kommentare ja aussagekräftiger gestalten. Denke daran scheitert es am häufigsten. Manche Fragen nach [...]

leider kann nicht jeder programieren und ich selber habe auch meine Probleme obwohl ich mich seit dem 8086 mit Computern beschäftige, aber durch viel lesen kann man das hinbekommen. Vieleicht wäre eine übersetzung der ini schon sinnvoll (in englsch steht ja schon alles)
> 
> Dass alles in der Ini vorhanden sein muss hab ich absichtlich erstmal so vorausgesetzt - ich weiss genau, dass dann die ersten User kommen und sagen "ich habe seit heute einen Shelly 1PM, was muss ich einstellen".

das ist auch sinnvoll so, den es kommen immer mehr Hausautomation dazu.
> 
> Keine Ahnung was am besten ist. 🌝

wie du schon sagst, es gibt immer einen dau

Duch machst hier einen super job und kümmerst dich um Fragen.
Vielen Dank dafür!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

fatzo

closed 2023-07-24 by reserve85