reserve85/HoymilesZeroExport #72: Mehrere Wechselrichter

Kante001, 2023-06-08:
Hallo,

ich hoffe man kann hier noch schreiben ? Erstmal vorne weg deine Regelung für einen Wechselrichter funktioniert super. Danke dafür.

Ich habe jetzt 4 Anlagen in Betrieb, das geht auch mit 4 Protokollen. Ich habe jetzt nur das Problem das sich über die Nacht 3 von 4 Protokollen aufhängen. Manchmal auch einzelne während des Tages.

Hast du da eine Idee woran das liegen könnte ?

Bzw hast du schon das für mehrere WR geschrieben worüber du nachgedacht hattest ?

Viele Grüße

reserve85, 2023-06-08:
Hi, ja das ist implementiert. Die aktuelle Version unterstützt schon seit längerem mehrere WR. Du kannst diese in der config deklarieren.

Kante001, 2023-06-08:
Hallo,super habe mir gerade die Datei geladen.Muss aber nochmal dumm nachfragen. Gebe ich die Seriennummer der Inverter da nicht an ? Und wenn ich bei Common - Inverter Count 4 für die 4 Inverter an. Kopiere ich dann bei list of inverters, based on Common/Count - Inverter 1 und füge das darunter ein und nenne es dann Inverter 2  und so weiter ? Sorry für die dumme Frage aber irgendwie erschließt sich mir das schon wieder nicht. 😭

reserve85, 2023-06-08:
Ja genau so, in der Reihenfolge wie in Ahoy angegeben. Die Seriennummern werden dann automatisch aus Ahoy gelesen.

Kante001, 2023-06-08:
OK nutze zwar dtu aber wird ja genauso funktionieren. Muss ich die Inverter in der dtu mit Inverter 1 nennen oder können die andere Namen haben ?

reserve85, 2023-06-08:
Super dass alles läuft. Ich würde die alte Version löschen, aber ich weiß ja nicht wie du das installiert hattest.

closed 2023-06-12 by reserve85

Kante001, 2023-06-14:
Hallo,Also zu der Frage mit dem deinstallieren. Habe ich nach der Anleitung gemacht. Im Ordner ist aber kein uninstall.sh dabei wie bei der neuen Version. Aufrufen kann ich den Status mit sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f Vielleicht sagt dir das ja was wie ich den Service wieder löschen kann ?

reserve85, 2023-06-14:
führe einfach die Befehle aus dem uninstall.sh script aus dann ist es weg.
Hattest du die alte version denn in einen anderen Ordner installiert?

Kante001, 2024-04-14:
Hallo. Ich habe mal wieder ein Problem. Habe die Open dtu geupdatet. Jetzt schreibt er mir

Error exception at gethoymilesactualpower
Error 'AC'
Error 'AC'
Dann fragt er wieder den Shelly ab und dann wieder Error exception...

Habe das wenn ich den resperi neustarte und das Skript lädt geht es drosselt die WR. Wenn ich einmal Strom erhöhe macht er auch wieder die WR auf 100 Prozent und dann wenn er wieder drosseln will kommt die Fehlermeldung.

reserve85, 2024-04-14:
Ja du musst das Script aktualisieren, es gab seitens der OpenDTU Änderungen an der API Schnittstelle.

Kante001, 2024-04-14:
OK wollte das neue schon mal laden aber da waren so viele Einstellung mehr als bei dem alten deswegen wollte ich da bleiben aber gut dann muss ich da noch mal Ran 😅. Oder kann man das Skript einfach so akualisieren das die Einstellungen übernommen werden ?

reserve85, 2024-04-14:
Ich habe in der neuen Version einen updater hinzugefügt , das funktioniert aber nur wenn du deine Einstellungen in die config_overwrite.ini einträgst. Daher würde ich dir empfehlen deine aktuelle config sicherheitshalber zu kopieren und das Script komplett neu zu installieren wie in der readme beschrieben.

Kante001, 2024-04-14:
Hab ich probiert jetzt das Problem. Hoffe er sendet Bilder ?

Kante001, 2024-04-14:
Hat das mit dem Bild geklappt ?

reserve85, 2024-04-14:
Nee, ich sehe nix.

Kante001, 2024-04-14:
Ich probiere es mal nochmal.

Ansonsten startet er den Service.
Und gibt dann Fehler
From packaging import Version
ModuleNotFoundError: No Module named 'packaging'
Hoymileszeroexport.service Main process exited , Code=exited , Status=1/failure
Hoymileszeroexport.service faild with result Exit code

Und das selbe dann noch paar Mal

Kante001, 2024-04-14:
Falls die E-Mails mit den Bildern gefiltert werden noch mal in Text Form.

Ansonsten startet er den Service.
Und gibt dann Fehler
From packaging import Version
ModuleNotFoundError: No Module named 'packaging'
Hoymileszeroexport.service Main process exited , Code=exited , Status=1/failure
Hoymileszeroexport.service faild with result Exit code

Und das selbe dann noch paar Mal

reserve85, 2024-04-14:
Geh einfach nach Anleitung vor, dann installiert sich das alles automatisch mit?

Kante001, 2024-04-14:
Hab ich gemacht nach Anleitung. Kann ich gleich noch mal probieren. Muss ich das alte deinstallieren ? Also den ./uninstall hab ich gemacht aber sonst noch etwas ?

reserve85, 2024-04-14:
Ne, kannst aber den ganzen Ordner löschen. Hast du die Datei update.sh im Ordner? Dann führe das mal aus. Gibt es da Fehlermeldungen?

Kante001, 2024-04-14:
Ist das normal ich habe den uninstall_service ausgeführt und den Ordner gelöscht und neu gestartet das er trotzdem nach dem Neustart sudo journalctl trotzdem noch das gleiche wie vorher steht mit dem Fehler was ich vorhin geschrieben habe. Hätte jetzt erwartet das er den Service garnicht mehr findet.

Kante001, 2024-04-14:
So habe nochmal probiert neu zu installieren.

Erster Teil klappt gut. Bei sudo chmod +x... Kommt dann Error : externally-managed-enviroment

Danach macht er weiter und am ende wieder der Fehler modulenotfounderror

Kante001, 2024-04-14:
Und das die config_override leer ist soll das so sein ?

Kante001, 2024-04-14:
Ich sehe gerade noch das

Python3 cant Open File .... Hoymileszeroexport.py Error 2 no such File or directory

Python3 ist in Version 3.11.2 installiert

Kante001, 2024-04-14:
Habe nochmal python3 installiert und neu gestartet jetzt macht er den alten Fehler wieder diesen modulenotfounderror schreibt.

Ich entschuldige mich vielmals für die nervigen E-Mails

reserve85, 2024-04-14:
Sorry ich bin leider auch nicht am pc…
Die overwrite ist immer leer, da musst du deine inverter Daten dann reinkopieren. Also alles was du ursprünglich in der config geändert hattest muss da rein. 

Aber dein Problem ist das externally managed. Schau mal hier: https://github.com/reserve85/HoymilesZeroExport/issues/173
Bzw. die Lösung von @lullatscho
https://github.com/reserve85/HoymilesZeroExport/issues/137#issuecomment-1942654871

Kante001, 2024-04-14:
Super vielen Dank. Hab die Anleitung befolgt und hat geklappt. Morgen mal abwarten ob die Steuerung funktioniert. Schönen Abend noch und Entschuldigung für die vielen E-Mails

#137 (comment)<https://github.com/reserve85/HoymilesZeroExport/issues/137#issuecomment-1942654871>

reserve85, 2024-04-15:
gerne, du kannst ab sofort dann auch das "update.sh" ausführen wenn eine neue Version erscheint.

reserve85, 2024-04-15:
und, wie schauts?

Kante001, 2024-04-15:
Guten morgen, also ich wollte das noch ein bisschen beobachten aber bist jetzt 100 Prozent alles am laufen. Vielen Dank noch mal für die unterlerstützung. Der Ablauf sollte auf jeden Fall mit in die Anleitung. Hat auch super geklappt mit den Codes. Hab bis 23 Uhr dran gesessen aber Ergebnis passt.

reserve85, 2024-04-15:
ja ich werde das verlinken, hatten ja relativ viele das Problem...