reserve85/HoymilesZeroExport #73: OpenDTU mit HMS Wechselrichter

mark82436, 2023-06-10:
Hallo Tobi,

ich wollte Dir Rückmeldung geben ich habe jetzt seit 3 Wochen
Dein Script mit einer OpenDTU-S und 2 Hoymiles Wechselrichter der HMS Serie (1x HMS-1800, 1x HMS-2000) am Laufen.
Habe gerade auf 1.39 umgestellt, läuft ohne Probleme.

Besten Dank für das tolle Script!

mark82436, 2023-06-11:
Eine Frage hätte ich noch dazu.

Ich betreibe die 2 WR an 2 Phasen mit jeweils einem Tasmota Zwischenstecker, gibt es die Möglichkeit
im Script auch mit 2 Tasmota Zwischenzählern zu arbeiten?

reserve85, 2023-06-11:
Nee gibts nicht, aber du kannst beide Werte z.B. in HomeAssistant oder ioBroker zusammenfassen und darüber drauf zugreifen. Es läuft aber auch sehr gut ohne Zwischenzähler.

mark82436, 2023-06-11:
ok danke für die Rückmeldung, dann lasse ich die Zwischenzähler weg
läuft auch so ganz gut, nur mit der OpenDTU als Zähler.

closed 2023-06-12 by reserve85