reserve85/HoymilesZeroExport #74: ERROR Exception HTTPConnectionPool und Anhalten des H..ZeroExp. Services

SMGatAUXde, 2023-06-13:
Erst mal Vielen Dank an reserve85 für die super Arbeit.

Ich betreibe das Script im Battery Mode, nun seit drei Tagen. Leider nicht ganz unterbrechungsfrei.

Vor zwei Tagen hat das Script mit folgendem Fehler einfach ausgesetzt: 

2023-06-11 12:36:36,089 ERROR    Exception at GetPowermeterWatts
2023-06-11 12:36:36,090 ERROR    HTTPConnectionPool(host='192.168.178.23', port=80): Max retries exceeded with url: /status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb59f4850>: Failed to establish a new connection: [Errno 113] No route to host'))
2023-06-11 12:36:36,090 ERROR    HTTPConnectionPool(host='192.168.178.23', port=80): Max retries exceeded with url: /status (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xb59f4850>: Failed to establish a new connection: [Errno 113] No route to host'))
Unter der IP ist der Shelly 3EM (RSI ~ -80dB +- 5 dB) ausgefallen

Das Ganze ist mir dann leider erst am Abend aufgefallen.
Nun lief der Inverter in der Zeit einfach mit dem letzten Limit weiter. Glücklicherweise bei ca. 300Watt.

Nun mehrere Anliegen/Fragen:
Warum setzt das Script hier einfach aus,
wenn z.B. der Inverter nicht erreicht wird, läuft die Schleife einfach weiter bis wieder ok -->  
2023-06-12 21:29:34,194 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (HM-1500) not reachable
.
.
.
2023-06-12 21:33:06,354 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (HM-1500) not reachable
2023-06-12 21:33:14,514 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (HM-1500) not reachable
2023-06-12 21:33:20,719 INFO     OpenDTU: Inverter "HM-1500" reachable: True

Könnte man hier einen Automatismus einbauen, das der Service neu gestartet wird? Oder man benachrichtigt wird um ggf. selber zu Restarten?

Wäre es möglich einen Leistungslimit für diesen Fall zu haben, dass man das Limit dann sozusagen erst mal auf die Grundlast einstellt?

MfG 

Sascha

reserve85, 2023-06-13:
Hi Sascha,

zu Punkt 1: Powermeter nicht erreichbar
Der Code läuft weiter und versucht immer wieder den Zähler zu erreichen - wenn er wieder erreichbar ist wird weitergeregelt.
Im Fehlerfall könnte ich das theoretisch erweitern, dass bei Batteriebetrieb ein "Not-Aus" an den Inverter gesendet wird. Ohne Batteriebetrieb würde ich das dann sogar auf 100% setzen?

Zu Punkt 2: Inverter nicht erreichbar
Hier ist deine DTU nicht erreichbar gewesen. Daher die Exception. Somit kann hier auch kein Limit eingestellt werden oder der Inverter ausgeschaltet werden. Ein Neustart des Scripts ändert hier leider auch nichts. Benachrichtigen hab ich so erstmal nicht vorgesehen, ggf. könnte in Zukunft über eine E-Mail Unterstützung o.Ä. nachgedacht werden.

Es ist immer schwierig die Fehlerbehandlung für jeden Anwendungsfall optimal zu gestalten.

reserve85, 2023-06-13:
> Punkt 1 " Der Code läuft weiter..." Verstehe ich nicht, da bei mir der letzte Eintrag im log-fiel um 12:36 war; dann kommt nichts mehr bis mir der Fehler aufgefallen ist, und ich Raspberry --> sudo reboot; neu gestartet habe. D.h. doch, das Script hat angehalten?

ggf. war auch dein Raspi ohne LAN Verbindung? Der Reboot hat es dann gefixt. Wenn ich das Log anschaue fällt mir auf, dass du ziemlich oft Verbindungsprobleme hast. 
Ich habe nochmal in den Code reingeschaut, bei `Exception at GetHoymilesAvailable, Inverter 0 (HM-1500) not reachable` ist die DTU nicht erreichbar. Wenn die DTU erreichbar ist, dann lautet die Meldung `OpenDTU: Inverter "HM-1500" reachable: False`.
Die Fehlermeldung ändere ich demnächst noch damit es klarer wird.

Spannungswert = inf ist noch ein Problem von mir, da kam von der OpenDTU wahrscheinlich kein valider Wert zurück. Das fixe ich noch.

Dein Loop Intervall ist viel zu schnell. Mit Ahoy schaffe ich einen stabilen Betrieb bei 10s. Erhöhe hier mal auf min. 10-15s. Sonst hat die DTU/Inverter keine Chance die Werte zu aktualisieren weil die DTU mit Limitänderungen restlos ausgelastet ist. Wahrscheinlich ist das auch das Problem mit Inverter reachable = false und den fehlerhaften Batteriespannungen.

SMGatAUXde, 2023-06-13:
Hallo,

vielen Dank für die Tips. Ich habe nun schon mal die Einstellung mit LOOP_INTERVAL_IN_SECONDS = 11 getestet.

OpenDTU: Inverter "HM-1500" reachable: False 

ist auf jeden Fall seltener und auch nicht mehr so oft direkt hintereinander.

Werde nun die Einstellung der DTU auf Niedrig testen.

MfG Sascha

delacor, 2023-06-13:
Welche Hardware für die DTU benutzt du? Eine ESP8266 ist zu schwach für so schnelle Abfragen. Du könntest desweiteren noch gucken mit welcher Frequenz dein ESP läuft. Die kann man auf 240MHZ hochdrehen.

SMGatAUXde, 2023-06-13:
Folgende HW:

Espressif ESP32 WLAN Dev Kit Board Development Bluetooth Wifi v1 WROOM32 NodeMCU &
NRF24L01+ PLUS - PA LNA SMA Antenne Long Range Funk Modul Raspberry Arduino

ESP32 läuft auf 240 MHz.

Bei meiner gestrigen Problemlösungssuche bin ich auf der OpenDTU - Seite noch auf die Empfehlung gestoßen folgende Antenne einzusetzen --> 

EBYTE - NRF24L01+ PLUS - PA LNA SMA Antenne Long Range Funk Modul Raspberry Arduino.

Evtl. ist die ja noch besser und zuverlässiger.

Eine Überlegung wäre auch ein LAN - Erweiterung für die ESP32 Platine zu ergänzen die OpenDTU & den Raspy kabelgebunden zu betreiben. WR alternativlos Funk. Powermeter könnte ggf. noch ausgetauscht werden.
Wobei bei mir ein LAN Verbindung zum Sicherungskasten zum Router leider sehr aufwändig wäre.

reserve85, 2023-06-13:
Evtl löst ein billiger Devolo die Verbindungsprobleme, Strom hast du ja überall. WLAN ist leider für 24h Betrieb oftmals unzuverlässig wenn die Verbindung nicht optimal ist. Der Raspi gehört am besten auch direkt ins LAN (hast du am Router keinen Port mehr frei? Switch?).

reserve85, 2023-06-13:
Ich glaube das läuft überall wo du ein Python Script laufen lassen kannst. Viel passiert in dem code ja nicht.

SMGatAUXde, 2024-01-25:
Hallo reserve85,

Punkt ist eigentlich geschlossen, nun hatte ich den oben beschriebenen Fehler:

**2024-01-25 15:35:42,889 ERROR    Exception at GetHoymilesActualPower
2024-01-25 15:35:42,889 ERROR    HTTPConnectionPool(host='192.168.178.25', port=80): Max retries exceeded with url: /rpc/Switch.GetStatus?id=0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xf6830d10>: Failed to establish a new connection: [Errno 113] No route to host'))
2024-01-25 15:35:52,944 INFO     OpenDTU: Inverter "HM-1500" reachable: True_****

erneut, nun aber leider bei voller Leistung von 1500 Watt.

Wird nun bei einem aktuellen Stand deiner Software dieser Fehler abgefangen und die Leistung auf z.B. den reduzierten Spannungswert gesetzt?

MfG 
SMGatAuxDe

reserve85, 2024-01-25:
nein, aber nach dem Timeout-Fehler (der kann durchaus mal vorkommen) läuft die Regelung doch wieder weiter?

SMGatAUXde, 2024-01-26:
Nein, im Prinzip ist die WLAN-Verbindung von meinem Shelly1PM, welcher als SHELLY_IP_INTERMEDIATE = 192.168.178.25 fungiert unterbrochen. Die Regelung liest zwar weiter die Leistung am 3EM und bekommt Werte Auszug (log file habe ich angehängt ab Zeile 40612 tritt der Fehler auf):

2024-01-25 15:36:07,497 ERROR    Exception at GetHoymilesActualPower
2024-01-25 15:36:07,498 ERROR    HTTPConnectionPool(host='192.168.178.25', port=80): Max retries exceeded with url: /rpc/Switch.GetStatus?id=0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xf6830f30>: Failed to establish a new connection: [Errno 113] No route to host'))
2024-01-25 15:36:17,657 INFO     OpenDTU: Inverter "HM-1500" reachable: True
2024-01-25 15:36:17,697 INFO     Lowest panel voltage: 45.20000076 Volt
2024-01-25 15:36:17,752 INFO     OpenDTU: Inverter "HM-1500" temperature: 59.1 degC
2024-01-25 15:36:17,804 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:18,885 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:20,115 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:21,345 INFO     powermeter Shelly 3EM: -1517  Watt
2024-01-25 15:36:22,397 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:23,450 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:24,514 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:25,563 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:26,611 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:27,750 INFO     powermeter Shelly 3EM: -1517  Watt
2024-01-25 15:36:31,817 ERROR    Exception at GetHoymilesActualPower
2024-01-25 15:36:31,818 ERROR    HTTPConnectionPool(host='192.168.178.25', port=80): Max retries exceeded with url: /rpc/Switch.GetStatus?id=0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xf6801e10>: Failed to establish a new connection: [Errno 113] No route to host'))
2024-01-25 15:36:41,929 INFO     OpenDTU: Inverter "HM-1500" reachable: True
2024-01-25 15:36:41,975 INFO     Lowest panel voltage: 45.20000076 Volt
2024-01-25 15:36:42,031 INFO     OpenDTU: Inverter "HM-1500" temperature: 59.2 degC
2024-01-25 15:36:42,234 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:43,463 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:44,794 INFO     powermeter Shelly 3EM: -1517  Watt
2024-01-25 15:36:45,845 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:46,895 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:47,946 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:48,996 INFO     powermeter Shelly 3EM: -1516  Watt
2024-01-25 15:36:50,044 INFO     powermeter Shelly 3EM: -1515  Watt
2024-01-25 15:36:51,088 INFO     powermeter Shelly 3EM: -1515  Watt
2024-01-25 15:36:52,135 INFO     powermeter Shelly 3EM: -1515  Watt
2024-01-25 15:36:56,201 ERROR    Exception at GetHoymilesActualPower
2024-01-25 15:36:56,202 ERROR    HTTPConnectionPool(host='192.168.178.25', port=80): Max retries exceeded with url: /rpc/Switch.GetStatus?id=0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xf6801a30>: Failed to establish a new connection: [Errno 113] No route to host'))
2024-01-25 15:37:06,309 INFO     OpenDTU: Inverter "HM-1500" reachable: True
2024-01-25 15:37:06,377 INFO     Lowest panel voltage: 45.20000076 Volt
2024-01-25 15:37:06,437 INFO     OpenDTU: Inverter "HM-1500" temperature: 59.3 degC
2024-01-25 15:37:06,491 INFO     powermeter Shelly 3EM: -1515  Watt
2024-01-25 15:37:07,542 INFO     powermeter Shelly 3EM: -1515  Watt
2024-01-25 15:37:08,593 INFO     powermeter Shelly 3EM: -1515  Watt
2024-01-25 15:37:09,769 INFO     powermeter Shelly 3EM: -1529  Watt
2024-01-25 15:37:10,819 INFO     powermeter Shelly 3EM: -1530  Watt
2024-01-25 15:37:11,872 INFO     powermeter Shelly 3EM: -1529  Watt
2024-01-25 15:37:12,923 INFO     powermeter Shelly 3EM: -1530  Watt
2024-01-25 15:37:13,969 INFO     powermeter Shelly 3EM: -1529  Watt
2024-01-25 15:37:15,016 INFO     powermeter Shelly 3EM: -1529  Watt
2024-01-25 15:37:16,060 INFO     powermeter Shelly 3EM: -1529  Watt
2024-01-25 15:37:20,137 ERROR    Exception at GetHoymilesActualPower
2024-01-25 15:37:20,138 ERROR    HTTPConnectionPool(host='192.168.178.25', port=80): Max retries exceeded with url: /rpc/Switch.GetStatus?id=0 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xf6801370>: Failed to establish a new connection: [Errno 113] No route to host'))
2024-01-25 15:37:30,269 INFO     OpenDTU: Inverter "HM-1500" reachable: True
2024-01-25 15:37:30,320 INFO     Lowest panel voltage: 45.20000076 Volt
2024-01-25 15:37:30,363 INFO     OpenDTU: Inverter "HM-1500" temperature: 59.4 degC
[log_20240125.txt](https://github.com/reserve85/HoymilesZeroExport/files/14061850/log_20240125.txt)

Er liest ja ~ -1500 Watt aus aber passt die Leistung nicht mehr an. Die Exception wird immer wiederholt und es erfolgt kein Aufruf mehr von : def SetLimitOpenDTU(pInverterId, pLimit):

Aktuell habe ich den SHELLY_IP_INTERMEDIATE auf false gesetzt.

Idealerweise müsste das Script bei anhaltenden Verbindungsproblemen zu den Zählern die Leistung zumindest auf den HOY_BATTERY_REDUCE_WATT oder auf einen bestimmbaren Wert z.B. Grundlast... setzten.

Vlt. wäre es die möglich so was zu implementieren. Werde aber auch mal mein WLAN  verbessern.

Gruß

reserve85, 2024-01-26:
Ok, verstehe. Ich habe jetzt folgendes implementiert:

- Wenn der Intermediate Meter nicht ausgelesen werden kann, dann wird versucht die DTU abzufragen.
- Wenn die DTU Abfrage auch nicht funktioniert, dann wird jeder Inverter auf "HOY_MIN_WATT_IN_PERCENT" reduziert.

- Wenn der normale Zähler nicht ausgelesen werden kann, dann wird jeder Inverter auf "HOY_MIN_WATT_IN_PERCENT" reduziert.

Es wäre super wenn du das bei dir mal laufen lassen könntest.
ggf. kannst du mal einen Verbindungsausfall zu deinen Zählern simulieren.
https://github.com/reserve85/HoymilesZeroExport/tree/reduce_limit_on_error

SMGatAUXde, 2024-02-02:
Sry, schaffe es jetzt erst. 

Wenn ich die aktuelle Version via... https://github.com/reserve85/HoymilesZeroExport/tree/reduce_limit_on_error installiere bekomme ich die Version 1.68 und nicht 1.70. 
Habe versucht die beiden files [HoymilesZeroExport.py](https://github.com/reserve85/HoymilesZeroExport/blob/reduce_limit_on_error/HoymilesZeroExport.py) und [HoymilesZeroExport_Config.ini](https://github.com/reserve85/HoymilesZeroExport/blob/reduce_limit_on_error/HoymilesZeroExport_Config.ini) manuell nach der Installation zu überschreiben mit dem Stand 1.70. 
Script startet dann aber nicht. 

Wie installiere ich denn Version 1.70?
[image]

SMGatAUXde, 2024-02-02:
Folgende Fehlermeldung bekomme ich im log sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f:

- Logs begin at Fri 2024-02-02 10:47:08 CET. --
Feb 02 10:47:20 piFlexCellEdgeDevice systemd[1]: Started HoymilesZeroExport Service.
Feb 02 10:47:22 piFlexCellEdgeDevice python3[737]: Traceback (most recent call last):
Feb 02 10:47:22 piFlexCellEdgeDevice python3[737]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Feb 02 10:47:22 piFlexCellEdgeDevice python3[737]:     from packaging import version
Feb 02 10:47:22 piFlexCellEdgeDevice python3[737]: ModuleNotFoundError: No module named 'packaging'
Feb 02 10:47:22 piFlexCellEdgeDevice systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Feb 02 10:47:22 piFlexCellEdgeDevice systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Feb 02 10:47:22 piFlexCellEdgeDevice systemd[1]: HoymilesZeroExport.service: Service RestartSec=100ms expired, scheduling restart.
Feb 02 10:47:22 piFlexCellEdgeDevice systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 1.
Feb 02 10:47:22 piFlexCellEdgeDevice systemd[1]: Stopped HoymilesZeroExport Service.
Feb 02 10:47:22 piFlexCellEdgeDevice systemd[1]: Started HoymilesZeroExport Service.

reserve85, 2024-02-02:
`pip3 install packaging`
ausführen oder nach anleitung installieren

SMGatAUXde, 2024-02-02:
packeging installiert und python auf 3.9 
Fehler bleibt.
bei dem pachakging wir auf python 3.7 verwiesen.

pi@piFlexCellEdgeDevice:~ $ pip3 install packaging
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: packaging in ./.local/lib/python3.7/site-packages (23.2)

 Default version sollte aber 3.9 sein...

pi@piFlexCellEdgeDevice:~ $ python --version
Python 3.9.15

mach ich hier noch was falsch?

reserve85, 2024-02-02:
probiere mal manuell mit der spezifischen version zu starten:
`python3.9 HoymilesZeroExport.py`

ich vermute dass er das mit version 3.7 öffnen will... warum auch immer?

SMGatAUXde, 2024-02-02:
Bin nun etwas verwirrt, 
das Script scheint zu laufen, da im Shell von Thonny das log file abläuft, aber im Terminal kommt auf: sudo systemctl status  HoymilesZeroExport.service
failed...
[image]

SMGatAUXde, 2024-02-02:
Kommt folgendes

pi@piFlexCellEdgeDevice:~/HoymilesZeroExport $ python3.9 HoymilesZeroExport.py
Traceback (most recent call last):
  File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 20, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'

SMGatAUXde, 2024-02-02:
installieren von Requests... geht leider nicht

pi@piFlexCellEdgeDevice:~/HoymilesZeroExport $ pip install requests
bash: /usr/bin/pip: /usr/bin/python: Defekter Interpreter: Datei oder Verzeichnis nicht gefunden

reserve85, 2024-02-02:
probiere nochmal das:

`python3.9 -m pip install requests`

SMGatAUXde, 2024-02-02:
hab das gleich  noch mit packaging gemacht.

python3.9 -m pip install packaging

nun läuft es mit dem manuellen start python3.9 HoymilesZeroExport.py
aber nicht im autostart nach dem reboot
also wenn ich nun python3 --version anzeige, sagt er wieder 3.7.3 
eine änderung via
pi@piFlexCellEdgeDevice:~ $ sudo rm /usr/bin/python
pi@piFlexCellEdgeDevice:~ $ sudo ln -s /usr/bin/python3.9 /usr/bin/python

führt seltsamerweise wieder zu 3.7.3
pi@piFlexCellEdgeDevice:~ $ python3 --version
Python 3.7.3

SMGatAUXde, 2024-02-02:
Um zum eigentlichen Problem zurückzukommen... 

Version wird ja V1.69 genannt das ist dann wohl die aktuellste im Changelog, auch wenn hier oben 1.70 steht

In meinem log file bekomme ich gleich am Anfang den Fehler:

2024-02-02 12:35:06,233 ERROR    Exception at GetHoymilesActualPower
2024-02-02 12:35:06,234 ERROR    HTTPConnectionPool(host='192.168.172.25', port=80): Max retries exceeded with url: /status (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xf682c598>, 'Connection to 192.168.172.25 timed out. (connect timeout=10)'))
2024-02-02 12:35:16,243 INFO     ---Start Zero Export---

der intermediate ShellyPM ist die IP ...25

laut meiner Frizbox ist er eigentlich erreichbar. 
aber das Script läuft zumindest weiter und dieses exception kommt nur einmal am anfang 

müsste in nun mal ganz offline machen und schauen ob das auch keine Exception bzw. ein Stoppen des Scripts auslöst
...

reserve85, 2024-02-02:
1.69 ist nicht richtig du brauchst die "Dev" Version hier: https://github.com/reserve85/HoymilesZeroExport/blob/reduce_limit_on_error/HoymilesZeroExport.py

einfach die *.py ersetzen

lullatscho, 2024-02-18:
> hab das gleich noch mit packaging gemacht.
>
> python3.9 -m pip install packaging
>
> nun läuft es mit dem manuellen start python3.9 HoymilesZeroExport.py aber nicht im autostart nach dem reboot also wenn [...]

Kann das sein, dass du den Link falsch setzt? Die Python Version sollte glaub eher hier liegen: /usr/local/bin/python3.9?

reserve85, 2024-02-21:
kann das zu? Im Script ist der "Not-Aus" implementiert und ich konnte es auch ein mal beobachten.

closed 2024-03-01 by reserve85