reserve85/HoymilesZeroExport #75: Script bleibt stehen

reserve85, 2023-06-14:
So, ich denke ich habe das derzeit beste für mich eingestellt.
Ich habe 3 WR: 400, 800 und 1500er. 
Meine Konfig habe ich folgendermaßen angepasst:
[COMMON]
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = false

[CONTROL]
POWERMETER_TARGET_POINT = -50
POWERMETER_TOLERANCE = 25
POWERMETER_MAX_POINT = 0

Damit speise ich zwischen 400 und 700 Wh pro Tag ein. Ich beziehe (wenn nicht gerade der Herd läuft oder alles gleichzeitig) tagsüber ca. 1 kWh aus dem Netz (wenn ich nicht da bin weniger). Gutes Wetter mal vorausgesetzt.

Beispiel gestern (Klima lief ab Mittag):
[image]

Ich rechne damit, dass im August mein Speicher kommt. Dann wird es nochmal spannend.

Aber: seit Montag steigt das Script bis zu zweimal am Tag aus, bleibt einfach stehen. zu unregelmäßigen Zeiten morgens und abends. Ich muss dann neustarten. Hat das von euch auch schon jemand gehabt, bzw. jemand eine Idee warum das passiert? Ich habe an meiner Konfig nichts geändert und der Rest (Influx, ioroker, Grafana etc.) läuft ja stabil weiter. Wenn ich das nicht mitbekomme, dann speise ich natürlich deutlich mehr ein...

_Originally posted by @keinprofi in https://github.com/reserve85/HoymilesZeroExport/issues/28#issuecomment-1590630555_

keinprofi, 2023-06-14:
V 1.41 - Was habe ich übersehen?
[image]

keinprofi, 2023-06-14:
ist doch da...
[image]

reserve85, 2023-06-14:
Schicke mir mal deine config oder hänge sie hier an, ich schaue dann nach.

keinprofi, 2023-06-14:
Bitte schön...
[HoymilesZeroExport_Config.txt](https://github.com/reserve85/HoymilesZeroExport/files/11749707/HoymilesZeroExport_Config.txt)

reserve85, 2023-06-14:
Bitte schön... :-)
[HoymilesZeroExport_Config.txt](https://github.com/reserve85/HoymilesZeroExport/files/11749876/HoymilesZeroExport_Config.txt)

keinprofi, 2023-06-14:
Danke!

Was wars?

reserve85, 2023-06-14:
Es hat gefehlt :-) glaube du hattest nicht gespeichert...
Mache einfach einen Diff der beiden Dateien mit [WinMerge](https://winmerge.org/), damit kannst du die Unterschiede recht schnell sehen. Das nutze ich immer.

keinprofi, 2023-06-14:
in WinMerge zeigt er mir das auch an. Aber öffne mal die Textdatei die ich Dir geschickt habe (https://github.com/reserve85/HoymilesZeroExport/issues/75#issuecomment-1591830915) im Editor. Da ist der Eintrag drin!
[image]

Und sag jetzt nicht bei Dir wäre das nicht so, sonst muss ich Medis einwerfen....

reserve85, 2023-06-14:
Nimm schon mal die medis, ist nicht drin

keinprofi, 2023-06-14:
Bin ich jeck? Sieh selbst...

https://1drv.ms/v/s!AhDwS_KPMrnmhIwCbsuRQkca8Pa12A?e=QAsXdr

reserve85, 2023-06-14:
Jaaaa, das ist der 3EM, da fehlt aber der EM ohne 3

keinprofi, 2023-06-14:
ok, ich geh medis nehmen! 

Nochmals danke!

keinprofi, 2023-06-15:
Moin,

die gute Nachricht: Es gab heute nicht den Ausfall der letzten Tage. Ich denke dass das Problem mit Deinem Timeout behoben ist.

reserve85, 2023-06-15:
Klingt erstmal gut, aber schaue dir mal die Logs an wie oft der Timeout auftritt. Das sollte eigentlich nie passieren, ansonsten ist wahrscheinlich dein Netzwerk instabil.

keinprofi, 2023-06-15:
Sollte...
Das Netz ist grundsätzlich stabil. Der Tasmota ist mit -65 dBm im WLAN gut aufgestellt. Dennoch verliert er die Verbindung - kein WLAN. Aber das erst seit einigen Tagen - wie bereits erwähnt.
Habe Hichi mal angeschrieben... und gerade die Antwort bekommen...

Es gibt da scheinbar bekannte Verbindungsprobleme die man mit einem "reset 3" fixen kann. Ich teste das mal. Sollte das nicht klappen denke ich über einen Sensor nach, den ich per Kabel an den Pi anschließe. Aber soweit ist es nicht. Schließlich lief der Sensor mit WLAN ein gutes Jahr einwandfrei.

keinprofi, 2023-06-15:
Das Script ist bisher nicht mehr stehen geblieben. Ich denke mal das dieses Problem erledigt ist.

Mein Problem mit dem Tasmota Sensor habe ich nicht abschließend beheben können. Deswegen habe ich jetzt den TTL Sensor bestellt, den ich direkt an den Raspi anschließen werde.
Da ich im Portainer keinen Port 8087 angelegt habe muss ich den vom "alten Tasmota" nehmen. Muss ich den noch irgendwo anpassen außer im Bereich [IOBROKER] in der ini? Nicht das ich da was übersehen habe.

reserve85, 2023-06-16:
Moin, da bin ich überfragt - bei mir läuft der Hichi schon seit Ewigkeiten ohne Problem mit Tasmota 12.3.1.4
Aber super, dass das Problem an sich wohl behoben ist. Ich werde es dann demnächst nach MAIN mergen.

keinprofi, 2023-06-16:
Lief bei mir jetzt auch über ein Jahr lang. Aber Verbindungsausfälle von 20 Minuten oder mehr über den Tag verteilt machen jede Steuerung witzlos.
12.3.1.4? Okay, ich habe noch die 9.5.0 drauf. Irgendwie hatte ich in Erinnerung das kein Update gemacht werden darf und da es funktionierte hab ich nie eins gemacht. Dann kann ich ja mal ein Update wagen...
Danke für den Tipp.

reserve85, 2023-06-16:
ja darfst kein update machen, nur mit dem Flasher.
siehe https://ottelo.jimdofree.com/stromz%C3%A4hler-auslesen-tasmota/ - da gibt es vorgefertigte images

keinprofi, 2023-06-16:
MIt dem Flasher? Demnach läuft das per OTA:
[image]

reserve85, 2023-06-16:
du brauchst so ein set: 
https://www.ebay.de/itm/144492092596?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=tg9hdLAtSmS&sssrc=2047675&ssuid=EfnMFbq4SbG&widget_ver=artemis&media=COPY
dann kannst du das vorkompilierte image downloaden und draufflashen (ich nutze dazu Tasmotizer).
https://bit.ly/3pZkE95

Ist alles in ein paar Minuten erledigt und du hast einen Ersatzcontroller für ein paar Euro. Kannst du dann einfach im Gehäuse austauschen.

keinprofi, 2023-06-16:
Hat OTA funktioniert:
[image]

reserve85, 2023-06-16:
Top, evtl bringt es ja schon eine Verbesserung

keinprofi, 2023-06-16:
Die 1M Version per Firmware Upgrade einspielen (ich habe sicherheitshalber zuvor die Minimalversion eingespielt, obwohl das nicht mehr nötig sein soll). Danach Restore Configuration und fertig.

Ich habe jetzt ein Dauerping auf den Sensor und meine Überwachung lauft ja auch noch. Wäre schön wenn es funktioniert. Dann ahbe ich den TTL Sensor als Reserve.

reserve85, 2023-06-16:
perfekt, funktioniert

keinprofi, 2023-06-16:
Ich würde sagen mit der neuen Firmware ist auch das Problem erledigt:
[image]

reserve85, 2023-06-16:
Ja perfekt, dann mach ich hier zu.

closed 2023-06-16 by reserve85