reserve85/HoymilesZeroExport #76: Inverter is not found and stays on "yet unknown" but is working

fummmmel, 2023-06-15:
Hi,
first of all let me say thank you for the great script. 

It is working quite well so far. Although I was not able to setup the systemd service, but that might be an issue with my RasbPi
I have 4 Hoymiles Inverters on my AHOY-DTU and i'm using the Home Assistant power value. 
My Issue is that the Last of the Inverters is not named properly by the script. The script is able to control it though.
Log:
`2023-06-15 09:52:08 INFO     Ahoy: Inverter "HM-600" reachable: True
2023-06-15 09:52:08 INFO     Ahoy: Inverter "HM-1200" reachable: True
2023-06-15 09:52:09 INFO     Ahoy: Inverter "HM-600-2" reachable: True
2023-06-15 09:52:09 INFO     Ahoy: Inverter "yet unknown" reachable: True
2023-06-15 09:52:09 INFO     powermeter HomeAssistant: 628  Watt
2023-06-15 09:52:09 INFO     Retry Counter exceeded: Inverterlimit already at 3600 Watt`
The Name of the Inverter would be "HM-1200-2". 
Here is my config file:
[conf.txt](https://github.com/reserve85/HoymilesZeroExport/files/11754624/conf.txt)

Any Idea what might be the issue?

Thank you.

reserve85, 2023-06-15:
Hi,

the names are read out of the RestAPI, can you enter this into your browser to check whats the result:
`http://192.168.224.59/api/inverter/list`

fummmmel, 2023-06-15:
i have been checking the API for live data and it seems one inverter is missing although it is showing up in the UI!?
[image]

fummmmel, 2023-06-15:
Okay, it seems that the issue is in the AHOY-DTU and not in the export script

closed 2023-06-15 by fummmmel

reserve85, 2023-06-16:
@fummmmel: i updated the script, see https://github.com/reserve85/HoymilesZeroExport/tree/dev
It would be nice if you can update and see if it´s working for you. I tested it locally and it seems to be running fine.