reserve85/HoymilesZeroExport #78: Fehlermeldung seit Update auf 1.42 OpenDTU: Inverter "yet unknown" seit Update

XpertXXL, 2023-06-17:
Die Steuerung funktioniert nicht mehr :-(
siehe log:

Jun 17 13:24:44 RasPI-Disco python3[885]: 2023-06-17 13:24:44 INFO     powermeter Shelly 3EM: -345  Watt
Jun 17 13:24:45 RasPI-Disco python3[885]: 2023-06-17 13:24:45 INFO     powermeter Shelly 3EM: -348  Watt
Jun 17 13:24:46 RasPI-Disco python3[885]: 2023-06-17 13:24:46 INFO     powermeter Shelly 3EM: -347  Watt
Jun 17 13:24:47 RasPI-Disco python3[885]: 2023-06-17 13:24:47 INFO     powermeter Shelly 3EM: 163  Watt
Jun 17 13:24:47 RasPI-Disco python3[885]: 2023-06-17 13:24:47 INFO     Retry Counter exceeded: Inverterlimit already at 2000 Watt
Jun 17 13:24:52 RasPI-Disco python3[885]: 2023-06-17 13:24:52 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Jun 17 13:24:52 RasPI-Disco python3[885]: 2023-06-17 13:24:52 INFO     powermeter Shelly 3EM: 150  Watt
Jun 17 13:24:52 RasPI-Disco python3[885]: 2023-06-17 13:24:52 INFO     Retry Counter exceeded: Inverterlimit already at 2000 Watt
Jun 17 13:25:12 RasPI-Disco python3[885]: 2023-06-17 13:25:12 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Jun 17 13:25:13 RasPI-Disco python3[885]: 2023-06-17 13:25:13 INFO     powermeter Shelly 3EM: -341  Watt
Jun 17 13:25:14 RasPI-Disco python3[885]: 2023-06-17 13:25:14 INFO     powermeter Shelly 3EM: -344  Watt
Jun 17 13:25:15 RasPI-Disco python3[885]: 2023-06-17 13:25:15 INFO     powermeter Shelly 3EM: 767  Watt
Jun 17 13:25:15 RasPI-Disco python3[885]: 2023-06-17 13:25:15 INFO     Retry Counter exceeded: Inverterlimit already at 2000 Watt
Jun 17 13:25:33 RasPI-Disco python3[885]: 2023-06-17 13:25:33 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Jun 17 13:25:33 RasPI-Disco python3[885]: 2023-06-17 13:25:33 INFO     powermeter Shelly 3EM: 1395  Watt
Jun 17 13:25:33 RasPI-Disco python3[885]: 2023-06-17 13:25:33 INFO     Retry Counter exceeded: Inverterlimit already at 2000 Watt
Jun 17 13:25:53 RasPI-Disco python3[885]: 2023-06-17 13:25:53 INFO     OpenDTU: Inverter "yet unknown" reachable: True
Jun 17 13:25:53 RasPI-Disco python3[885]: 2023-06-17 13:25:53 INFO     powermeter Shelly 3EM: 153  Watt
Jun 17 13:25:53 RasPI-Disco python3[885]: 2023-06-17 13:25:53 INFO     Retry Counter exceeded: Inverterlimit already at 2000 Watt

könntet ihr die älteren Versionen des Scripts und der config auch noch bereitstellen, habe mir die funktionierende Version leider nicht gesichert
vielen Dank für euere Arbeit

XpertXXL

reserve85, 2023-06-17:
Du findest jede Version in der history zum Download. https://github.com/reserve85/HoymilesZeroExport/commits/main

Was genau funktioniert nicht? Ich sehe nur unknown inverter (hat sich hier auch an der openDTU API was geändert?) - ansonsten sieht es doch gut aus?

XpertXXL, 2023-06-17:
Hallo reserve85,
danke für deine Bemühungen :-)
Ich verwende aktuell folgenden OpenDTU:

Hostname	OpenDTU-277BB8
SDK-Version	v4.4.4
Konfigurationsversion	0.1.25
Firmwareversion / git Hash	v23.6.14
PIO Umgebung	generic
Firmware-Aktualisierung	Aktuell!

Die DTU regelt den WR nicht und der steht immer auf 2000 W, da ändert sich nichts. Wir verbrauchen ca. 180W, der Rest von ca. 450 W wandert ins Netz.

XpertXXL

Benno10, 2023-06-17:
Hallo reserve85,
ich habe das gleiche Problem.

Hostname OpenDTU-08EE90
SDK-Version v4.4.4
Konfigurationsversion 0.1.25
Firmwareversion / git Hash v23.6.14
PIO Umgebung generic
Firmware-Aktualisierung Aktuell!
Cip-Modul  ESP32-D0WD-V3

ini VERSION = 1.40
Py version__ = 1.42
Heute so als Zip heruntergeladen

Die DTU regelt den WR 1200W nicht mehr und der steht immer auf gleichem Wert.

reserve85, 2023-06-17:
Ok schau ich nachher mal aber in dem log oben passt es doch? Du beziehst Energie und daher wird nix gedrosselt.

reserve85, 2023-06-17:
Ahso ja geht nicht weil inverter unbekannt. Mein Denkfehler! Ich schau es demnächst an.

reserve85, 2023-06-17:
@XpertXXL @Benno10 Hupsi, da hat sich ein Copy-Paste Fehler eingeschlichen. Sollte jetzt laufen. Sorry dafür!

XpertXXL, 2023-06-18:
läuft wieder :-), danke für den super Support!

XpertXXL

Benno10, 2023-06-18:
Vielen Dank auch von mir für die schnelle Hilfe.
Läuft jetzt wieder.

Benno10

closed 2023-06-18 by reserve85