reserve85/HoymilesZeroExport #79: Tasmota Konfiguration

Esojma-Silverbullet, 2023-06-18:
Hi zusammen,

ich versuche mich auch gerade ein wenig an dem Skript.
Es läuft und ich kann sowohl die openDTU als auch meinen Tasmota finden.
Als völliger noob scheitere ich aber schon an der config zum Tasmota.
Im log liefert mein Tasmota keine Werte.
Könnte mir einer kurz und knackig zeigen, was genau ich dort befüllen muss.
Verstehe das Beispiel einfach nicht wie es richtig sein soll.

Das bekomme ich geliefert mit der Abfrage aus der config.

StatusSNS
        Time "2023-06-18T12:44:26"
MT691
       Total_in XXX
       Power_cur XXX
       Meter_id "XXX"

Das verstehe ich nicht wirklich :(
TASMOTA_JSON_STATUS = StatusSNS
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
TASMOTA_JSON_POWER_MQTT_LABEL = Power_cur (So heist mein output, reicht das dort?)
TASMOTA_JSON_POWER_CALCULATE = FALSE
TASMOTA_JSON_POWER_INPUT_MQTT_LABEL =
TASMOTA_JSON_POWER_OUTPUT_MQTT_LABEL =

Bin für jeden Tip/Hilfe dankbar

eso

reserve85, 2023-06-18:
Ersetze SML durch MT691 dann sollte es gehen.

Esojma-Silverbullet, 2023-06-18:
Treffer 👍 
Danke @reserve85 genau das war es, jetzt bekomme ich keinen Error mehr im log.
Dann mache ich mich jetzt mal ans optimieren.
Klasse Skript und vielen Dank für deine Arbeit.

eso

closed 2023-06-18 by reserve85