reserve85/HoymilesZeroExport #80: Config

feeti1, 2023-06-20:
[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = 0
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 10
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = true, then the limit will jump to 100% by reaching this point.
POWERMETER_MAX_POINT = 1500

# List of INVERTERS, based on COMMON/COUNT
[INVERTER_1]
# power rating of your inverter
HOY_MAX_WATT = 1500
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = true
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47.5
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48.4
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 49.3
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 850
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 400
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 49.5

; [INVERTER_2]
; # power rating of your inverter
; HOY_MAX_WATT = 1500
; # minimum limit in percent, e.g. 5%
; HOY_MIN_WATT_IN_PERCENT = 5
; # factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
; HOY_COMPENSATE_WATT_FACTOR = 1
; # battery powered?
; HOY_BATTERY_MODE = false
; # voltage to turn off the inverter
; HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
; # voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
; HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
; # voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
; HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
; # maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
; HOY_BATTERY_NORMAL_WATT = 1500
; # maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
; HOY_BATTERY_REDUCE_WATT = 300
; # reenable inverter at this voltage
; HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51

Sorry, war etwas eingebunden.

reserve85, 2023-06-21:
Kannst du auch deine Einstellungen unter [COMMON] noch teilen?
Und mir wäre ein Log sehr hilfreich wenn du POWERMETER_MAX_POINT reduzierst auf z.B. 100.

Gruß

feeti1, 2023-06-24:
[image]

Hallo,

Ich habe alles nochmal auf einem anderen NUC neu aufgesetzt. Das Fehlverhalten konnte ich jetzt nicht mehr feststellen.

mfg Dirk

feeti1, 2023-06-26:
Hallo,

mit der aktuellen 1.40 habe ich jetzt ein neues Problem.
Es ist jetzt schon 2x passiert das der Inverter ziemlich genau um 0:00 abgeschaltet wird.
Das geschieht obwohl die Batterie noch mehr als 48V hat.
Wenn ich dann den Inverter über ahoydtu neu starte, läuft er die Nacht durch.
Natürlich habe Ich dann nicht immer Zugriff per SSH auf das Log.
Mit dem Script 1.33 passiert das nicht. 
Damit gab es aber das Problem das es manchmal komplett hängte.
Gibt es die Möglichkeit nochmal das Script und die py mit der 1.30 zu bekommen?
Das lief meiner Meinung nach absolut stable.

lg

reserve85, 2023-06-26:
Welche Version vom Script? Nicht die Version der ini. Für mich interessant ist immer die Version des scripts und die siehst du beim Startup oder in der *.py.

Aktiviere doch bitte mal die Speicherung der Logs, dann kann man das sehen. Irgend eine Aktion um 0:00 ist nicht programmiert. Vermutlich gibt Ahoy 0V Panelspannung zurück und dann wird ausgeschaltet. Eingeschaltet dann erst wieder bei 49,5V. Schaue ich mir mal an, ich denke so könnte das passieren. Ggf. auch wenn wlan weg ist…

Mit 
`sudo journalctl -u HoymilesZeroExport.service -n 20000 -e`
müsstest du auch zu der Uhrzeit scrollen können, das gibt dir die letzten 20000 log Zeilen aus.

Versionen sind immer historisiert, https://github.com/reserve85/HoymilesZeroExport/commits/main
Da gibts alle.

feeti1, 2023-06-26:
Also, das Script das heute ausgestiegen ist,  habe ich gestern um 17:00 installiert.
Kann also nur die aktuellste Version sein.

reserve85, 2023-06-26:
Ok habe da eine Vermutung

reserve85, 2023-06-26:
Hast du in ahoy reset values at Midnight aktiviert?

feeti1, 2023-06-26:
Ja, habe ich.
Habe es allerdings gerade deaktiviert.

reserve85, 2023-06-26:
Ok, habe in dem dev Zweig eingebaut, dass immer die letzten 5 Werte von der DTU betrachtet werden. Also das Maximum der letzten 5 Ablesungen. Damit sollte das Problem auch weg sein. Sollte eine 0V Panel Spannung zurück kommen wird es damit ignoriert (außer es passiert 5 mal in Folge).

closed 2023-07-24 by reserve85