reserve85/HoymilesZeroExport #87: Ist HoymilesZeroExport auf einem Mikrocontroller lauffähig, z.B ESP32?

OlafWalther1969, 2023-07-13:
Hallo,
einen lieben Gruß in die Runde.

Kann das HoymilesZeroExport Projekt auf einem Mikrocontroller ausgeführt werden.
Ich hatte per Zufall ein Produkt aus Östereich entdeckt. Das nennt sich Smart-Mikro Solar.
Dort scheint die ZeroExport Software auf einem Mikrocontroller zu laufen.
In der Beschreibung sind genau die Funktionen der HoymilesZeroExport beschrieben.

Viele Grüße
Olaf Walther

reserve85, 2023-07-13:
Nein, das läuft so leider nicht. Dafür ist es hier kostenlos :-).

OlafWalther1969, 2023-07-14:
> Nein, das läuft leider nicht auf einem Microcontroller. Dafür ist es hier kostenlos :-). Vielleicht will es ja jemand irgendwann mal portieren, aber das ist schon Aufwand.

Guten Morgen, Ich danke Dir für die Rückmeldung!
Das ist hier ein tolles Projekt und Du machst wirklich eine tolle Arbeit.

Viele Grüße
Olaf

costa2, 2023-07-14:
Hallo.
Ich habe mal testweise Micropython auf einem ESP32 installiert. 
Das Script läuft nicht.
Es scheitert schon daran, dass Micropython urequests anstatt requests nutzt und mit ini dateien nicht zurecht kommt.
Gruß,
Volker

reserve85, 2023-07-14:
Ja, ich hatte da auch schonmal interessehalber geschaut weil die Frage schonmal kam. Ohne GUI oder Display o.Ä. ist das mMn. ziemlich kompliziert.
Vor allem war das Script lediglich für meine privaten Zweck gedacht und ich wollte das von anfang an auf dem RasPi laufen lassen. Daher von mir eine klares nein - werde ich nicht umsetzen.

@OlafWalther1969: besten Dank, ich habe es gesehen. Danke nochmal dafür 🥇

OlafWalther1969, 2023-07-14:
> Daher von mir ein klares nein - werde ich nicht umsetzen.

Hallo, ich hoffe nicht das Du das falsch verstanden hast. Ich habe nicht erwartet das Du etwas umsetzt.
Das war nur meine Unwissenheit. Ich hatte gegoogelt, dass auf einem auf einem Microcontroller Micro Python läuft.

Ich selbt habe zu Hause einen RaspberryPI4 und IoBroker. Da ist das mit Phyton kein Problem.

Hat Jemand das Projekt auf einer günstigeren Plattform, wie dem Raspberry PI?

LG

OlafWalther1969, 2023-07-14:
> Es scheitert schon daran, dass Micropython urequests anstatt requests nutzt und mit ini dateien nicht zurecht kommt.
> Gruß,
> Volker

Danke Volker für Deine Rückmeldung!

reserve85, 2023-07-14:
> Hallo, ich hoffe nicht das Du das falsch verstanden hast. Ich habe nicht erwartet das Du etwas umsetzt. Das war nur meine Unwissenheit. Ich hatte gegoogelt, dass auf einem auf einem Microcontroller Micro Python läuft.

Kein Problem, ich hab das auf niemanden bezogen, wollte nur meine Meinung schreiben :-)

costa2, 2023-07-14:
Auch ein Lob von mir.
Dein Script läuft bei mir auf dem Raspi einwandfrei.
Was das Micropython betrifft, ich werde mich da irgendwann mal näher mit beschäftigen, man müsste das Script umschreiben, damit es ohne ini auskommt, die Daten also direkt im Script eingeben.
Als Vorlage könnte man das Script "hoymiles-tarnkappe" nutzen, das ist für Micropython geschrieben.
Dort ist requests z.B. so definiert: `import urequests as requests`
Das Script selbst muss main.py heißen, damit es beim Start der Hardware automatisch gestartet wird.

Volker

delacor, 2023-07-14:
> Hat Jemand das Projekt auf einer günstigeren Plattform, wie dem Raspberry PI?
> 
> LG
Kennst du schon orangePi oder BananaPi? Klar, kosten jetzt nicht die 5€ wie ein ESP, aber sind wenigstens ein bisschen günstiger.

closed 2023-07-24 by reserve85