reserve85/HoymilesZeroExport #88: HM-1500 über Nacht drosseln

hartloeter, 2023-07-17:
Hallo, wäre es möglich einen HM-1500 für eine Nachteinspeisung ab ca. 20 Uhr bis morgens zu drosseln. Ich will 2Module direkt anschließen und die anderen Eingänge abends mit einer Batterie speisen. Vorab schon einmal Danke für eine Antwort. mfg harti

reserve85, 2023-07-18:
Hi, das wird so nicht unterstützt. Aber ein ähnliches Vorgehen wie hier wäre evtl. denkbar: https://github.com/reserve85/HoymilesZeroExport/issues/28#issuecomment-1562665352

Du hast dann zwei Scripte, eins für tagsüber (Drosselung und Batterie deaktiveren) und eins für nachts (batteriebetrieb). Die Umschaltung zwischen beiden erfolgt dann mit einem CronJob.
Allerdings stelle ich mir das nicht so einfach vor, du musst die Umschaltung mit deinen Relais? ja irgendwie gleichzeitig vornehmen.

Außerdem: ob das mit zwei Panels problemlos regelbar ist kann ich dir auch nicht sagen, den Anwendungsfall hatte ich noch nicht.

hartloeter, 2023-07-18:
Danke für die schnelle Antwort, der hm-1500 läuft auch mit 2 Modulen tagsüber. Abends stoppe ich das script und setze den hm-1500 auf 40% fest für die Nacht. Ist halt umständlich aber ich schaue mir das mal an.

hartloeter, 2023-07-21:
Hi, muss noch einmal stören. Ist es möglich die Zeilen (# voltage to turn off the inverter und # reenable inverter at this voltage) heraus zu nehmen. Dann würde das script laufen. Da die Spannung der Module unter 1V fallen schaltet der Inverter ab. Kleiner 1V kann ich nicht einstellen da eine Fehlermeldung kommt. Gruß harti

reserve85, 2023-07-21:
Moin, 
ja, das ist dann etwas tricky mit einem Inverter.
Kannst du mir sagen welcher Fehler da kommt? Hast du das Log?

Für deine Zwecke sollte das eigentlich so klappen:
```
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = -50
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 50
```

Alternativ könnte ich die config erweitern, damit die Spannungen bestimmter Panels im Batteriebetrieb ignoriert werden.

hartloeter, 2023-07-21:
Danke erstmal probiere ich aus.
hier die Fehlermeldung:   HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V.append(config.getfloat('INVERTER_' + str(i + 1), 'HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V'))
Jul 20 23:36:15 solaranzeige python3[11367]:   File "/usr/lib/python3.9/configparser.py", line 823, in getfloat
Jul 20 23:36:15 solaranzeige python3[11367]:     return self._get_conv(section, option, float, raw=raw, vars=vars,
Jul 20 23:36:15 solaranzeige python3[11367]:   File "/usr/lib/python3.9/configparser.py", line 808, in _get_conv
Jul 20 23:36:15 solaranzeige python3[11367]:     return self._get(section, conv, option, raw=raw, vars=vars,
Jul 20 23:36:15 solaranzeige python3[11367]:   File "/usr/lib/python3.9/configparser.py", line 803, in _get
Jul 20 23:36:15 solaranzeige python3[11367]:     return conv(self.get(section, option, **kwargs))
Jul 20 23:36:15 solaranzeige python3[11367]: ValueError: could not convert string to float: '0,1'
Jul 20 23:36:15 solaranzeige systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Jul 20 23:36:15 solaranzeige systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Jul 20 23:36:15 solaranzeige systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Jul 20 23:36:15 solaranzeige systemd[1]: Stopped HoymilesZeroExport Service.
Jul 20 23:36:15 solaranzeige systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Jul 20 23:36:15 solaranzeige systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Jul 20 23:36:15 solaranzeige systemd[1]: Failed to start HoymilesZeroExport Service.

hartloeter, 2023-07-21:
Danke script läuft melde mich nach dem test heute nacht

reserve85, 2023-07-21:
> Danke erstmal probiere ich aus. hier die Fehlermeldung: HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V.append(config.getfloat('INVERTER_' + str(i + 1), 'HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V')) Jul 20 23:36:15 [...]

musst ein Punkt anstelle Komma schreiben.

hartloeter, 2023-07-23:
Danke für die Unterstützung das script läuft. Ab 21 Uhr speise ich max 300 Watt ein bis der Laderegler abschaltet.

closed 2023-07-23 by hartloeter

reserve85, 2023-07-23:
Wie hast du es jetzt gelöst für Tag und Nacht?

hartloeter, 2023-07-23:
2 Module sind immer am hm-1500 angeschlossen. 2 Module laden über einen Victron 100/20 eine 24v Batterie auf. Der Lastausgang wird ab26,4v aktiv und ab 21 Uhr schaltet ein Relais die Leitung zum Wechselrichter frei. Ich habe im script eingestellt das ab 27v der hm-1500 auf 600w limitiert wird. So speise ich ca. 2200w nachts ein.

hartloeter, 2023-07-23:
Dadurch das am HM-1500 ja immer noch eine Spannung von den Modulen vorhanden ist fließt kein großer Strom da die Elkos geladen sind.