reserve85/HoymilesZeroExport #89: Warten auch ACK

reserve85, 2023-08-08:
@reserve85 

Kannst Du hier bitte mal reinschauen und ggf. Dein Script anpassen?
https://github.com/lumapu/ahoy/issues/1072#issuecomment-1668660122

_Originally posted by @keinprofi in https://github.com/reserve85/HoymilesZeroExport/issues/28#issuecomment-1669031005_

reserve85, 2023-08-08:
Kann ich anschauen, dauert aber etwas da ich im wohlverdienten Urlaub bin

Ollipop030, 2023-08-14:
Moin Tobias,

ich probiere gerade die 1.50 aus. Sollte die Option SET_LIMIT_RETRY nicht obsolet sein? Das wurde doch meines Wissens nach eingebaut, da Ahoy das Limit nicht immer angenommen hat. Somit kann ja man das wiederholte Senden eines neuen Limits sparen, wenn aktuelles Limit=neues Limit ist, also SET_LIMIT_RETRY = 0.

reserve85, 2023-08-14:
Ne, kann dennoch sein dass ein Fehler zurück kommt. Ich warte max. 10s auf das Limit setzen, wenn bis dahin kein Acknowledge von Ahoy kommt gehts weiter in die nächste Runde.

Ollipop030, 2023-08-14:
Alles klar, danke für die Erklärung. Läuft übrigens super bisher.

reserve85, 2023-08-14:
Ich schau mal, dass ich nur wiederhole wenn der letzte Befehl nicht bestätigt wurde.

reserve85, 2023-08-15:
ist eingecheckt in der DEV.

Ollipop030, 2023-08-15:
> Wäre euch dankbar wenn das jemand mit mehr als einem Inverter austesten kann.

1.52 hier, sieht gut aus (habs auf das Wesentliche gekürzt):

```
2023-08-15 16:41:59,155 INFO     setting new limit to 242 Watt
2023-08-15 16:41:59,157 INFO     Ahoy: Inverter "HM-1500": setting new limit from 274 Watt to 201 Watt
2023-08-15 16:41:59,858 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
2023-08-15 16:41:59,859 INFO     Ahoy: Inverter "HM-300": setting new limit from 54 Watt to 40 Watt
2023-08-15 16:42:00,471 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
2023-08-15 16:42:00,492 INFO     Ahoy: Inverter "HM-1500" reachable: True
2023-08-15 16:42:00,515 INFO     Ahoy: Inverter "HM-300" reachable: True
2023-08-15 16:42:12,856 INFO     powermeter EMLOG: -42  Watt
2023-08-15 16:42:13,880 INFO     intermediate meter EMLOG: 204  Watt
2023-08-15 16:42:13,881 INFO     Inverterlimit already at 242 Watt
2023-08-15 16:42:13,996 INFO     Ahoy: Inverter "HM-1500" reachable: True
2023-08-15 16:42:14,021 INFO     Ahoy: Inverter "HM-300" reachable: True
2023-08-15 16:42:26,354 INFO     powermeter EMLOG: -31  Watt
2023-08-15 16:42:27,380 INFO     intermediate meter EMLOG: 203  Watt
2023-08-15 16:42:27,381 INFO     Inverterlimit already at 242 Watt
2023-08-15 16:42:27,506 INFO     Ahoy: Inverter "HM-1500" reachable: True
2023-08-15 16:42:27,530 INFO     Ahoy: Inverter "HM-300" reachable: True
2023-08-15 16:42:39,871 INFO     powermeter EMLOG: -61  Watt
2023-08-15 16:42:40,896 INFO     intermediate meter EMLOG: 203  Watt
2023-08-15 16:42:40,897 INFO     Inverterlimit already at 242 Watt
2023-08-15 16:42:41,025 INFO     Ahoy: Inverter "HM-1500" reachable: True
2023-08-15 16:42:41,050 INFO     Ahoy: Inverter "HM-300" reachable: True
2023-08-15 16:42:51,336 INFO     powermeter EMLOG: 213  Watt
2023-08-15 16:42:51,337 INFO     setting new limit to 510 Watt
2023-08-15 16:42:51,338 INFO     Ahoy: Inverter "HM-1500": setting new limit from 201 Watt to 425 Watt
2023-08-15 16:42:52,085 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
2023-08-15 16:42:52,086 INFO     Ahoy: Inverter "HM-300": setting new limit from 40 Watt to 85 Watt
2023-08-15 16:42:52,697 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
2023-08-15 16:42:54,724 INFO     intermediate meter EMLOG: 209  Watt
```

reserve85, 2023-08-15:
shit, ich hab meine Antwort ausversehen in dein Post editiert, wollte eigentlich antworten...
Dann halt jetzt:

> Nachtrag: SET_LIMIT_RETRY kann dann ja auch aus der ini verschwinden.

ne, wenn SetLimit fehl schlägt (kein Ack), dann wird so oft wiederholt wie SetLimitRetry...

reserve85, 2023-08-19:
Bei mir läuft das super zuverlässig, rollout? @Ollipop030 hast du noch zufällig was entdeckt? @keinprofi konntest du es testen?

keinprofi, 2023-08-19:
Sorry, hab zwar mitgelesen, aber bisher nicht geschafft auch nur eine Version einzuspielen. Ich versuche das morgen mal.

Ollipop030, 2023-08-20:
Also von meiner Seite aus steht dem Rollout nichts im Weg. Hab grad nochmal in die Logs der letzten zwei Tage gesehen, keine Probleme festgestellt.

keinprofi, 2023-08-20:
ini 1.50; py 1.53, Ahoy 7.36
[image]

Was läuft falsch?

Ollipop030, 2023-08-20:
> Was läuft falsch?

Er findet das Modul "packaging" nicht. Das muss seit Version 1.53 vorhanden sein, siehe hier:
https://github.com/reserve85/HoymilesZeroExport/blob/dev/CHANGELOG.md

Einfach "sudo pip3 install packaging" ausführen, dann läuft es.

keinprofi, 2023-08-20:
ok, hab ich übersehen.

Hab das gerade nochmal versucht, läuft aber immer noch nicht, liegt aber an mir, nicht am Script etc.
Ich muss das die Tage mal ganz in Ruhe machen, so auf die Schnelle wird das nix. Bin eben kein Profi....

closed 2023-08-29 by reserve85