reserve85/HoymilesZeroExport #90: Keine Installation in einem Docker Container möglich

iamheresy, 2023-08-10:
Hallo zusammen. Die Installation in einem Docker Container schlägt fehl.

```
root@ubuntu: /# cd HoymilesZeroExport / root@ubuntu:/HoymilesZeroExport#./install.sh
Installing packages
Requirement already satisfied: requests in /usr/local/lib/python3. 10/dist-packages (2.31.0)
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python3.10/dist-packages (from requests) (2023.7.22)
Requirement already satisfied: urllib3<3,>=1.21.1 in /usr/local/lib/python3.10/dist-packages (from requests) (2.0.4)
Requirement already satisfied: charset-normalizer<4,>=2
in /usr/local/lib/python3.10/dist-packages (from requests) (3.2.0)
Requirement already satisfied: idna<4,>=2.5 in /usr/local/lib/python3.10/dist-packages (from requests) (3.4)
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https: //pip.pypa.io/warnings/venv
Packages install completed
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
[Unit]
Description=HoymilesZeroExport Service
After multi-user.target
[Service]
Type=simple
Restart=always
ExecStart=/usr/bin/python3/HoymilesZeroExport/HoymilesZeroExport.py
[Installj
WantedBy=multi-user.target
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
Installation of HoymilesZeroExport completed
root@ubuntu: /HoymilesZeroExport#
```

reserve85, 2023-08-10:
Hi, du musst systemd aktivieren:
https://github.com/microsoft/WSL/issues/8883#issuecomment-1331933092

closed 2023-08-19 by reserve85

iamheresy, 2023-12-19:
Im Container oder auf der DS muss das aktiviert werden?