reserve85/HoymilesZeroExport #91: Hi, ist es möglich, dieses script direkt in Home Assistant laufen zu lassen, bzw. als Integration zu integrieren?

ottyalamo, 2023-08-11:
Hi, ist es möglich, dieses script direkt in Home Assistant laufen zu lassen, bzw. als Integration zu integrieren?

_Originally posted by @ragman11 in https://github.com/reserve85/HoymilesZeroExport/issues/28#issuecomment-1672698139_
Das fände ich auch total super.
Zur Zeit läuft das Script auf einem extra Server Raspi. 
Das Script ist top. Vielen Dank dafür

ottyalamo, 2023-08-14:
Da steh jetzt auf dem Schlauch. Wie mach ich das? Hast du dafür eine kurze Erklärung?

reserve85, 2023-08-14:
Debian auf den Raspi installieren, HomeAssistant als Supervisor installieren und du hast Debian Linux + HA laufen. So kannst du z.B. auch PiHole noch nebenbei laufen lassen.

ragman11, 2023-08-18:
Hi, ich hätte da noch eine Frage bzgl. update des Script. Ist es möglich ohne die Konfiguration zu verlieren das Script auf die 1.53 Version zu bringen?

reserve85, 2023-08-18:
Musst du leider händisch zusammenführen. Ich ersetze immer nur die *.py und passe die *.ini dann händisch an. Hat sich meist nur wenig geändert.

ragman11, 2023-08-18:
Danke für die Info, aber da fehlt mir das Know How wie ich die .py ersetzte. Die .ini anpassen ist kein Problem. Hättest du hierfür bitte ein kleines How To.  Danke...

reserve85, 2023-08-18:
https://github.com/reserve85/HoymilesZeroExport/issues/46#issuecomment-1515717688

closed 2023-08-19 by reserve85