reserve85/HoymilesZeroExport #92: AhoyDTU quasi nicht mehr ansprechbar bei laufenden Script

scharrrfi, 2023-08-14:
Hallo zusammen,

bei mir ist mein AhoyDTU (mit einem WR) bei laufenden Script quasi nicht mehr ansprechbar. Weder das Webinterface aktualisiert (bzw. lässt sich überhaupt laden) noch Updates über MQTT werden gesendet. Die AhoyDTU scheint aber noch da zu sein, da das Script korrekt mir ihr redet. Auch ist sie beim Stoppen des Scriptes direkt wieder korrekt erreichbar. 

Da die DTU bei mir schon auf einem ESP32 läuft, irgendwelche Ideen womit das zusammenhängen könnte?

Danke für die Unterstützung!

reserve85, 2023-08-14:
Liegt mit Sicherheit an den Intervallen. Poste bitte mal die Sektion Common aus der ini und die Einstellungen in Ahoy (Interval [s] + Max retries per Payload, findest du in Setting/Inverter)

scharrrfi, 2023-08-14:
Klaro:
Einmal die .ini:
`[COMMON]
# Number of Inverters
INVERTER_COUNT = 1
# max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
# if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20
# interval time for setting limit to Hoymiles
LOOP_INTERVAL_IN_SECONDS = 20
# delay time after sending limit to Hoymiles Inverter
SET_LIMIT_DELAY_IN_SECONDS = 5
# delay time after sending limit to Hoymiles Inverter when using more than one Inverter
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
# polling interval for powermeter (must be < LOOP_INTERVAL_IN_SECONDS)
POLL_INTERVAL_IN_SECONDS = 1
# if your powermeter exceeds POWERMETER_MAX_POINT: immediatelly set the limit to predefined percent of HOY_MAX_WATT (if you have more than one inverter it´s the sum of all HOY_MAX_WATT)
# value = 0 disables the feature. Values are possible from [0 to 100]
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100
# max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
# enable logging to file
ENABLE_LOG_TO_FILE = false
# how many logfiles you wish to keep
LOG_BACKUP_COUNT = 30
# defines how often a identical limit will be set, set it to "-1" for disabled (infinite repeat)
SET_LIMIT_RETRY = 10
# log the inverter temperature
LOG_TEMPERATURE = false
# delay time after turning the inverter off or on
SET_POWER_STATUS_DELAY_IN_SECONDS = 10`

Und hier hier die Ahoy Konfig:
'Interval [s]: 6
Max retries per Payload: 5'

reserve85, 2023-08-14:
Passt, da ist normal sogar noch Luft um schneller zu werden. Wie ist die Verbindung zum WR? Findet du unter Ahoy/System/Radio. Welche Ahoy Version nutzt du?

scharrrfi, 2023-08-14:
Nutze die aktuelle 0.7.26.

Hier meine Werte unter Radio:
```
nrf24l01+
is connected
Datarate
250 kbps
Power Level
LOW
TX count
855
RX success
805
RX fail
0
RX no answer
43
RX fragments
1608
TX retransmits
62
```

reserve85, 2023-08-15:
Sieht eigentlich top aus, bei mir läuft das so ohne Probleme auf der 0.7.30. hast du bei Mqtt ein Intervall eingestellt oder steht das auf default 0?

Ollipop030, 2023-08-15:
Bei mir läuft das Script mit 2 Invertern und Ahoy 0.7.33 auf ESP32, komme jederzeit auf Webinterface ohne Verzögerungen.

Hast du noch einen anderen ESP liegen? Dann den mal probieren. Anderes Netzteil hilft auch manchmal. Hast du einen Kondensator eingelötet?
Evtl ist auch beim Flashen was schief gelaufen. Bei Ahoy melden sich auch immer mal wieder Leute, die ein eigenartiges Verhalten der DTU melden, und nach einem erneuten Flash (und Erase) geht es dann.

tester277, 2023-08-15:
Bei mir lief es mit ahoy 0.7.26. stabil und zuverlässig (ca 3 Tage). Es fing damit an, dass ich die 0.7.31 installieren wollte. Ein paar Sekunden nach Upload war ahoy Webserver nicht mehr erreichbar (mit gleichlautender Fehlermeldung). Die Installation ging auch nach entfernen der Spannungsversorgung nicht mehr. Ebenso funktionierte das Speichern der Settings nicht. Der Wechselrichter war außer Tritt gekommen und musste auch neu gestartet werden. Jetzt habe ich ahoy 0.7.33. .
Ich habe die Vermutung, dass es möglicherweise an den kurzen Abfrage Intervallen (6s) liegen könnte.
Teste jetzt auch erst mal weiter mit 0.7.33.

reserve85, 2023-08-15:
Probiert bitte auch mal die neue dev Version, ein ähnliches Problem hatte auch schon @keinprofi

Mit der dev Version 1.52 (https://github.com/reserve85/HoymilesZeroExport/tree/dev) werden die Limit Befehle reduziert, da auf das acknowledge von Ahoy gewartet wird.

siehe: https://github.com/reserve85/HoymilesZeroExport/issues/89

ich konnte das Problem bei mir allerdings nicht nachvollziehen, lief immer super.

tester277, 2023-08-16:
Habe nun die dev Version 1.52 installiert und werde mich melden wenn es Probleme geben sollte.
Danke Tobias!

reserve85, 2023-08-16:
Gerne, gib bitte auch Rückmeldung wenn es läuft :-)

tester277, 2023-08-16:
Auch wenn es gut läuft melde ich mich (ein paar Tage wird es wohl dauern)

scharrrfi, 2023-08-19:
Also bei mir schien es wohl tatsächlich am WLAN gelegen zu haben. Bisschen Position der DTU geändert und schon sind die Probleme behoben. DTU bleibt erreichbar und wird ordentlich gesteuert vom Script. Danke trotzdem für eure Mühen!

tester277, 2023-08-20:
Bei mir lief es stabil in der Konstellation Ahoy 0.7.37 HoymilesZeroExport 1.52.
Nun habe ich auf die 1.53 upgedated. Mal schauen, wie es sich jetzt entwickelt.
Es ist ja (Dank der fleissigen Entwickler) viel Bewegung drin.

closed 2023-09-18 by reserve85